betaflight/betaflight #4259: BF3.2 RC6 Flip over after crash (aniti-turtle) does not reverse motors

HaiSaiFPV, 2017-10-01:
Just flashed RC6 and enabled flip over after crash. everything works as it should accept for reversing the motors, only two motors spin with roll and pitch input as it should but they spin the normal direction.

Kakute F4 FC
Wraith 25A Blheli32 ESCs

UltiFix, 2017-10-01:
I also have this issue....

Omnibus F4 FC
Kiss 24A Esc's flashed with V108F

brycedjohnson, 2017-10-01:
Are you using the beta version of blheli32?  Did it work before?

fiveangle, 2017-10-02:
Just in case it helps track down the area of the issue, on RC5, OMNIBUS(F3) target, [generic micro 20x20 20A 4:1 esc,](https://www.gearbest.com/esc/pp_618556.html), BLHeli_S 16.67, DSHOT600, each ESC set reversed in BLHeli Configurator, BF set to "Motor direction is reversed", props reversed, and _Turtle Flip Mode_ on switch functions as expected.

glenmurphy, 2017-10-04:
Same issue with my Tinyfish - in flip mode the motors go in the wrong direction (pulls that side towards the ground).

I wired all my motors backwards, so I reversed the motors in BLHeli, which let me keep everything the regular way around in BetaFlight (props are normal direction and "motor direction reversed" is "off").

Betaflight: Using DSHOT600 / FC Yaw Offset of -45
BLHeli: S_H_50  + 16.7 Bidirectional Reversed

brycedjohnson, 2017-10-04:
@glenmurphy Yeah the escs in 3d mode I don't think will do turtle mode successfully.

For everyone else, Reverse prop directions or esc motor direction reversed have been tested with my setup should be working fine.

glenmurphy, 2017-10-04:
@brycedjohnson thanks! I updated to 16.7 at the same time I swapped from regular reversed to bidirectional reversed (16.6 with regular reversed didn't work) - going back to regular reversed with 16.7 worked great!

mikeller, 2017-10-04:
@HaiSaiFPV: Did an update of the ESC firmware to 32.15 solve this for you?

claytonbakerjr, 2017-10-05:
I will be following this thread. Seems I too have something odd going on pertaining to Flip Over After Crash. This has given me some other items to check/pay attention to, to determine if I too have an issue.
Setup: CL Racing F4 flashed with RC6 (Flip After Crash enabled) Small Angle set to Default (Accelerometer Disabled)
Cicada 30Amp BB2 esc's flashed with 16.67
2205/2600Kv Motors.
Running Reverse Prop Rotation
Tested DShot 600 and DShot 300
Actions: once upside down in grass (2 or 3 inches)Arm switch in off position, Activate FOAC, Arm, jog Pitch/Roll not much happens other than the props try to spin but never accelerate enough to flip in fact hardly turn at all.
On a clean surface very little grass following proper procedure for FOAC Quad flips violently in each case (DShot600/DShot300) Very Violently. Very similar to having props on incorrectly. Almost flips 2 times before settling. 
Once rain lets up I will test DShot150, and set small angle to 180 (but with acc disabled does it matter?)
I want to pay attention to motor direction during test to see if this is also my situation.

brycedjohnson, 2017-10-05:
For testing you don't need props on and don't need to be upside down.  Arm
and check that the motors move when you tilt the pitch and roll sticks.
They will increase in speed the more it tilts.

I haven't had any problems but I'm running a night between RC5 and RC6.

claytonbakerjr, 2017-10-05:
@brycedjohnson 
Will do and thanks

controlflow, 2017-10-07:
Anti-turtle mode do not reverses my motors too...
It works perfectly on BLHeli_32 quad, but no luck with all my other BLHeli_S quads.
DSHOT/ESC beeper works perfectly on all of the quads. Betaflight RC6.
One of the quads has, for example, T-Motor 45A, BLHeli_S C-H-40 16.67_dshotcmd.
Motors direction reversed, bringing it back to normal do not helps.
Two quads (BLHeli_32-based with anti-turtle mode working and BLHeli_S based that do not reverses props) have motors remapped (FC rotated 180 degrees yaw) if this matters.
Bidirectional modes are not used.

controlflow, 2017-10-08:
Ah, nevermind, that quad is gone 😭
I think I've managed to get turtle mode motor reverse working after flashing different BLHeli_S firmwares a few times (ending with 16.67_dshotcmd) and switching motor directions back and forth once.

Jimgitsit, 2017-10-15:
I've got this same issue with BF 3.2 release, blheli_32 spedix escs running dshot 1200, and a dys F4 FC. Motors don't spin in reverse.

brycedjohnson, 2017-10-15:
@Jimgitsit Can you take a screenshot of what version of blheli32 you are running?

fiveangle, 2017-10-15:
> On a clean surface very little grass following proper procedure for FOAC Quad flips violently in each case (DShot600/DShot300) Very Violently. Very similar to having props on incorrectly. Almost flips 2 times before settling.

This seems to be the consensus from others too.  Someone filed an request on CF tracker to set a max ESC limit for turtle flip mode, but that just seems like a kludge.

@brycedjohnson - I tried to trace it back, but couldn't figure it out: do expo or super rates get applied to stick inputs before getting sent to the mechanism to power the motors in reverse ?

The trouble is that it really needs some kind of auto-attenuation during the roll since the normal force is dropping as a _sin_ of the roll angle as it rotates towards upright.  Once it's at 90, there is no gravity to counter-act so it's effectively got an additional 9.8m/s<sup>2</sup> force once it's at the top, and no matter how fast someone is on the sticks, that will be hard to prevent from over-shooting.

Is it possible to add a gyro mix to drop motor force automatically by sinθ of the rotation ?  If that's too complex, even just shut it complete off at that point (requiring stick-center to reset ?).  I'm not sure the best attack but I don't think it would take much to improve the experience.  One may only get a single chance to right their craft on a roof building so we probably want to make it the best effort possible ;)

brycedjohnson, 2017-10-15:
@fiveangle No rates get applied to the pitch or roll.  It is linear 0 - 100% motor spining from center throttle to the edge.  Actually on RC4 or 5 there was a bug that it only went to ~70-80%.  On my 2 inch quad to flip over it definitely helped to have that extra 20%.

Limiting the max throttle is basically the same thing as not push the stick over quite as far.

Are they only using the 2 props spinning on cleanflight as well?  Or is it the older code with all 4 props spinning? 

The ACC could get brought in to auto level or auto turtle, but many don't run with that on.

It could be the ESCs aren't receiving the command to switch all the time.  There was a early version of 16.63 that was more stringent (there was multiple versions of 16.63 release for some reason).  If someone notices that it is only sometimes flipping violently that may be the issue.

It might be a technique thing.  Just blip the throttle and stop before you all ready flipped over.

nrangus, 2017-10-15:
New here Iam also having the same issue! The mode works, just esc are not reversing motor direction. Beeper mode also not working I see the fc flashing the mode, esc not beeping. Esc beep on arm. Using dys F4 pro pdb fc with dys 35a aria 32bit esc. Is this a 32 bit issue? I am not able to use the blheli configurator app in chrome. Read a comment that the developer wont support 32bit esc? Only able to use blheli suit 32, esc firmware is current as far as suite is concerned. Personally feel the esc are the issue but idk! Check the betaflights firmware flasher only version up is 3.2.0 no rc.

fiveangle, 2017-10-15:
@brycedjohnson - to be clearly, I did not advocate for a throttle limit:

> Someone filed an request on CF tracker to set a max ESC limit for turtle flip mode (cleanflight#2963), but that just seems like a kludge.

> It might be a technique thing. Just blip the throttle and stop before you all ready flipped over.
Have you used the feature IRL ?  :)

My quad is working exactly as expected with correctly configured components but the violent flip over for the pure physics reasons I described is very difficult to "technique" your way out of.

While watching the quad on the floor, I am able to manipulate the quad to flip it over if I practice the finess required to do so.  Doing so in goggles with any precision is an act of futility.  Anyway, it's better than nothing I suppose.  I was just trying to think of how to make it actually work on the first try 70% of the time.

mikeller, 2017-10-15:
As a somewhat simpler way to kludge around this, a 'burst time' parameter could be introduced that would allow the user to set a configurable time limit for flip over bursts. This would allow users to configure the required amount of thrust in LOS, and then rely on the burst time limit to keep their craft from doing a 360 in FPV.

brycedjohnson, 2017-10-15:
@fiveangle. Lol! I have likely used this feature more then anyone else by far.  I haven't had much problems with both a top mount battery or bottom mount, 2 5 and 6 inch.  Mostly has been on grass, so maybe concrete or very trimmed grass would be cause an issue.

If you had a video of it fpv or los that works be useful to visualize what you are describing.

brycedjohnson, 2017-10-15:
@nrangus did you want to get a screenshot of your blheli32 version.  Check with blheli and see if they are supported yet.

nrangus, 2017-10-16:
@brycedjohnson Id like a screenshot of my blheli32 version how do i acquire this?

brycedjohnson, 2017-10-17:
Hit save screenshot

[image]

Rabter1, 2017-11-21:
I have the same issue with Omnibus F4 v3 
And BLHeli_32 ESC's 
I updated to Betaflight 3.2.2
and ESC's to 32.1 
but i still have the issue

brycedjohnson, 2017-11-21:
@Rabter1 32.1 doesn't support dshot commands. Check with your manufacturer or blheli32.  Not a betaflight bug.

Rabter1, 2017-11-21:
But it also didnt work at 32.0 :/ 
I dont know where i can find the right informations about this. Sorry if i its not an Betaflight Bug and i am wrong here...

brycedjohnson, 2017-11-21:
@Rabter1 https://github.com/bitdump/BLHeli.  You'll need 32.3 installed on there

Rabter1, 2017-11-21:
@brycedjohnson Thanks man! 
32.3 fixes the issue.
I watched the video of joshua bardwell an he said every blheli32 firmware supports dshot commands... so i didnt expect the firmware to be the issue :/

Dgr8one85, 2018-01-24:
Does anyone know if a work around or solution to this has been found?  When "Flip Over After Crash" mode is activated, my motors still spin their normal direction.  Only 2 motors spool up at a time (as they should) but the direction is not reversed.  I have a suspicion this issue is prevelant with Spedix ESC's used along with prop directions reversed configurations.  I have not resolved this issue, this is my current setup:

CLR4
BF 3.2.4
Spedix IS30 Dshot ESCS
Prop Directions Reversed

brycedjohnson, 2018-01-24:
Likely you escs need to be updated.  16.7 supports dshot commands such as
reverse motor.

Dgr8one85, 2018-01-24:
as far as i can tell, i'm running the latest firmware on everything.  16.7 on my escs.

brycedjohnson, 2018-01-24:
Prop directions reversed doesn't matter either.  I'm running that on all my
quads.   It has been out quite a while so I'd be surprised if there was a
bug.  Everything I've heard about has been configuration errors or wrong
version.  You could check to see if the dshot beeper is working?  That uses
the same method as the reverse motors.

leeahoward, 2018-01-27:
@brycedjohnson I have been having this issue(motors won't reverse, or they will reverse into turtle mode and won't go back to normal) blheli_s 16.67 or 16.67.  I have narrowed it down and found a way to make it work for me, but I do think there is a bug.  

I have a switch set up for prearm.  I also have the beeper on the same switch.  Turtle mode is on another switch. With the beeper on the prearm switch the motors will occasionally reverse to turtle mode, but most of the time they won't.  If I disable the beeper on the prearm switch turtle mode works every time. 

I have been having this issue on 4 different quads and just worked it out today.

possibly related to #4602

yannickspill, 2018-02-02:
I experience a similar problem, but managed to solve it. However it sounds like a bug to me.
First, I downgraded my ESCs to 16.67_dshotcmd. Then in betaflight configurator, I select DSHOT300 (I was using DSHOT600 before). Hit save. But here comes the catch. I get 5 short beeps, then 3 long ones. I go back to configure tab, and I see it's still on DSHOT600. So I re-select DSHOT300 and hit save. Now I hear 5 short beeps and 2 long ones. In the config tab, DSHOT300 is selected. But if I try turtle, the props spin in the normal direction. So in the config tab, I just hit save without changing anything. Again, 5 beeps then 2, but this time turtle mode inverts motor spin, like it should. Also, note that it is best to do this save operation 3 times in a row without touching the TX. Touching the TX somehow prevents that stuff from working.

claytonbakerjr, 2018-02-13:
@fiveangle 
Thanks for the reply early on in this and sorry I delayed any responses. I pretty much gave up on FOAC. However with the fact that most everyone is using it in races now I am going to jump back on the wagon and set this back up again since the weather here in Cinci is starting to get better.
After going back through this thread (thanks for everyones inputs) it seems that I need to be at least on 16.7 (for BL Heli S) and 32.2 on (BLHELI32) I will be running DSHOT600 in both cases. I will also make sure to be running the latest BF version as well during my setup of FOAC.

claytonbakerjr, 2018-02-15:
So after yesterday's post pertaining to FOAC I did more research and I still have a lingering question that is not sticking out...and likely due to my lack of knowledge...so try not to beat me up too much :)
I see in the set up phase of FOAC, setting small angle to 180... I have a question...
I turn off my accelerometer, doesn't this essentially tell the quad it can arm (in a manner of speaking) at any angle? Hence do I still need to set small angle? I just want to make sure doing both will not or has not caused my problems in the past where the quad freaks out when flipping. Early on I was told early on in quads turning off acc will help me arm if Im upside down or in a odd predicament, so I just kept doing it...??

brycedjohnson, 2018-02-15:
@claytonbakerjr Small angle isn't checked for FOAC mode anymore to activate.  Regular arming checks that.  If you disable the acc then nothing checks it.

claytonbakerjr, 2018-02-15:
@brycedjohnson ok great thanks for the info Bryce, so I need not set it as per YouTube videos.
Does anyone have an updated procedure out there or within GitHub so as I do it correctly?

brycedjohnson, 2018-02-15:
Make sure escs are up to date with the latest (16.7 for blheli_s)
Enable the FOAC mode to a switch
Remove props
Test that just 2 motors spin in the opposite direction using pitch and roll input.  With no pitch and roll none of the props spin.

Some people do small angle = 180 or disable the Acc in case the quad is flipped but at > 30 degrees (default small angle i think)

claytonbakerjr, 2018-02-15:
@brycedjohnson thank you Sir

claytonbakerjr, 2018-02-19:
I was able to get to the FOAC setup last night and confirmed that the motors are all spinning in the correct direction when FOAC is enabled so now I need to just get out and do a real world test.
Thanks again @brycedjohnson

beastlykings, 2018-05-09:
@leeahoward

I've just set up turtle mode and am having similar issues to you I think. I was testing it on my new build because I haven't gotten around to updating my other builds to bf 3.2, but it was only reversing the motors sometimes, not all the time. The thing we have in common? I have my esc beeper set to the same switch and switch position as turtle mode. It's 2am here so I really don't have time to test further, I need sleep, but tomorrow I will for sure. And tomorrow I'll check all my versions, I haven't gotten that far yet. But ESC commands are working because beeper works, and the escs reverse for turtle mode at least sometimes.

Really frustrating bug! But I truly think this is a bug, in something.

brycedjohnson, 2018-05-09:
Yeah you are sending beeps and turtle to the ESC at basically the same time (sends the turtle command on arming) and it can miss commands because the esc can't receive another command when a beep is happening.  Also depending on ESC version, some are worse then others.  Update everything should help.

But the issue will be reduced due to https://github.com/betaflight/betaflight/pull/5707

closed 2018-06-15 by stale[bot]