betaflight/betaflight #4263: Carte YOUPIF4 et GPS

Stef263, 2017-10-01:
# If your issue looks like a hardware fault or a configuration problem please don't raise an issue here.

## Please consider using other user support options such as asking the manufacturer of the hardware you are using, [RCGroups](https://rcgroups.com/forums/showthread.php?t=2464844), [Slack](https://slack.betaflight.tech) or other user support forums & groups (e.g. facebook).

## If you believe there is an issue with the firmware itself please follow these steps:
1.) Describe your problem;
2.) Include ways to reproduce the problem;
3.) Provide information about your flightcontroller and other components including how they are connected/wired;
4.) Add the used configuration and firmware version;
5.) Create a `diff` and post it here in a code block. Put \`\`\` (three backticks) at the start and end of the `diff` block; and
6.) Remove this Text :).

Stef263, 2017-10-01:
Bonjour,
je n'arrive pas à installer un GPS sur une carte YUPIF4!!!
pouvez vous modifier le code ? 
sur le logiciel INAV il n'y a pas de problème

Hello,
I can not install a GPS on a YUPIF4 card !!!
can you change the code?
about INAV software there is no problem

jflyper, 2017-10-02:
@RcNet @Faduf Can you folks take care of this? Preferably at your support site?

Faduf, 2017-10-07:
I've tested today on a fixed wing with the 3.2 RC6 and no problem with the GPS on the UART1 of the YupiF4. I think this issue is only a configuration problem. 
@Stef263 which type of GPS are you using? which protocol have you chosen, NMEA or UBLOX?

closed 2017-10-21 by jflyper