betaflight/betaflight #4264: Clean blackbox logs in OSD (not really issue)

Bugg69s, 2017-10-01:
Hi, I found out that is annoying me to clean the logs, I have just old PC with no SD card reader and to delete logs I need to use my phone or dongle that is slow as hell and not really working right. Will be possible to delete logs in OSD ?
 It will save a lot of time.

jflyper, 2017-10-02:
You mean you want to selectively delete one or more log files in _**SD card**_?

Btw, deleting the entire file can be done through CMS.

Bugg69s, 2017-10-02:
I want to use OSD to delete some files, I will have like full Card and want to keep some new logs. That I can delete some through OSD. Or you can add command line to OSD

Bugg69s, 2017-10-02:
It will be no need for PC or laptop to setup your quad, just use command line and notepad with commands you wanna use

Arakon, 2017-10-02:
You do realize that even on a 4 gig card, you can keep several hundred logs before it gets full?

Bugg69s, 2017-10-03:
I have only 1gb card and i'm too lazy to insert into my phone, it will save my time

jflyper, 2017-10-03:
@Bugg69s I've put a label for feature request and edited the title a little.

closed 2026-09-26 by blckmn