betaflight/betaflight #4268: Accelerometer not working correct - quad flies crooked after throttle punches

VoicOfReason, 2017-10-03:
With 2 different flight controllers (Betaflight F3 and F7V2) with 2 different sets of motors and receivers (Crossfire and Frsky) my quad is flying crooked after throttle punches after a stable/level lift off. I'm using 3.2RC5 and just tried RC6. Same problem on both. I take off level, but after a few hard throttle punches it flies so off-level that I can barely control it, i.e. have to crank stick hard to one side to keep it level. It sometimes get progressively worse. 

I think it's related to this issue, but I'm not a programmer here and don't know status of this fix: https://github.com/betaflight/betaflight/pull/4218

Is this a problem with ACC filtering? Will it be fixed soon?

Related post: https://github.com/betaflight/betaflight/issues/797

RipperDrone, 2017-10-05:
It sounds like this one might rather be related to #3723 

Maybe you want to share a log file for the devs to analyze further.

I am also plagued by the same issue - on all my BF 3.2 copters (3 of them right now, different FCs and ESCs). They were ok on BF 3.1.7, though. I suspect it's related to PID sum overflow, anti_gravity or something else in the controller loop which generates a PID offset error...

Seems there is no clear direction right now :-(

VoicOfReason, 2017-10-05:
I just finished a 4" build last night with BF3.2RC6 on Omnibus F4 and it
does NOT have this problem. However, I've had 3 other builds that DO have
the problem. I hope they escalate this issue as it involves ability of
aircraft to fly and for safety reasons. My kids are learning to fly and I
don't want a drone hitting me or their mother in the face.

jflyper, 2017-10-06:
Why do different builds with the same firmware behaves differently?

VoicOfReason, 2017-10-06:
I don't know. The settings between models are almost the same in terms of
gyro rates, filter settings, anti-gravity setting, etc.

mikeller, 2017-10-06:
Could this be caused by erratic behaviour of acc / gyro that's not properly filtered out?

RipperDrone, 2017-10-06:
Haven't touched the gyro filtering at all btw versions 3.1.7 (working ok) and 3.2 RC5/6 (not ok) on my builds, though.

Seems to be linked to today's strong motor (most pronounced dead on strongest motors), this might be an explanation why, it does not show up on 4" builds or low power builds consistently. Strong motors make high acceleration bursts, fast attitude changes, this is when it happens... Must be something in PID controller handling fast control inputs / high rate attitude changes, like capped pidsum, anti_gravity, ... (Tried anti_gravity off, values up to 9,000, all the same :-()

tarekmaalouf, 2017-10-06:
I ended up doing the following - went to an older version of the firmware 3.1.* and then loaded - all tested perfectly. - Went back to the firmware for 3.2.* and this time it was fine. It seems to me that there is an issue in the software only - not motor specific in the sense that this issue displays in BF when just configuring without motor action. Not sure what the cause is but reloading the firmware over itself a second time has yielded a more positive outcome. Wish this could get fixed...

mikeller, 2017-10-06:
There seem to be a lot of problems at the moment with configuration settings not properly being migrated when updating the firmware. As a general recommendation, when updating the firmware, it is always a good idea to do a backup in CLI, update the firmware, reset to defaults, restore the backup.

VoicOfReason, 2017-10-06:
Cyclone3D, so you loaded 3.1 default settings and then immediately flashed
3.2 again and set up and it was fixed?

tarekmaalouf, 2017-10-06:
Voice - that is exactly what I did and the damn thing calibrated without error and flew perfect. Not sure what gives on these - but it works for now

VoicOfReason, 2017-10-06:
I wish I could do that, but Omnibus V7 doesn't have a 3.1 version at all.
Today I tried re-flashing 3.2 and re-doing settings, but it didn't fix the
problem.

tarekmaalouf, 2017-10-06:
Can you post your diff readings from the CLI? Also, do you have barometer turned on? Just curious. I wanted to compare screen shots so that I could see if I could duplicate your error. I have roughly 1000 FC's in stock for my clients and can grab some to test and see if I can help.

VoicOfReason, 2017-10-06:
# name

name WPF-Talon

# resources

# mixer

# servo

# servo mix

# feature

feature INFLIGHT_ACC_CAL

feature MOTOR_STOP

feature TELEMETRY

feature RSSI_ADC

feature LED_STRIP

feature ANTI_GRAVITY

feature DYNAMIC_FILTER

# beeper

# map

# serial

serial 1 0 115200 57600 0 115200

serial 2 64 115200 57600 0 115200

serial 5 8192 115200 57600 0 115200

serial 6 0 115200 57600 0 115200

# led

led 0 3,14::C:1

led 1 4,14::C:1

led 2 5,14::C:1

led 3 6,14::C:1

led 4 7,14::C:1

led 5 8,14::C:1

led 6 9,14::C:1

led 7 10,14::C:1

# color

# mode_color

# aux

aux 0 0 0 1925 2100

aux 1 1 1 900 1600

aux 2 13 2 1900 2100

aux 3 28 1 1350 2100

aux 4 35 3 1900 2100

# adjrange

# rxrange

# vtx

# rxfail

# master

set align_gyro = CW270

set gyro_notch1_hz = 0

set gyro_notch2_hz = 0

set align_acc = CW270

set baro_hardware = NONE

set max_check = 2000

set serialrx_provider = CRSF

set blackbox_device = NONE

set motor_pwm_protocol = DSHOT600

set align_board_yaw = 180

set bat_capacity = 1300

set vbat_max_cell_voltage = 44

set vbat_warning_cell_voltage = 10

set gps_provider = UBLOX

set gps_auto_baud = ON

set pid_process_denom = 1

set osd_cap_alarm = 1300

set osd_vbat_pos = 2228

set osd_rssi_pos = 2279

set osd_tim_1_pos = 2212

set osd_tim_2_pos = 33

set osd_flymode_pos = 333

set osd_throttle_pos = 225

set osd_crosshairs = 200

set osd_ah_sbar = 200

set osd_current_pos = 385

set osd_mah_drawn_pos = 2246

set osd_craft_name_pos = 362

set osd_gps_lon_pos = 82

set osd_gps_lat_pos = 65

set osd_gps_sats_pos = 57

set osd_home_dir_pos = 302

set osd_home_dist_pos = 303

set osd_compass_bar_pos = 266

set osd_altitude_pos = 247

set osd_pid_roll_pos = 423

set osd_pid_pitch_pos = 455

set osd_pid_yaw_pos = 487

set osd_debug_pos = 1

set osd_power_pos = 321

set osd_warnings_pos = 2473

set osd_avg_cell_voltage_pos = 2260

set osd_pit_ang_pos = 257

set osd_rol_ang_pos = 289

set osd_battery_usage_pos = 2504

set osd_disarmed_pos = 138

set osd_nheading_pos = 311

set osd_nvario_pos = 279

set osd_esc_tmp_pos = 82

set osd_esc_rpm_pos = 83

set osd_stat_bbox = OFF

set osd_stat_bb_no = OFF

set osd_stat_tim_1 = ON

set osd_stat_tim_2 = OFF

# profile

profile 0

set dterm_lowpass_type = PT1

set anti_gravity_gain = 5000

set p_pitch = 63

set i_pitch = 60

set p_roll = 43

set i_roll = 50

set p_yaw = 75

set i_yaw = 50

set p_level = 90

set i_level = 85

set d_level = 100

set level_limit = 85

# rateprofile

rateprofile 0

set roll_srate = 75

set pitch_srate = 75

set yaw_srate = 80

#

VoicOfReason, 2017-10-06:
`
# name
name WPF-Talon

# resources

# mixer

# servo

# servo mix

# feature
feature INFLIGHT_ACC_CAL
feature MOTOR_STOP
feature TELEMETRY
feature RSSI_ADC
feature LED_STRIP
feature ANTI_GRAVITY
feature DYNAMIC_FILTER

# beeper

# map

# serial
serial 1 0 115200 57600 0 115200
serial 2 64 115200 57600 0 115200
serial 5 8192 115200 57600 0 115200
serial 6 0 115200 57600 0 115200

# led
led 0 3,14::C:1
led 1 4,14::C:1
led 2 5,14::C:1
led 3 6,14::C:1
led 4 7,14::C:1
led 5 8,14::C:1
led 6 9,14::C:1
led 7 10,14::C:1

# color

# mode_color

# aux
aux 0 0 0 1925 2100
aux 1 1 1 900 1600
aux 2 13 2 1900 2100
aux 3 28 1 1350 2100
aux 4 35 3 1900 2100

# adjrange

# rxrange

# vtx

# rxfail

# master
set align_gyro = CW270
set gyro_notch1_hz = 0
set gyro_notch2_hz = 0
set align_acc = CW270
set baro_hardware = NONE
set max_check = 2000
set serialrx_provider = CRSF
set blackbox_device = NONE
set motor_pwm_protocol = DSHOT600
set align_board_yaw = 180
set bat_capacity = 1300
set vbat_max_cell_voltage = 44
set vbat_warning_cell_voltage = 10
set gps_provider = UBLOX
set gps_auto_baud = ON
set pid_process_denom = 1
set osd_cap_alarm = 1300
set osd_vbat_pos = 2228
set osd_rssi_pos = 2279
set osd_tim_1_pos = 2212
set osd_tim_2_pos = 33
set osd_flymode_pos = 333
set osd_throttle_pos = 225
set osd_crosshairs = 200
set osd_ah_sbar = 200
set osd_current_pos = 385
set osd_mah_drawn_pos = 2246
set osd_craft_name_pos = 362
set osd_gps_lon_pos = 82
set osd_gps_lat_pos = 65
set osd_gps_sats_pos = 57
set osd_home_dir_pos = 302
set osd_home_dist_pos = 303
set osd_compass_bar_pos = 266
set osd_altitude_pos = 247
set osd_pid_roll_pos = 423
set osd_pid_pitch_pos = 455
set osd_pid_yaw_pos = 487
set osd_debug_pos = 1
set osd_power_pos = 321
set osd_warnings_pos = 2473
set osd_avg_cell_voltage_pos = 2260
set osd_pit_ang_pos = 257
set osd_rol_ang_pos = 289
set osd_battery_usage_pos = 2504
set osd_disarmed_pos = 138
set osd_nheading_pos = 311
set osd_nvario_pos = 279
set osd_esc_tmp_pos = 82
set osd_esc_rpm_pos = 83
set osd_stat_bbox = OFF
set osd_stat_bb_no = OFF
set osd_stat_tim_1 = ON
set osd_stat_tim_2 = OFF

# profile
profile 0

set dterm_lowpass_type = PT1
set anti_gravity_gain = 5000
set p_pitch = 63
set i_pitch = 60
set p_roll = 43
set i_roll = 50
set p_yaw = 75
set i_yaw = 50
set p_level = 90
set i_level = 85
set d_level = 100
set level_limit = 85

# rateprofile
rateprofile 0

set roll_srate = 75
set pitch_srate = 75
set yaw_srate = 80

# `

VoicOfReason, 2017-10-06:
> 
# name
name WPF-Talon

# resources

# mixer

# servo

# servo mix

# feature
feature INFLIGHT_ACC_CAL
feature MOTOR_STOP
feature TELEMETRY
feature RSSI_ADC
feature LED_STRIP
feature ANTI_GRAVITY
feature DYNAMIC_FILTER

# beeper

# map

# serial
serial 1 0 115200 57600 0 115200
serial 2 64 115200 57600 0 115200
serial 5 8192 115200 57600 0 115200
serial 6 0 115200 57600 0 115200

# led
led 0 3,14::C:1
led 1 4,14::C:1
led 2 5,14::C:1
led 3 6,14::C:1
led 4 7,14::C:1
led 5 8,14::C:1
led 6 9,14::C:1
led 7 10,14::C:1

# color

# mode_color

# aux
aux 0 0 0 1925 2100
aux 1 1 1 900 1600
aux 2 13 2 1900 2100
aux 3 28 1 1350 2100
aux 4 35 3 1900 2100

# adjrange

# rxrange

# vtx

# rxfail

# master
set align_gyro = CW270
set gyro_notch1_hz = 0
set gyro_notch2_hz = 0
set align_acc = CW270
set baro_hardware = NONE
set max_check = 2000
set serialrx_provider = CRSF
set blackbox_device = NONE
set motor_pwm_protocol = DSHOT600
set align_board_yaw = 180
set bat_capacity = 1300
set vbat_max_cell_voltage = 44
set vbat_warning_cell_voltage = 10
set gps_provider = UBLOX
set gps_auto_baud = ON
set pid_process_denom = 1
set osd_cap_alarm = 1300
set osd_vbat_pos = 2228
set osd_rssi_pos = 2279
set osd_tim_1_pos = 2212
set osd_tim_2_pos = 33
set osd_flymode_pos = 333
set osd_throttle_pos = 225
set osd_crosshairs = 200
set osd_ah_sbar = 200
set osd_current_pos = 385
set osd_mah_drawn_pos = 2246
set osd_craft_name_pos = 362
set osd_gps_lon_pos = 82
set osd_gps_lat_pos = 65
set osd_gps_sats_pos = 57
set osd_home_dir_pos = 302
set osd_home_dist_pos = 303
set osd_compass_bar_pos = 266
set osd_altitude_pos = 247
set osd_pid_roll_pos = 423
set osd_pid_pitch_pos = 455
set osd_pid_yaw_pos = 487
set osd_debug_pos = 1
set osd_power_pos = 321
set osd_warnings_pos = 2473
set osd_avg_cell_voltage_pos = 2260
set osd_pit_ang_pos = 257
set osd_rol_ang_pos = 289
set osd_battery_usage_pos = 2504
set osd_disarmed_pos = 138
set osd_nheading_pos = 311
set osd_nvario_pos = 279
set osd_esc_tmp_pos = 82
set osd_esc_rpm_pos = 83
set osd_stat_bbox = OFF
set osd_stat_bb_no = OFF
set osd_stat_tim_1 = ON
set osd_stat_tim_2 = OFF

# profile
profile 0

set dterm_lowpass_type = PT1
set anti_gravity_gain = 5000
set p_pitch = 63
set i_pitch = 60
set p_roll = 43
set i_roll = 50
set p_yaw = 75
set i_yaw = 50
set p_level = 90
set i_level = 85
set d_level = 100
set level_limit = 85

# rateprofile
rateprofile 0

set roll_srate = 75
set pitch_srate = 75
set yaw_srate = 80

#

tarekmaalouf, 2017-10-06:
Just curious - there are a few things I would like you to try but these are only because I am comparing to a few other setups..just to see if it helps...

1. disable all features except telemetry
 - meaning no gps, no anti gravity, nothing. I would ask that you activate horizon mode to test in order for me to duplicate here...but outside of horizon, everything else on modes is off and on features, nothing is set except telemetry. Perhaps OSD if necessary but I do not know how far you have to fly this to duplicate the errors. 

Once everything is disabled except for these, can you tell me if the calibration is level in BF and also if your readings are set. 

I understand if you do not want to make these changes..so don't worry, but I just thought I could duplicate the issue here if those were made since we do not attach GPS to our side or use the other features for testing. 

In the meantime I am going to pull a board and see if I can duplicate using your settings. Is your FC turned to the back?

VoicOfReason, 2017-10-06:
I’m happy to troubleshoot. I’ll get back to you in a few hours, have some
things to take care of.

RipperDrone, 2017-10-06:
to complement the bug search: Here is my diff from one of the 3 quads displaying the same behavior:

Do you spot anything abnormal here?

# version
# Betaflight / OMNIBUSF4SD (OBSD) 3.2.0 Sep 28 2017 / 16:38:50 (ceb6f70) MSP API: 1.36

# reset configuration to default settings
defaults nosave

# name
name Thor

# resources

# mixer

# servo

# servo mix

# feature
feature TELEMETRY
feature ANTI_GRAVITY

# beeper

# map
map TAER1234

# serial
serial 0 64 115200 57600 0 115200
serial 2 32 115200 57600 0 115200

# led
led 0 8,15:S:CI:6
led 1 7,15:S:C:6
led 2 6,15:S:CI:6
led 3 6,14:S:CI:6
led 4 7,14:S:C:6
led 5 8,14:S:CI:6

# color

# mode_color
mode_color 6 1 9

# aux
aux 0 0 0 1300 1700
aux 1 28 1 900 1300
aux 2 2 1 1700 2100
aux 3 27 0 1700 2100
aux 4 13 0 1700 2100

# adjrange
adjrange 0 0 3 900 1025 0 2
adjrange 1 0 3 1025 1075 6 2
adjrange 2 0 3 1075 1125 7 2
adjrange 3 0 3 1125 1175 8 2
adjrange 4 0 3 1175 1225 9 2
adjrange 5 0 3 1225 1300 10 2
adjrange 6 0 3 1300 1375 11 2
adjrange 7 0 3 1375 1425 4 2
adjrange 8 0 3 1425 1475 5 2
adjrange 9 0 3 1475 1525 1 2

# rxrange

# vtx

# rxfail

# master
set gyro_notch1_hz = 0
set gyro_notch1_cutoff = 0
set moron_threshold = 32
set mag_declination = 150
set baro_hardware = NONE
set min_check = 1040
set max_check = 1980
set max_aux_channels = 4
set serialrx_provider = SBUS
set blackbox_device = NONE
set min_throttle = 1040
set motor_pwm_protocol = DSHOT600
set motor_pwm_rate = 32000
set failsafe_off_delay = 25
set failsafe_throttle = 1150
set failsafe_kill_switch = ON
set failsafe_throttle_low_delay = 50
set vbat_max_cell_voltage = 45
set vbat_scale = 112
set ibata_scale = 300
set yaw_motors_reversed = ON
set 3d_deadband_low = 1470
set 3d_deadband_high = 1530
set 3d_neutral = 1500
set 3d_deadband_throttle = 0
set small_angle = 180
set gps_provider = UBLOX
set gps_sbas_mode = EGNOS
set pid_process_denom = 1
set tlm_inverted = ON
set frsky_gps_format = 1
set osd_vbat_pos = 2049
set osd_rssi_pos = 40
set osd_tim_1_pos = 54
set osd_flymode_pos = 465
set osd_throttle_pos = 225
set osd_vtx_channel_pos = 377
set osd_ah_sbar = 200
set osd_ah_pos = 200
set osd_current_pos = 2071
set osd_mah_drawn_pos = 2105
set osd_craft_name_pos = 2060
set osd_gps_speed_pos = 34
set osd_gps_lon_pos = 82
set osd_gps_lat_pos = 65
set osd_gps_sats_pos = 1
set osd_home_dir_pos = 257
set osd_home_dist_pos = 257
set osd_compass_bar_pos = 256
set osd_altitude_pos = 65
set osd_pid_roll_pos = 423
set osd_pid_pitch_pos = 455
set osd_pid_yaw_pos = 487
set osd_debug_pos = 1
set osd_power_pos = 47
set osd_pidrate_profile_pos = 345
set osd_warnings_pos = 329
set osd_avg_cell_voltage_pos = 76
set osd_pit_ang_pos = 257
set osd_rol_ang_pos = 289
set osd_battery_usage_pos = 392
set osd_disarmed_pos = 2411
set osd_nheading_pos = 1
set osd_nvario_pos = 1
set osd_esc_tmp_pos = 82
set osd_esc_rpm_pos = 83
set osd_stat_max_spd = OFF
set osd_stat_min_batt = OFF
set osd_stat_bbox = OFF
set osd_stat_endbatt = ON
set osd_stat_bb_no = OFF

# profile
profile 0

set dterm_lowpass_type = PT1
set pid_at_min_throttle = OFF
set anti_gravity_gain = 9000
set i_pitch = 46
set d_pitch = 27
set p_roll = 50
set i_roll = 43
set d_roll = 25
set p_yaw = 75

# profile
profile 1

# profile
profile 2

# restore original profile selection
profile 0

# rateprofile
rateprofile 0

set rc_rate = 120
set rc_rate_yaw = 120

# rateprofile
rateprofile 1

# rateprofile
rateprofile 2

# restore original rateprofile selection
rateprofile 0

# save configuration
save

VoicOfReason, 2017-10-06:
Tarekmaalouf, I tried your suggestions for troubleshooting of turning off everything except telemetry and horizon/angle mode. Problem still exists. It didn't help anything. It still drifts hard to one side after 1 throttle punchout.

tarekmaalouf, 2017-10-06:
Ok, I wanted to check and see ....because I am duplicating this on my end and wanted to start with just the basics....i am about to take this one up and see...

VoicOfReason, 2017-10-07:
Hope you can solve this for us all.

tarekmaalouf, 2017-10-07:
ok..I have been able to somewhat duplicate and correct...
rather than give me a dif reading from CLI please send me a dump reading. You can copy and paste that to an email for me if you prefer...cyclone3dservices@gmail.com as I have multiple boards ready right now.

One of the key factors it seems is the calibration time span between each gyro calibration. It appears to me (and this is from quick testing - more extensive during the next few hours) that the time gap between updates creates a small window where the quad must try to fill in the gap with its own data. I know that my terms are not exact - sorry as my BF lingo is not up to par - but the fact remains that during certain movements, depending on the update frequency, the fc appears to try and compile its own "horizontal leveling" which, if done at the wrong time, seems to cause the fc to fly off level. 

Please understand that this is the testing that I have done thus far but I am diving into it for the next 5 hours (it is midnight almost here) and I try to work until 5 and then sleep for two hours and then start back up again. 

the dif from CLI gives me a small amount of data but the dump will give me a great deal more that I can evaluate from. So, while I continue, and I completely understand if this appears to be taking too long and you may not be interested, but if you are, send me that txt file. I need to compare it and plug in your exact data to these boards and then tune the update to see where the ideal mark would be. 

My guess, and again just a guess based on a the components here, is that the calibration/alignment process is recognizing your punch and also updating in the process which is causing the reading to be skewed. Again, my terminology makes sense in  my head although I have a feeling there are more accurate terms for what I am trying to explain.

In either case, if you can send the file, please do. I would like to review. I feel like I am close...but I will not know for a few more hours perhaps. I hope!

mikeller, 2017-10-07:
@tarekmaalouf:

> You can copy and paste that to an email for me if you prefer...cyclone3dservices@gmail.com as I have multiple boards ready right now.

Can you please avoid conducting support over email?
The reason for this is that GitHub is also meant to serve as an archive of past support cases, with users being able to find solutions to their problems if other users have had the same problem in the past.

tarekmaalouf, 2017-10-07:
My apologies. I am new to this board and did not mean to cause an issue. Of course, please paste here and I will do my best. I will make sure to avoid this mistake in the future.

VoicOfReason, 2017-10-07:
Here's my dump:

# version
# Betaflight / OMNIBUSF7V2 (OB72) 3.2.0 Oct 5 2017 / 14:43:41 (36c91c76f)
MSP API: 1.36

# name
name WPF-Talon

# resources
resource BEEPER 1 D15
resource MOTOR 1 B00
resource MOTOR 2 B01
resource MOTOR 3 E09
resource MOTOR 4 E11
resource MOTOR 5 NONE
resource MOTOR 6 NONE
resource MOTOR 7 NONE
resource MOTOR 8 NONE
resource SERVO 1 NONE
resource SERVO 2 NONE
resource SERVO 3 NONE
resource SERVO 4 NONE
resource SERVO 5 NONE
resource SERVO 6 NONE
resource SERVO 7 NONE
resource SERVO 8 NONE
resource PPM 1 NONE
resource PWM 1 NONE
resource PWM 2 NONE
resource PWM 3 NONE
resource PWM 4 NONE
resource PWM 5 NONE
resource PWM 6 NONE
resource PWM 7 NONE
resource PWM 8 NONE
resource LED_STRIP 1 D12
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 NONE
resource SERIAL_TX 3 B10
resource SERIAL_TX 4 NONE
resource SERIAL_TX 5 NONE
resource SERIAL_TX 6 C06
resource SERIAL_TX 7 NONE
resource SERIAL_TX 8 NONE
resource SERIAL_TX 9 NONE
resource SERIAL_TX 10 NONE
resource SERIAL_TX 11 NONE
resource SERIAL_TX 12 NONE
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 A03
resource SERIAL_RX 3 B11
resource SERIAL_RX 4 NONE
resource SERIAL_RX 5 NONE
resource SERIAL_RX 6 C07
resource SERIAL_RX 7 E07
resource SERIAL_RX 8 NONE
resource SERIAL_RX 9 NONE
resource SERIAL_RX 10 NONE
resource SERIAL_RX 11 NONE
resource SERIAL_RX 12 NONE
resource I2C_SCL 1 NONE
resource I2C_SCL 2 NONE
resource I2C_SCL 3 NONE
resource I2C_SCL 4 NONE
resource I2C_SDA 1 NONE
resource I2C_SDA 2 NONE
resource I2C_SDA 3 NONE
resource I2C_SDA 4 NONE
resource LED 1 E00
[... 566 lines cut ...]
set horizon_tilt_expert_mode = OFF
set gps_pos_p = 15
set gps_pos_i = 0
set gps_pos_d = 0
set gps_posr_p = 34
set gps_posr_i = 14
set gps_posr_d = 53
set gps_nav_p = 25
set gps_nav_i = 33
set gps_nav_d = 83

# rateprofile
rateprofile 0

set rc_rate = 100
set rc_rate_yaw = 100
set rc_expo = 0
set rc_expo_yaw = 0
set thr_mid = 50
set thr_expo = 0
set roll_srate = 75
set pitch_srate = 75
set yaw_srate = 80
set tpa_rate = 10
set tpa_breakpoint = 1650

tarekmaalouf, 2017-10-07:
Reading from a post Boris put up regarding the changes in 3.2 there were some things that I noticed on your config that are off from mine and more importantly seem to be steps that he recommended only after the quad proved to be testing fine...

I have taken your entire line of dump file, written a small script to compare to mine, and noticed that our differences appear to fall in line with what Boris mentioned. In your case, your settings are set to "what he would change if everything was fine" while mine are set to more of a default. 

Here is a link to the article he put out which seems to be in line with some of what I am seeing here....

I have also copy and pasted the settings on mine that differ from yours and many of them appear to be in line with his post as well. Granted we do not have blackbox info from your quad but assuming you are using soft housing and all is well then I would have to see if you were to revert to the "Default" as stated in his article...to see if you are able to get clean flying. 

For example - changing the first line from PT1 to biquad would put you right back to default and the start. Although this is his recommendation to change first if things are flying fine, he does seem to note that this is not the default for an "out of the box" flying experience. (unless I am reading his article completely wrong)

Either way, I think that my settings are more in line with the default and I would ask that you simply change them to match and see how it goes. Cant hurt to try and see if the "default" settings are in place, perhaps it may create a better starting point. 

So, here are the settings from all the lines that I see as different:

Your   Config | My Config
-- | --
set dterm_lowpass_type = PT1 | set dterm_lowpass_type = BIQUAD
set anti_gravity_gain = 5000 | set anti_gravity_gain = 1000
set p_pitch = 63 | set p_pitch = 58
set i_pitch = 60 | set i_pitch = 50
set p_roll = 43 | set p_roll = 40
set i_roll = 50 | set i_roll = 40
set p_yaw = 75 | set p_yaw = 70
set i_yaw = 50 | set i_yaw = 45
set p_level = 90 | set p_level = 50
set i_level = 85 | set i_level = 50
set d_level = 100 | set d_level = 75
set level_limit = 85 | set level_limit = 55

Also, here is a link to that post from Boris:
https://github.com/betaflight/betaflight/wiki/Gyro-&-Dterm-filtering-recommendations

I am not sure if this will help but I think it might be a good start. If it does not work, simply change the values back. Just trying to help out...

VoicOfReason, 2017-10-07:
Absolutely! I'll try those settings out this afternoon and report back. Were you able to duplicate the malfunction?

RipperDrone, 2017-10-07:
@tarekmaalouf - here is my dump instead of diff: 

[Dump Level OFfset Problem after Punches.txt](https://github.com/betaflight/betaflight/files/1366209/Dump.Level.OFfset.Problem.after.Punches.txt)

To your point of defaults: I have started to fly this quad off the defaults which displayed the same level offsets after throttle punches from the very beginning. Only then I started to add ANTI_GRAVITY, even bringing it up to a 9000 level without any success to cure the problem. Maybe you are onto something with the timing of gyro readouts. I'm not so sure about the inflight re-calibration of level that you mentioned. I thought BF is calibrating level only a) before flight, right after initializing OR (inflight, if the GYRO_LEVEL_INFLIGHT feature is set to ON)...

VoicOfReason, 2017-10-07:
By the way, I don't have this problem on a 4" Floss I just built with Hyperlite 2204-3022kv motors. It flies wonderfully and stays level no matter what I do. I'm running 3.2RC6 on it, as well. Settings are virtually identical to my larger build that has leveling issues.

tarekmaalouf, 2017-10-08:
hey guys - I know this is off topic a bit..but I just finished the final design and test for the new racing quad I have to ship out. I need some folks to test a few for me...for 30 days...and in return, just give me feedback ...a lot of feedback...if anyone is interested, and is skilled in FPV, please let me know. Sorry, been working for 3 months on the design and finished everything while I was assisting with this topic as well. I am back to the FC now to see whats up....

VoicOfReason, 2017-10-08:
Ill try it out. I dont Race but have been doing drones for 4 years and own
5 high speed quads I built.

VoicOfReason, 2017-10-12:
Any progress on this ACC issue?

tarekmaalouf, 2017-10-12:
I have not been able to duplicate the issue on my board and I have a feeling that this goes beyond the realm of what I know. I am sorry for not being able to do more but I am going to be communicating with you here to send you out one of my quads for testing. I will continue to try and see if I can duplicate on  my end...

Maxidingue, 2017-10-17:
changing the first line from PT1 to biquad, to work for me, thanks

tarekmaalouf, 2017-10-17:
@Maxidingue that worked for you?

VoicOfReason, 2017-10-17:
What do you mean by "that worked"? Do you mean that duplicated the error or it fixed the error (drift)?

Maxidingue, 2017-10-17:
yes, with air mode on, and pt1, the drone takes off on its own, does not do it on biquad

VoicOfReason, 2017-10-17:
Very interesting. I will test also. Good news. At least disabling PT1 will be a temporary fix. If so, this should give developers direction as to what might be wrong?

VoicOfReason, 2017-11-03:
Tried disabling PT1 and turning back to BIQUAD. Didn't help. Still terrible drift problems.