betaflight/betaflight #4269: Kakute F4 OSD rolling flicker

MylesShannon, 2017-10-03:
I'm getting a visual bug with the OSD of the Kakute F4 AIO flight controller when "cpu_overclock" is set to "ON". Here is a [DVR video](https://youtu.be/HXmSvdUkhIA) showing what I am seeing. This is a fresh flash of Betaflight 3.2 RC6. I've seen this same issue with previous 3.2 RCx releases.

The only solution I've found is using [this firmware](betaflight_3.2.0_KAKUTEF4-max_half_clock.hex.zip) supplied by @jflyper in this now closed issue, #4144

nathantsoi, 2017-10-03:
you need to set `max7456_clock`, see https://github.com/betaflight/betaflight/pull/4150/files#diff-3df161829f53e111012cc1907754987bR756

closed 2017-10-03 by nathantsoi

jflyper, 2017-10-03:
@MylesShannon 

- The half-clock firmware provided in #4114 works,
- Full-erase RC6 doesn't.

Is this your case?

MylesShannon, 2017-10-03:
@nathantsoi
No problem. Thanks for the info. I'll give that a try and report back here tomorrow.

@jflyper
That's right.

jflyper, 2017-10-03:
@MylesShannon 
Overlay chip driver in RC6 is designed to detect the original MAX7456 chip, but it may not be working.

Can you try the `set max7456_clock = HALF` first to find out this option works?

MylesShannon, 2017-10-07:
Sorry for the delay.

I got around to testing ```set max7456_clock = HALF``` and it did not solve my issue. I double checked the setting with a ```get max7456_clock``` too.  Do you have any other ideas?

Thanks for the help.

jflyper, 2017-10-10:
@MylesShannon 
We need some debugging.
Can you flash this hex

(1) Flash with this firmware
[betaflight_3.2.0_KAKUTEF4-4224-0.hex.zip](https://github.com/betaflight/betaflight/files/1371551/betaflight_3.2.0_KAKUTEF4-4224-0.hex.zip)

(2) `set debug_mode = MAX7456_SPICLOCK`

(3) After rebooting, record the values on debug variables 1, 2 & 3 for over clocking on and off.

krankenreif, 2018-01-24:
I tested this with the my current build. The values are with max7456_clock = half (in order of Debug 1, 2 and 3) 0, 16, 0 and with max7456_clock set to full they are 0, 8, 0. My OSD is completely invisible with cpu_overclock = on and the values do not change by setting cpu_overclock to on or off.

max7456_clock = half and cpu_overclock = on: 0, 16, 0
max7456_clock = half and cpu_overclock = off: 0, 16, 0
max7456_clock = full and cpu_overclock = on: 0, 8, 0
max7456_clock = full and cpu_overclock = off: 0, 8, 0

jflyper, 2018-01-25:
@krankenreif 
Is there a firmware you had a visible OSD? If so, which version was it?

krankenreif, 2018-01-25:
I tried out different checkouts, even one before you reworked the SPI stuff but my OSD never shows up when setting cpu_overclock = on. Any specific version you want me to try?

jflyper, 2018-01-25:
@krankenreif 
Try one of these (that matches with your board)
[betaflight_3.3.0_KAKUTEF4-slowmax-0.hex.zip](https://github.com/betaflight/betaflight/files/1664586/betaflight_3.3.0_KAKUTEF4-slowmax-0.hex.zip)
[betaflight_3.3.0_KAKUTEF4V2-slowmax-0.hex.zip](https://github.com/betaflight/betaflight/files/1664587/betaflight_3.3.0_KAKUTEF4V2-slowmax-0.hex.zip)
Try setting `max7456_clock` to `slow`.

krankenreif, 2018-01-26:
When seting debug_mode = max7456_signal I still get 0, 0, 0, 0. You probably multiplied by 4, not by 2. I tried that aswell yesterday...

jflyper, 2018-01-26:
`max7456_signal` is for debugging video format. You should use `max7456_spiclock`.

krankenreif, 2018-01-26:
Yeah, I know. I also tried max7456_spiclock and it shows the expected behaviour (1, 0, 32). The thing is that with max7456_signal it shows up as non zero when the OSD is working. That's why I was referencing it.

jflyper, 2018-01-26:
There's nothing more I can do in terms of SPI clock for MAX, except halving it again, which I think will have no effect. There must be something else that is causing it to fail on OC.

krankenreif, 2018-01-27:
I thought that aswell. Maybe I'll get around monitoring the SPI clock with an oscilloscope today... That could give some insights.

krankenreif, 2018-02-03:
Adding this line to my target.h worked for me:

#define MAX7456_SPI_CLK         (SPI_CLOCK_STANDARD * 2)

jflyper, 2018-02-04:
That's so weird. I'm interested in debug outputs.

krankenreif, 2018-02-04:
Max7456_clock: 1, 0, 32, 0
Max7456_signal: 0, 4, 0, 13

nibrocd, 2018-02-07:
Any update? I have the same issue and would like a fix. Thanks

NikoN-FPV, 2018-02-15:
i have the same problem on my kakute f4 v2, but nothing works.. tested firmware from above then osd is completely gone, and the black flickers stay on all different settings.. slow half full default.. if OC is off then perfect video feed but without osd and i cant run on 32k/16k.. with 3.3.0 RC1 and OC on, the osd is here again slighlty, but also the black flickers.. set max7456_clock to half, default, full, but no changes.. but in rc1 with OC off i have perfect video and osd!! but without oc i cant run on 32k.. :/ someone any ideas?

krankenreif, 2018-02-18:
I built 3.3 RC1 with my proposed change 

> #define MAX7456_SPI_CLK (SPI_CLOCK_STANDARD * 2)

Please leave feedback 👍 

[KAKUTEF4.zip](https://github.com/betaflight/betaflight/files/1735257/KAKUTEF4.zip)

nibrocd, 2018-02-19:
That works perfect, thanks. I get a build with 3.3 RC2? And can this be added to the offical code?

NikoN-FPV, 2018-02-19:
@nibrocd you dont need it in RC2 .. in RC2 you can manually set the cpu_overclock to 192/216/240mhz.. i set it to 216mhz on my v2 and works perfect :) 32k/16k on 35% cpu-load ;) thanks guys :D

jflyper, 2018-02-19:
@krankenreif It's not that simple...

krankenreif, 2018-02-19:
@jflyper Yeah, I know. We should not change the target.h but I find it interesting that this insertion seems to half the clock whereas setting max7456SpiClock is not working correctly...
Maybe we could remove the `#ifdefs` in https://github.com/betaflight/betaflight/blob/552ca4c33b70da0b3f39699544a6150b98f1056f/src/main/drivers/max7456.c#L174 and just leave
`#define ENABLE_MAX7456        {spiSetDivisor(MAX7456_SPI_INSTANCE, max7456SpiClock);IOLo(max7456CsPin);}`

I'll try that out later. What do you think?

jflyper, 2018-02-19:
Ah... I'm starting to see the problem...

jflyper, 2018-02-19:
Those of you having problem with osd, can you see if this makes a difference?

```
resource flash_cs none
save
```

nibrocd, 2018-02-19:
You mean just in the cli? ill give that a shot.

jflyper, 2018-02-20:
Doh...

jflyper, 2018-02-22:
@krankenreif
> Adding this line to my target.h worked for me:
>
>#define MAX7456_SPI_CLK (SPI_CLOCK_STANDARD * 2)

I think this is the only practical solution at the moment. I believe a plain `SPI_CLOCK_STANDARD` will also work.

The reason behind this observation is that current code mixes up two things, (1) a feature to allow multiple devices with different clock requirements on the same bus, that is enabled by defining `MAX7456_SPI_CLK` in `target.h`, and (2) a facility to detect slower chip and reduce the clock when overclocked, which also rely on `MAX7456_SPI_CLK` being defined (thus it does not work if `MAX7456_SPI_CLK` is not defined.

Adding the `MAX7456_SPI_CLK` to `target.h` is justified by following reasons.

1. It is required when multiple devices sharing the same bus have different clock requirements.

2. The two features can not be separated with a small change and need regression test involving multiple targets.

3. The co-existence problem will be addressed in a more generic way in 3.4 (such as #4902).

Please test whether the plain `SPI_CLOCK_STANDARD` work. I will file a PR based upon the test result.

krankenreif, 2018-02-22:
@jflyper I'll try that and report back asap

krankenreif, 2018-02-23:
It worked @jflyper 
Attached is the up to date HEX with the fix (Merge pull request #5269 from mikeller/fixed_fport_efficiency). Please test 👍 
[betaflight_3.3.0_KAKUTEF4.hex.zip](https://github.com/betaflight/betaflight/files/1751374/betaflight_3.3.0_KAKUTEF4.hex.zip)

jflyper, 2018-02-23:
So, single `SPI_CLOCK_STANDARD` worked?

krankenreif, 2018-02-23:
Yes, it worked for me.