betaflight/betaflight #4271: BF 3.2 RC6 - Profiles are not working properly

Rickcau, 2017-10-03:
## If you believe there is an issue with the firmware itself please follow these steps:
Problem Description:
I have setup 3 different rate profiles for Profile 1 for my racing builds.  I also like to use Profile 2 for higher rates & d setpoint, just in case I want to do acro and feel I need faster response.  If I select profile 2 and set my settings and save them.  When I reconnect, I now see profile 1 rate 1 has the same settings I just made for profile 2 rate 1.  This is duplicated across all rate settings.  It appears the bug is referencing the same array structure that Profile 1 is referencing, it should be an easy fix.

How to reproduce:
1. Set 3 different rate settings for profile 1 and save them.
2. Set 3 different rate settings for profile 2 and save them.
3. disconnect and reconnect then take a look at the rate settings for profile 1, then have changed to the rate settings that you just set in profile 2.

Clearly it appears to be a bug that is referencing the same array structure as the profile 1.

chime13, 2017-10-04:
@Rickcau, 
As was pointed out to me:
FCs will now store 3 profiles and only 3 global rate profiles.
Yeah, I too had different rates for each of the different profiles. Otherwise, everything is working fine for me.

mikeller, 2017-10-04:
@Rickcau: As @chime13 is pointing out, profiles and rateprofiles have been simplified in 3.2, and they are now completely independent. You now get 3 profiles and 3 rateprofiles, and any profile can be used in conjunction with any rateprofile.

closed 2017-10-04 by mikeller

Rickcau, 2017-10-08:
I hear you on the simplification of things.  But, many folks setup 3 separate rate profiles for racing only and as you know racing rates are much, much lower than acro rates.   I run and avg of 975 dps for acro and 325dps for my racing rates.  Hence, having only 3 rate profiles is really limiting.  I would encourage you to use 3 arrays so you can have 3 rate profiles for each PID profile.  Currently as it stands I really cannot adequately fly acro on any of my race quads as I have to go into BF and change them, them change them back.   I hope you can understand how time consuming this is.  

I ask that you please triage this with the other devs and reconsider adding it back.