betaflight/betaflight #4272: Unstable crossfire connection with telemetry ON

adamtfc87, 2017-10-03:
# If your issue looks like a hardware fault or a configuration problem please don't raise an issue here.

## Please consider using other user support options such as asking the manufacturer of the hardware you are using, [RCGroups](https://rcgroups.com/forums/showthread.php?t=2464844), [Slack](https://slack.betaflight.tech) or other user support forums & groups (e.g. facebook).

## If you believe there is an issue with the firmware itself please follow these steps:
1.) Using Crossfire 2.0.8 micro tx and micro rx v2, signal fluctuates quite a lot with telemetry turned on in betaflight and on the receiver.

codecae, 2017-10-03:
I worked with @adamtfc87  to troubleshoot this issue and have found that a recent change in the rx code may be inhibiting telemetry transport.

Reverting the following change seemed to correct the issue:
https://github.com/betaflight/betaflight/commit/287a45659d88f7bc5f97cb265eb47a5d811cc61e#diff-d00dc7f5c196f87a1c750b70bd40795eL236

codecae, 2017-10-03:
Reverting the following change smoothed out the dips in LQ (packet success ratio) in telemetry:

https://github.com/betaflight/betaflight/commit/860ec23fbe1916d96e458edca498af26cc304fc2#diff-d00dc7f5c196f87a1c750b70bd40795eL236

mikeller, 2017-10-04:
There is something odd going on here. https://github.com/betaflight/betaflight/commit/860ec23fbe1916d96e458edca498af26cc304fc2#diff-d00dc7f5c196f87a1c750b70bd40795eL236 should not make a difference unless `serialrx_halfduplex` is set to `on`.

codecae, 2017-10-04:
Appears to be a false positive result.  Something peculiar is going on and I can't quite put my finger on it.  When applying @adamtfc87's diff, it appears to trigger something that doesn't occur when manually configuring.

codecae, 2017-10-04:
I found that simply disabling the accelerometer is causing dips of 10-12% LQ.

codecae, 2017-10-04:
Created PR #4279 for this issue.

mikeller, 2017-10-05:
@codecae: Not sure I'd call this a workaround. Not sending attitude data if we don't have attitude data seems to be the right thing to do.

codecae, 2017-10-11:
After researching this issue deeper it has been indicated that the problem existed on the crossfire side.  The firmware is now available in public beta as v2.11 ... Please update your crossfire firmware to find resolution to this issue.

closed 2017-10-11 by DieHertz