betaflight/betaflight #4273: Does anyone have a new document to build the betaflight source code in Eclipse C/C++ ?

helpmelearnplease, 2017-10-03:
Hello guys,
If you have a new documentation for Building the source code for BetaFlight in Eclipse IDE for C/C++ please help me with it.
In spite of following all the steps in https://github.com/martinbudden/betaflight/blob/master/docs/development/Building%20in%20Eclipse.md
and also form https://github.com/cleanflight/cleanflight/blob/master/docs/development/Building%20in%20Eclipse.md#checkout-cleanflight

i am not able to succeed with the process.
I am getting "make:** empty variable name.stop." Error.

AndersHoglund, 2017-10-03:
A project like this is not the ideal thing to start learing crossplattform embedded software design in Eclipse. And building such things are not a strong point of Eclipse, especially not with 100++ different targets. Not sure anyone here have the time to guide you through it all. Get experience on Eclipse on some simpler project. 
If this is only about building binaries, I think you should forget Eclipse. Instead:
Open a Linux terminal window shell, a real or virtual via Vagrant, Virtualbox, CygWIN, Docker etc etc. 
cd to the betaflight project and `make arm_sdk_install` then `make <some_target>`,  Or only `make` to build them all.  `make help` is also your friend.

mikeller, 2017-10-04:
@helpmelearnplease: @AndersHoglund is right, if you are trying to start out by building in Eclipse, you are trying to run before you have learned to walk. Follow the steps in https://github.com/martinbudden/betaflight/blob/master/docs/development/Building%20in%20Windows.md to get a (non-Eclipse) Windows development environment set up. Only when you've got this done, and can build in Windows, move on to Eclipse.

helpmelearnplease, 2017-10-04:
@AndersHoglund @mikeller Thank you for your suggestions, But I am not able to download the packages from git (Devel/git-completion) after installing CyGwin, why is it so?
Is there any particular version of the software you would recommend?

mikeller, 2017-10-04:
Don't worry about Devel/git-completion, it is marked as optional.

helpmelearnplease, 2017-10-04:
Okay @mikeller , Will try it and will post the update in the improvements

conkerkh, 2017-10-08:
I’ve set up eclipse in 2 mins it’s pretty straightforward. But the problem is it’s not indexing files properly as certain defines are target related. Therefore the best option is to import bf and then create a new make file in eclipse.

AndersHoglund, 2017-10-08:
The betaflight Makefile (and included sub-makefiles) are essential for correct build configurations. Trying to duplicate this is not such a good idea. Hard to get that right, and keeping it in sync. And even if you do succeed in this duplication, you can only do it for ONE particular target/configuration at a time and get the sources indexed for that one single target. When coding and designing, you must have a broader view than just one single target. Debugging with eclipse is another thing, then it is natural to have only one single target configured. I use Eclipse with J-link and Seeger that way.

This is the major flaw of eclipse, depending on build config conditional items but not importing/parsing the definitions in a makefile based build system.  Eclipse is good for single target projects where everything is defined inside eclipse. But we do not want to be tied to one particular IDE, and we do want to support multiple targets.

conkerkh, 2017-10-08:
You can build in Eclipse with “external male command” which basically allows to build any target you want without setting up much stuff. The only problem is that eclipse is blind and won’t index properly. 

FYI If you use JLink, Ozone is much better for debugging purposes than Eclipse.

AndersHoglund, 2017-10-08:
Yeah yeah, sure you can build using make. But that is so much easier without eclipse, you only need a shell. As you say, eclipse is blind. Never tried Ozone, interesting. Maybe ditch eclipse totally...
For quick edits and fixes I never launch eclipse anyhow, emacs, grep, find etc etc are all the friends you need.

conkerkh, 2017-10-08:
Yeah, pretty much you are right regarding the building process in Eclipse. Try Ozone you gonna love it, it truly allows to use the potential of JLink.

mikeller, 2017-10-08:
@conkerkh: I've had limited support to get Eclipse to index properly for different targets by using a build configuration for each target, and then essentially replicating the target's relevant defines from `target.h` into the build configuration's defines.

helpmelearnplease, 2017-10-12:
@AndersHoglund and @conkerkh or anyone in this thread, do you suggest me to build the code in Ubuntu? Linux?
is there any steps or documentation available for the same?

AndersHoglund, 2017-10-12:
On Linux (real or virtual):
Get the sources, install sdk and build one target like this:
```
git clone https://github.com/betaflight/betaflight.git
cd betaflight
make arm_sdk_install
make BETAFLIGHTF3
```

Try a few other make variants:
```
make help
make test
```
You may find some components missing or other issues. Install and fix as needed. As this depends a lot on your base environment it is hard to tell what and howto fix. A few of them are listed here: https://github.com/betaflight/betaflight/wiki/Known-Build-Environment-Problems 

Not sure we are the best internet school of programming and building. I am certainly not. You might want to get experience on some smaller project.

helpmelearnplease, 2017-10-13:
thank you @AndersHoglund , I will install Ubuntu and try running it,
I have another query, after building the code in Linux environment, will i be able to make some changes to the betaflight/Cleanflight code?
I know I am asking too many questions but I am in that kind of situation that I want to access this code for my project.
I Appreciate that you are helping me out with this

AndersHoglund, 2017-10-13:
OK, but if you do not already have a Ubuntu maschine, try instead installing Vagrant on windows. Read the Vagranfile in the beatflight source code root dir. You need to get an initial copy of the source tree, or only the Vagrantfile. Or use docker, virtualbox or any other VM with linux installed. Even a Linux-like env like CygWin will do, even though that may incure some file format issues. ( Classic WinDOS problems.)
Of cource anything can be changed, only needs the sources and an editor. But since you ask; I have no idea if YOU can do it or not.

mikeller, 2017-10-13:
@helpmelearnplease: Or you could use https://hub.docker.com/r/betaflight/betaflight-build/ .

Jahnkeanater, 2017-10-16:
It seems like when I make a target it only updates the first time. When I change the code than build it again it doesn't update. I have tried deleting the obj folder. Do I need to run a command to clean up before I rebuild it again? Make clean all didn't seem to do anything.

mikeller, 2017-10-16:
@Jahnkeanater: That's odd. `make` normally picks up changed files by comparing the timestamp of the input (*.c) file to the one of the output (*.o) file, and recompiling if the input is newer than the output. What operating system and file system (local / remote mount) are you working with?

Also, the command to clean the artefacts for one target is `make <target name>_clean`.

Jahnkeanater, 2017-10-16:
@mikeller linux mint, local file system? Maybe I needed to clear the specific target I built instead of make clean_all. Or maybe when I edited the .c file I messed up a file's permissions?

It should just recompile when I make target right.

should the beta flight folder be root permissions?

mikeller, 2017-10-16:
Surprised that you have problem on linux with a local file system (NFS on the other hand is somewhat famous for screwing with make). As long as you are able to see the file as the user that runs the build command, you should be fine.

> It should just recompile when I make target right.

Yes.

> should the beta flight folder be root permissions?

It should be owned by and accessible to the user that is running make. And this should preferrably not be root.

Jahnkeanater, 2017-10-16:
I cloned the repo with sudo. I didn't run make with sudo. I don't think make had permission to delete old files. Hopefully make works fine now that the betaflight folder isn't root.

wind0r, 2017-10-16:
Never ever clone a repo with sudo!

mikeller, 2017-10-16:
> Never ever clone any repo with sudo!

Never do anything with `sudo` that will work without `sudo`!
(i.e. try first without, if you get 'you don't have permission', reconsider, and if you are still convinced what you want to do is the right thing to do, do it with sudo.)

Jahnkeanater, 2017-10-16:
I'm going to assume cloning with sudo is a security concern. :)

Since we're on the topic of Linux. How do I fix FC drivers? Betaflight says the board is in DFU mode every time I plug it in. I always just use a windows PC to flash the firmware.

mikeller, 2017-10-16:
@Jahnkeanater:

Yes it is - mostly. Even with the security concerns set aside, a lot of processes that create files will create them with the wrong permissions when run with sudo, causing other processes trying to use these files later on to fail - see your problems with building.

I suspect you haven't worked through https://github.com/betaflight/betaflight/wiki/Installing-Betaflight#platform-specific-linux then. The board being put into DFU mode is typically caused by ModemManager running on the host, and trying to grab every `/dev/ttyACM<x>` device that is recognised...

Jahnkeanater, 2017-10-16:
@mikeller 
I have tried that. I tried it again. It doesn't go into DFU mode now. It just fails to connect. I think I edited the file wrong? I'll figure it out one of these days. :)

# DFU (Internal bootloader for STM32 MCUs)
ACTION=="add" , SUBSYSTEM=="usb" , ATTRS{idVendor}=="0483" , ATTRS{idProduct}=="5740" , MODE="0664" , GROUP="plugdev"

mikeller, 2017-10-16:
@Jahnkeanater: It's not that part, it's the part about killing ModemManager (and making sure it stays dead).

closed 2018-04-21 by DieHertz