betaflight/betaflight #4276: CL_racing F4 SDcard fatal error on BF3.2 RC5, RC6. works OK on BF 3.17

peteoz, 2017-10-04:
I can't get BB logging to work on CL_racing F4 board, it fails to record data every time when trying on BF3.2, but works fine on BF3.17.

tried 3 different SD cards that are known to be good because they work on my other quads with different FCs.

the whole behaviour in BF3.2 is weird, for example when in configurator and on BB page, it comes initially as error, when I switch to CLI and instantly come back to configurator then it reads SD card "ready", but regardless then it fails to record any data.

posted the question on RCgroups CL-RacingF4 thread 10 days ago asking whether anybody got BB working and received no positive confirmation to date.

jflyper, 2017-10-04:
@bnn1044 Do you have any info on this?

bnn1044, 2017-10-04:
@jflyper @peteoz 
just test RC6. works. no problem at all.

bnn1044, 2017-10-04:
@peteoz do you have the configurator upgraded?

peteoz, 2017-10-04:
@bnn1044 I'm running 3.2.1, but it's not just the configurator Fatal Error. at the end of the flight when OSD stats come up, I always get "fault" for the BB and no log file is ever recorded.

in CLI when I run status I get "Filesystem: Fatal - no FAT MBR partitions", then I come back from CLI to configurator and under BB it reads SDcard "Ready", I close down configurator and re-open it and it comes up stating "Fatal Error".
and yet those same SD cards work fine when running version 3.17, or when running same SD cards on other boards (tested with 2 different Kiwi F4 boards and LuxV2), those other FCs work fine with BF3.2

bnn1044, 2017-10-04:
@peteoz did you try to format the card either using the configurator or computer?

peteoz, 2017-10-05:
@bnn1044 firstly I formatted my SD cards on my Mac, choosing "MS DOS Fat" formatting option. and because it still wasn't working I then also downloaded the SDcard formatter tool from sdcard.com and formatted again, still it doesn't work.

I'm using these cheap 2GB SD cards : https://www.amazon.com/gp/product/B000VOU91U/
I bought 10 of them and I'm using them successfully on my other quads with different FCs.

I wonder if I should try to buy some modern SD card with much faster read/write access speed, but they usually come in sizes 32GB/64GB which is kind of waste.

What SD card did you use in your test ?

I also note when the SD card is inserted in SDcard reader then I quite struggle to connect to BF configurator, very often it freezes when I insert the USB cable, I have to do it many times before I connect. However I don't have this issue when the SD card is removed from the FC and BB is disabled. So I wonder if it is just a case of the SD card reader on the FC unable to handle slow read/write speed of these older SD cards.

bnn1044, 2017-10-05:
I use one of my 1gb cheap sd card experiencing the same as you. The FC won’t boot up. The green light constantly on. But when I change to better sd card problem go away.

peteoz, 2017-10-05:
@bnn1044 I tested the SDcard reader today with 32GB Sandisk ExtremePro (this one is Class 10, U3) and it worked. 

You can close this issue I guess, but it's worth noting that the older/slower 2GB card worked on BF3.17, I suspect something has changed with BF3.2 that now makes olded SD cards unable to handle required read/write speed and/or the new compression algorithm.

mikeller, 2017-10-06:
@peteoz: The problem in 3.2 might be caused by the default for `sdcard_dma` being switched from `on` to `off`, because it was causing conflicts in many cases. If you feel like it, you can see if changing this back to `on` will fix it.

peteoz, 2017-10-06:
@mikeller I see, didn't know that one. anyway, I tried that now setting it to "on" and the I still get the fatal error with the older 2GB SD card.
the faster ExtremePro SD card works ok for me so I just use that one.

mikeller, 2017-10-06:
Thanks for testing. SD card / SD card DMA support is still not as stable as it could be, so any additional insights are appreciated.

eutxi, 2017-10-17:
The same problem with my SPRACINGF4EVO and another one from a friend.
Many times it does not start and stays with the red LED on permanently. 
I have to connect and disconnect the battery many times to make it work and this only happens with the sd connected, if I remove everything ok.
With 3.1.7 all ok

closed 2018-04-21 by DieHertz