betaflight/betaflight #4278: LED profiles (switchable via OSD)

Elbau, 2017-10-04:
I'd love to see programmable profiles for LED modes and an OSD interface for selecting them.

For instance, a race mode that switches to an assigned pilot color, a beacon mode that flashes white, or a status mode with battery, RSSI, warnings.

Typically I fly FPV and only use my lights as locating beacons after a crash -- blinking white seems to provide the best visibility. 

When flying with others I like to have my lights on solid color mode so I show up well in other pilots' cameras. It's fun to use this with a potentiometer on your transmitter to control the color and assign yourself a unique pilot color.

It would be nice to change between these use cases easily, instead of having to re-program my LED strip for each occasion. Also, instead of dedicating a transmitter channel, LED profiles and color selection could be achieved through an OSD menu. 

The interface should be trivial on either end.
The configurator would have profile pulldowns similar to the PIDs and Rates tab.
The OSD could have a top-level LED menu that would allow switching between 3 programmable profiles as well as a simple 'race' mode that sets all wired LEDs the selected color.

This would allow easy on-the-fly assignment of pilot colors via OSD without precluding other  #lighting functions. LED profile changes could also be handled via the adjustments functions (identical to changing rate profiles)-- for those that have spare switches and channels on their transmitters.

lazd, 2017-10-16:
I'm gonna need this on a switch so I can [flip the lights and pull other quads over](https://youtu.be/KG5jbo4tNN4).

BradB111, 2018-01-20:
I would like to see this or even a list of say 8 custom colours to choose from on the osd, I have just installed LEDs on my quad and I am surprised it is not an option

arierep60, 2018-07-17:
I really want this too! :D

cybrsqrl, 2018-11-06:
I also want this. Especially useful on my tiny whoop to switch between racing and turning on my headlights for night flight. Would like to have a couple profiles on a switch. Would also like to turn the LEDs off but leave the battery indicator led on. The current switch option is all or nothing in modes setup.

pkruger, 2018-12-29:
PR: https://github.com/betaflight/betaflight/pull/7303

closed 2019-01-19 by mikeller