betaflight/betaflight #4288: BF and LUX4OSD

tarekmaalouf, 2017-10-05:
The issue appears to be with the new firmware used on the Lumenier F4AIO which requires the LUXF4OSD 3.2.0 version has an issue with the calibration of the accelerometer. The issue does not seem to be present when using the optional 3.1.* VGOODDHF4 Firmware although this is not recommended from Lumenier if running BF 3.2.*. 

I have 100 of these to custom make for a racing group and I wanted to know if there is something on my end that may be the cause of the problem. Again, I do not see the issue with the HGLRCF4 on the same frame nor do I see the same issue when using the VG*** firmware. Thought I would see if you could recommend something - perhaps changing the refresh etc...

Thank you in advance...
[image]

[LumAIO Config.txt](https://github.com/betaflight/betaflight/files/1360569/LumAIO.Config.txt)

tarekmaalouf, 2017-10-05:
Well, after uploading the VD** firmware and then switching back, I was able to correct the issue. I will see if it happens again with the other builds and keep you posted. With 100 of these to build and test, I better get back to work!

tarekmaalouf, 2017-10-05:
Sorry, it took all of 10 seconds the drone went back to off level. I did not have to move it or do anything. By the time I was done typing this last post it had gone from working to not working. Any input would be appreciated.

tarekmaalouf, 2017-10-05:
Changed the PID Loop to 8kHz and the calibrations seems to be holding. I will update as I see changes. 

[image]
[image]

capjuancho89, 2017-10-11:
do you have voltage reading on the fc?

jflyper, 2017-10-11:
@tarekmaalouf 
Do all the boards behaves the same way?

capjuancho89, 2017-10-11:
Well i have 3 boards f4 lumenier an the target is luxf4osd and con the three of them have the same problem no voltage :(

tarekmaalouf, 2017-10-11:
Sorry for my delay..tending to some things here..
I have exactly 100 boards in stock...I can randomly pull some now...the ones I had issue with may have been due to the firmware but I noticed last night that there was now a lux4 available on BF rather than through the updates page..that has worked on my recent build..but I also made some configuration changes...

tarekmaalouf, 2017-10-11:
I wish I knew how to get in and change some of the code from the backend and I would tamper with it as well. I put about 18 hours a day in this industry and coding is the final step for me..ughh...

jflyper, 2017-10-11:
@capjuancho89
Your problem is totally different.
Can you try
1. Select onboard ads as source
2. Save on battery tab
3. Move to ports tab (whatever with save & reboot)
4. Save & reboot

mikeller, 2017-10-11:
@jflyper: Are 3. and 4. actually required steps? If so, we should change the button in the battery tab to be 'Save & Reboot`.

tarekmaalouf, 2017-10-11:
@mikeller I dont remember having to do that many steps..except on the ports for the telemetry to pass to the receiver.

capjuancho89, 2017-10-11:
@jflyper thanks for the reply! But i already did that all ThE steps reflashed several times and nothing

jflyper, 2017-10-11:
@mikeller I just wanted to make sure...

jflyper, 2017-10-12:
So, what's the state of the issue now? I'm a bit confused.

jflyper, 2017-10-23:
I'm closing this for now.

closed 2017-10-23 by jflyper