betaflight/betaflight #4295: BF3.2 RC6 Latest Built - DShot Beeper Command (Aux Channel) Not working

Ibanous, 2017-10-07:
I seem to be having the same issue as in #4258 which is closed. However I've tried setting beeper_dshot_beacon_tone = 1-5 with no fix to the issue. I do have beeper set to an Aux channel and verified its working. I'm running Dshot600 and I get motor tones on startup, in addition anti-turtle mode is working so I know the dshot commands are functioning. 

I'm running a Racerstar tattooF4S (AIO FC and ESC) with a target of LUXF4OSD which only has one version available (3.2.0-RC6) so I can't roll back to test RC5. 

Diff:

# diff

# version
# Betaflight / LUXF4OSD (LUX4) 3.2.0 Sep 28 2017 / 16:32:34 (ceb6f70) MSP API: 1.36

# name
name Ibby Lightning

# resources

# mixer

# servo

# servo mix

# feature
feature RSSI_ADC
feature ANTI_GRAVITY
feature DYNAMIC_FILTER

# beeper

# map
map TAER1234

# serial
serial 0 64 115200 57600 0 115200
serial 5 8192 115200 57600 0 115200

# led
led 0 0,15::ATI:0
led 1 1,15::ATI:0
led 2 2,15::ATI:0
led 3 3,15::ATI:0
led 4 4,15::ATI:0
led 5 5,15::ATI:0

# color

# mode_color

# aux
aux 0 0 1 1700 2100
aux 1 1 0 900 1200
aux 2 13 2 1400 2100
aux 3 26 0 1700 2100
aux 4 28 0 1300 2100
aux 5 35 3 1300 2100

# adjrange

# rxrange

# vtx

# rxfail
rxfail 4 s 1000

# master
set baro_hardware = NONE
set baro_noise_lpf = 0
set baro_cf_vel = 0
set baro_cf_alt = 0
set min_check = 1100
set rssi_scale = 30
set serialrx_provider = SBUS
set motor_pwm_protocol = DSHOT600
set failsafe_delay = 10
set failsafe_off_delay = 20
set failsafe_throttle = 1100
set failsafe_procedure = AUTO-LAND
set beeper_dshot_beacon_tone = 5
set servo_lowpass_hz = 400
set gps_provider = UBLOX
set pid_process_denom = 2
set osd_units = IMPERIAL
set osd_rssi_alarm = 30
set osd_cap_alarm = 1600
set osd_tim2 = 1537
set osd_vbat_pos = 472
set osd_rssi_pos = 2107
set osd_tim_1_pos = 56
set osd_tim_2_pos = 2136
set osd_flymode_pos = 364
set osd_throttle_pos = 129
set osd_vtx_channel_pos = 2081
set osd_crosshairs = 0
set osd_ah_sbar = 200
set osd_ah_pos = 200
set osd_current_pos = 2488
set osd_mah_drawn_pos = 111
set osd_craft_name_pos = 2472
set osd_gps_speed_pos = 391
set osd_gps_lon_pos = 82
set osd_gps_lat_pos = 65
set osd_gps_sats_pos = 386
set osd_home_dir_pos = 302
set osd_home_dist_pos = 303
set osd_compass_bar_pos = 266
set osd_altitude_pos = 161
set osd_pid_roll_pos = 322
set osd_pid_pitch_pos = 354
set osd_pid_yaw_pos = 386
set osd_debug_pos = 1
set osd_power_pos = 47
set osd_pidrate_profile_pos = 418
set osd_warnings_pos = 329
set osd_avg_cell_voltage_pos = 2520
set osd_pit_ang_pos = 257
set osd_rol_ang_pos = 289
set osd_battery_usage_pos = 2505
set osd_disarmed_pos = 138
set osd_nheading_pos = 311
set osd_nvario_pos = 279
set osd_esc_tmp_pos = 82
set osd_esc_rpm_pos = 83
set osd_stat_max_spd = OFF
set osd_stat_endbatt = ON

# profile
profile 1

# rateprofile
rateprofile 0

MylesShannon, 2017-10-07:
I'm having the same issue. The stable release of 3.2 doesn't have working DSHOT buzzer commands. I'm using DYS XSD30a ESCs, DHOST 600, and BLHeli_S v16.63 firmware.

mikeller, 2017-10-07:
@MylesShannon: Did you do `set beeper_dshot_beacon_tone = 4` (or any number `1` to `5`)?

Rreaper86, 2017-10-08:
I am having the same issue.

Beeper (Beacon) is not working.
Also when i plug in my Battery i get the esc startup tones TWICE!

Does anyone know What that could be?

Im using : 

Betaflight F3 FlightController.
Cobra BLheli_S 35A With 16.63 DShot Command with dshot600.

Jahnkeanater, 2017-10-09:
I can confirm turtle mode working with 3.2 stable and blheli 16.63. Physical beeper is working but motor beeper is not. FC is a Revo. Using an unknown 4-in-1 esc.
Is it as simple as a CLI command?

Jahnkeanater, 2017-10-09:
Can we get a switch at the bottom of the configuration tab under the beeper settings that enables motor beeper. I can see this helping a lot of people out. Just a switch that changes the beeper_dshot_beacon_tone from 0 to 1. Then the user still has the option to change the tone under the CLI.

MylesShannon, 2017-10-10:
@mikeller
Thanks for the reminder. This did it for me.

serpis, 2017-11-07:
@Ibanous make sure you save after setting the tone. This is what I forgot... :)

aberg12012, 2017-12-04:
@Ibanous I have the Lumineer version of that flight controller (Alpha AIO) and discovered that mine works only if I don't max out the volume slider in BlHeli suite.  Setting it to 255 for me makes it not work for some reason.  Setting it to 240... beeps like a charm.