betaflight/betaflight #4298: Potential roll gyro issue on Omnibus?

Arakon, 2017-10-07:
Latest BF, both on Omnibus F3 at 4k/4k and Omnibus F4 8k/8k.
Motors and FC softmounted. Added caps to no avail.
Look at the gyro pre-notch trace. It's going completely crazy.. this is just hovering.
Compare to ACC trace.. smooth as a baby's bottom.
Biggest thing, this happens ONLY on roll.. yaw and pitch are completely unimpressed, so it can't possibly be vibration.
On this copter, literally everything was replaced already. Motors, FC, ESC, props, battery, frame etc. None makes any difference at all. Dropped D-Term greatly (as far as 12), this slightly reduces dterm magnitude, but it's still going nuts.

I believe I excluded every possible hardware issue.

# diff

# version
# Betaflight / OMNIBUSF4 (OBF4) 3.2.0 Sep 30 2017 / 06:35:03 (ceb6f70) MSP API: 1.36

# name
name Shrieker

# resources

# mixer

# servo

# servo mix

# feature
feature LED_STRIP
feature ANTI_GRAVITY
feature DYNAMIC_FILTER

# beeper
beeper -ON_USB

# map

# serial
serial 0 64 115200 57600 0 115200

# led
led 0 8,15::CTOW:8
led 1 7,15::CTOW:8

# color

# mode_color

# aux
aux 0 0 2 1675 2100
aux 1 2 0 1600 2100
aux 2 13 1 1675 2100
aux 3 28 2 1675 2100

# adjrange

# rxrange

# vtx

# rxfail
rxfail 3 h
rxfail 4 s 1900

# master
set gyro_lowpass_hz = 100
set gyro_notch1_hz = 0
set gyro_notch1_cutoff = 0
set gyro_notch2_hz = 0
set gyro_notch2_cutoff = 0
set baro_hardware = NONE
set baro_cf_vel = 0
set min_check = 1015
set max_check = 1990
set rssi_channel = 8
set rc_interp = MANUAL
set rc_interp_ch = RPYT
set rc_interp_int = 9
set max_aux_channels = 4
set serialrx_provider = SBUS
set blackbox_on_motor_test = ON
set min_throttle = 1060
set max_throttle = 1990
set dshot_idle_value = 650
set motor_pwm_protocol = DSHOT600
set motor_pwm_rate = 16000
set failsafe_delay = 8
set failsafe_off_delay = 50
set failsafe_throttle = 1250
set failsafe_procedure = AUTO-LAND
set align_board_yaw = 90
set vbat_warning_cell_voltage = 34
set current_meter = NONE
set vbat_detect_cell_voltage = 33
set beeper_dshot_beacon_tone = 4
set yaw_motors_reversed = ON
set small_angle = 180
set disarm_kill_switch = OFF
set deadband = 5
set yaw_deadband = 4
set pid_process_denom = 1
set osd_rssi_alarm = 25
set osd_cap_alarm = 1000
set osd_tim2 = 1025
set osd_vbat_pos = 2446
set osd_rssi_pos = 2439
set osd_tim_1_pos = 54
set osd_tim_2_pos = 2454
set osd_flymode_pos = 2433
set osd_throttle_pos = 225
set osd_vtx_channel_pos = 377
set osd_crosshairs = 0
set osd_ah_sbar = 200
set osd_ah_pos = 200
set osd_current_pos = 385
set osd_mah_drawn_pos = 353
set osd_craft_name_pos = 2049
set osd_gps_speed_pos = 218
set osd_gps_lon_pos = 82
set osd_gps_lat_pos = 65
set osd_gps_sats_pos = 1410
set osd_home_dir_pos = 302
set osd_home_dist_pos = 303
set osd_compass_bar_pos = 266
set osd_altitude_pos = 247
set osd_pid_roll_pos = 423
set osd_pid_pitch_pos = 455
set osd_pid_yaw_pos = 487
set osd_debug_pos = 1
set osd_power_pos = 321
set osd_pidrate_profile_pos = 345
set osd_warnings_pos = 2064
set osd_avg_cell_voltage_pos = 76
set osd_pit_ang_pos = 257
set osd_rol_ang_pos = 289
set osd_battery_usage_pos = 392
set osd_disarmed_pos = 2283
set osd_nheading_pos = 311
set osd_nvario_pos = 279
set osd_esc_tmp_pos = 82
set osd_esc_rpm_pos = 83
set osd_stat_max_spd = OFF
set osd_stat_max_curr = OFF
set osd_stat_used_mah = OFF
set osd_stat_bbox = OFF
set osd_stat_endbatt = ON
set osd_stat_tim_1 = ON
set debug_mode = NOTCH

# profile
profile 0

set dterm_lowpass = 110
set dterm_notch_hz = 0
set dterm_notch_cutoff = 0
set vbat_pid_gain = ON
set anti_gravity_gain = 5000
set setpoint_relax_ratio = 20
set dterm_setpoint_weight = 180
set crash_recovery = ON
set yaw_lowpass = 100
set p_pitch = 70
set d_pitch = 23
set p_roll = 60
set i_roll = 47
set d_roll = 20
set i_yaw = 48

# rateprofile
rateprofile 0

set rc_rate = 107
set rc_rate_yaw = 91
set roll_srate = 80
set pitch_srate = 76
set yaw_srate = 82
set tpa_breakpoint = 1800

[image]
[BTFL_BLACKBOX_LOG_Shrieker_20171007_220131.zip](https://github.com/betaflight/betaflight/files/1366402/BTFL_BLACKBOX_LOG_Shrieker_20171007_220131.zip)

jflyper, 2017-10-08:
Isn't your CPU usage is 100% and Cycle time fluctuating?

Try with -DYNAMIC_FILTER.

jflyper, 2017-10-08:
I loaded a test quad with OMNIBUS (F3) loaded with your config.
No fatal behavior is observed.

On the way, I had to make the following changes.
- motor_pwm_protocol = DSHOT600 to ONESHOT125 (this test frame is still OS125)
- yaw_motors_reversed = ON to OFF
- align_board_yaw = 90 to 0
- Serial RX port from UART1 to UART3 (this is minor)

I think you need to review your quad and firmware configs.

cudakeer808, 2017-10-08:
Could you try roll the firmware back to nightly build #162? I might have the same problem and I don't have a black box but I was able to actually hover after I rolled it back. Just need confirmation that the Omnibus build onward of 162 has issue.

Arakon, 2017-10-08:
#162 made little to no difference. I can hover and even fly, I just have a constant vibration on roll going on.

cudakeer808, 2017-10-08:
How about putting back the d_term notch back to 260 on build #162?

Arakon, 2017-10-08:
I went down to 3.1.7 and left filters at default.. suddenly MUCH less vibration on the pre-notch traces. 
[BTFL_BLACKBOX_LOG_Shrieker_20171008_182811.zip](https://github.com/betaflight/betaflight/files/1367027/BTFL_BLACKBOX_LOG_Shrieker_20171008_182811.zip)

jflyper, 2017-10-08:
What if you do the same with 3.2; leave filters at default?

Arakon, 2017-10-08:
Very little to no difference to 3.2.0 with dynamic, except now P is going bonkers too (dynamic off, filters at default).
3.1.7 shows a lot less vibration 
[BTFL_BLACKBOX_LOG_Shrieker_20171008_195936.zip](https://github.com/betaflight/betaflight/files/1367083/BTFL_BLACKBOX_LOG_Shrieker_20171008_195936.zip)
than either.

ralenekov, 2017-10-11:
@Arakon It may sound strange but can you try to set the defaults for rc interpolation and try again. They should be something like:
set rc_interp = AUTO
set rc_interp_ch = RP
set rc_interp_int = 19

I had similar (roll axis only) problems on my F4 and NAZE32 targets with 3.2.0-rc3/4 and 3.1.7. Didn't try 3.2.0 stable to check if it has been fixed. ESC protocol didn't matter.

Arakon, 2017-10-11:
Done that too, it made no difference. Currently waiting for new motors to see if that changes anything at all.

romko1, 2018-03-19:
Hi, 
I have a board OMNIBUSF4SD (OBSD) 3.2.0 Sep 11 2017 / 06:44:20 (649ccd7c7) MSP API: 1.36
and I also have a problem when quad in the absolute plane is so changing the Pitch, Roll values (Betaflight 10.2. -Setup).
Of course the settings (see above) have the same and nothing is helpful.
Will you please advise me?
Thanks.