betaflight/betaflight #4301: Dshot not supported by F7 in known issues

goodnerdboy, 2017-10-08:
I don't know how to submit a change to the wiki so I put it here, apologies if I should not put it here.
I think Dshot not supported by F7 in 3.2.0 should be specified in the "known issues" of the release page so F7 owners know about it before flashing.

jflyper, 2017-10-08:
@goodnerdboy 
Yes, it's there. First thing in the Issues section.

https://github.com/betaflight/betaflight/wiki/BetaFlight-3.2#issues
> F7 support is quite early and dshot is currently broken. The devs know and are looking into it.
> See https://github.com/betaflight/betaflight/issues/4129

>

closed 2017-10-08 by jflyper

mikeller, 2017-10-08:
@goodnerdboy: Thanks, added it to the known issues in the release.

goodnerdboy, 2017-10-08:
@mikeller thanks! :)