betaflight/betaflight #4302: [Please add this back]  - 9 rate profiles are needed.

Rickcau, 2017-10-08:
I understand that someone made the decision to simplify the rate profiles and only have 3 shared across the 3 PID profiles.  I encourage you to think what this does to the *active* FPV race pilot.

1. For *all* my racing quads I use 3 rate profiles with the lowest DPS being at 274 and highest at 525.  <-  These types of low-rates are very common among racers, just ask Brian Morris.  
2. Most of us also fly freestyle and rates this low are not adequate for freestyle.  I run 950 and higher for freestyle and there is no way I can do that on any of my race quads without a major undertaking e.g. go into BF change the rates to my freestyle rates save and fly.
3.  Now when I am going to a race I have to make sure I have my 3 race rate profiles set back up.

I am sorry, but this is a lot of work and the amount of effort to add 2 additional arrays to restore this feature is worth brining it back.  This feature makes it easier for the pilots so it would be much appreciated if you bring it back.

mikeller, 2017-10-08:
The reason for PID profiles and rate profiles being separated was that the original system of making the rate profiles dependant on the selected PID profile was seen as overly complicated, and confusing for new users. Also, one common use case for PID profiles is to use them to try out different PID settings. In this case, with the old system, users were forced to make every change to the rates to 3 different rate profiles, in order to be able to test them with different PIDs.
 All in all, I am not sure that going back to the old way of having things is a good thing, having independent PID profiles and rate profiles seems to be easier to understand and manage.

But how about increasing the number of rate profiles (on boards that have the storage to do so)? Would it work for you if you had 3 PID profiles, and 9 rate profiles (possibly organised in 3 banks of 3, selectable with 1 (3 profiles available) or 2 (9 profiles available) 3 position switches?

OtterNas3, 2017-10-08:
As i'm same type of Pilot like Rickau, i did prefered the old system too.
Changing 3 Rate-Profiles for every PID-Profile "once" was much better then changing them now all the time.
And if you forgot to set the "Race" rates back to your FC before you hit the field without a laptop, it get's rly frustrating.
Yeah i know "Why not use OSD to change settings?"
Most Boards support that now with onboard OSD but there are still ppl out there that have old good working FC's mounted to their Drones which does not have a build in OSD and it is a PITA to carry a Laptop around just to change your rates :)

I understand that the separation makes it easier for the "common" Pilot, but it also gave a new Pilot more freedom to try out different Rates for different PID's.
"3 Rate profiles are more then enough" - Maybe
But having 2 Beer at a Bar is not enough for everyone that wanna get Drunk :)

Increasing of the Rate-Profiles to 6 or 9 would also be a good compromise!

Rickcau, 2017-10-09:
It would absolutely work if I had 9 rate profiles so yes what you are proposing would be awesome!

chime13, 2017-10-09:
I would rather have it go back to the way it was - 3 profiles with 3 rateprofiles per profile. I have a DX6i and can only toggle between 2 rates - this latest implementation really limits my choices. I don't believe it was confusing at all to setup 3 profiles with 2 or 3 rateprofiles per each profile as it was originally. I vote for restoring original rateprofile array functions.

Wes-Leigh, 2017-10-09:
Maybe add it in and have the Configurator set it as a expert feature. this way it will hide it for the people that are new to the hobby. Or make it a feature that needs to be enabled through cli commands?

martinbudden, 2017-10-09:
@Rickcau , this change was discussed and made ages ago. One of the things that is sometimes frustrating as a developer is the lack of direct feedback from active pilots. So if you are an active pilot I would encourage you to (at least some of the time) use the most recent development version of Betaflight and give your feedback and suggestions.

So back to the issue in question. Just because it was made ages age, that doesn't mean it shouldn't be changed back.

The simplest change would be to add more rate profiles, ie 6 or 9. I'd rather have 6 rate profiles than 9, unless you think 9 is absolutely necessary. Your view?

@chime13 , many people find 3 profiles with 3 rateprofiles per profile confusing. I'm not familiar with the DX6i - a quick look at the manual shows it has 6 channels, loads of switches and supports channel mixing. So I would think it would be possible to mix two switches onto a channel and so change between more than 2 rates.

chime13, 2017-10-09:
@martinbudden 
The Spectrum DX6i TX has 6 channels - but only 2 auxes, each aux is only a 2 position toggle switch. If I use StickArming then I can assign aux 1 for rateprofile selection and aux2 will either be beeper/buzzer or some other MODE selection.
If I were to use an Aux for arming, then that would leave me with only have 1 aux available.

"many people find 3 profiles with 3 rateprofiles per profile confusing"

With all due respect - it's pretty self-explanatory

martinbudden, 2017-10-09:
@chime13 , what mechanism do you use to change profile?

chime13, 2017-10-09:
@martinbudden, I use stick commands.  
Left stick down and to left + Right stick hard left = Profile 1
Left stick down and to left + Right stick straight up = Profile 2
Left stick down and to left + Right stick hard right = Profile 3

I use Aux 1 for rate profile/Airmode On toggle.
Aux1 (pos 1 = slow rate, pos 2 = fast rate+Airmode On)

martinbudden, 2017-10-09:
@chime13 ,  how about this for an idea:
1. Add 6 rate profiles
2. If in ACRO mode, then repurpose AccTrim stick commands to select rate profile 1 to 4

It will give you 4 rate profiles

chime13, 2017-10-09:
@martinbudden, That sounds awesome! I only fly ACRO without any other added features except for buzzer so any re-mappable unused StickCommands would be better than the current! Again, having only 2 AUXs on my TX is quite limiting, but assigning lesser-used rates* to StickCommands would give me back my 6 selections (AND NOW) independent of profiles - Very Cool!!!

*Though you can only change these values (1 and 4) while the craft is on the ground and disarmed...

OtterNas3, 2017-10-09:
@martinbudden 
"If in ACRO mode, then repurpose AccTrim stick commands to select rate profile 1 to 4"
Don't get me wrong, but that would be more confusing then the original 3-3 setup :D

I would prefer the old system, and if that is too complicated then 6 rate profiles (which would be enough in most cases i guess)

Most of the Pilots i fly with told me they did not like that they only have 3 rate profiles now.
We all got used to it by now but i still miss it, as most of them also do.

6 Rate profiles however would be a very nice compromise for them and for me too

mikeller, 2017-10-09:
@martinbudden:

> If in ACRO mode, then repurpose AccTrim stick commands to select rate profile 1 to 4

Not sure about this. This will introduce second layer of stick commands, making something that is already not well understood even more complex.

As an alternative, how about adding another `adjustment range` setting for rateprofile that allows the user to cycle through the available rate profiles with a momentary switch, much like they can cycle through PID settings with the respective adjustment ranges already?

chime13, 2017-10-09:
Any reason why we cannot revert to the original and forgo these extra complications and issues brought up by changing the code? 

It really wasn't that complicated and allowed for greatest flexibility - Now by oversimplifying, we crippled useful functionality for those that _can actually use it_.

Can't fool-proof everything in this world. This is a hi-tech hobby/passion that requires a degree of knowledge in electronics, soldering, programming, and not to mention, hand/eye coordination. 

Selecting rateprofiles should be a cinch if one has any of the above requirements.

Heck, soon we'll want these things to fly on their own! - ...oh, wait a minute..... - hahaha

DieHertz, 2017-10-09:
It was changed as part of parameter groups conversion.

mikeller, 2017-10-09:
> Any reason why we cannot revert to the original and forgo these extra complications and issues brought up by changing the code?

Yes. The original set up existed before parameter groups were introduced. It got simplified when the parameter groups were introduced, because supporting hierarchical configuration arrays was seen as overly complex, and would have resulted in considerable additional complexity in how parameter groups are done.

This could still be added , but the move to more independent rate profiles looks like a better approach.

As a compromise, we could add a 'rate profile bank selector' to the profiles, that would allow the user to define a custom starting point (in the list of profiles) for the 3 switchable rate profiles.

martinbudden, 2017-10-09:
> As a compromise, we could add a 'rate profile bank selector' to the profiles, that would allow the user to define a custom starting point (in the list of profiles) for the 3 switchable rate profiles.

No, let's not go there. This part of the code is complicated enough as it is. Let's just have 6 rate profiles - it satisfies most user needs and keeps the code simple.

chime13, 2017-10-09:
I'm all for innovation and simplification, but it seems to me that I will need more than the 2 AUXs with SPDT switches to be able to address banks of choices that are 2 menu levels deep.

I won't mention it again because it is annoying and possibly a sign of madness but.....

May someone explain what is so complicated about storing and addressing 2 or 3 rateprofiles per each profile? 

If the TX AUX(x) has a 3Pos toggle programmed for RateProfileSelections, the max number of addressable selections will be 3 per profile for a total of 9

In my case with a Spektrum D6Xi:

The TX AUX (1 and 2) has a 2Pos toggle for RateProfileSelections, the max number of addressable selections will be 2 per profile for a total of 6

Currently, while in the air I can only toggle between 2 rates.

Regardless, I appreciate all of you that contribute to the advancement of this hobby.

martinbudden, 2017-10-09:
> May someone explain what is so complicated about storing and addressing 2 or 3 rateprofiles per each profile?

1. Parameter groups don't support it.
2. It permeates the code and introduces dependencies between two different areas of code.
3. It complicates the UI.

Yes, it could be done. But it's a nightmare to program and there are much more important and interesting things to address.

mikeller, 2017-10-09:
@martinbudden:

> No, let's not go there. This part of the code is complicated enough as it is. Let's just have 6 rate profiles - it satisfies most user needs and keeps the code simple.

Not sure I made my proposal clear enough. What I meant to propose is 'leave the profiles and rate profiles independant as they are. Add an index (defaulting to 0) to every profile. When an 'adjustment range' switch is used to select rate profiles, add this index to the switch value when calculating the rate profile number to be selected.'
This will:
- solve the problem of being able to select more than 3 rate profiles with a 3 position switch;
- enable users to emulate the old 'profile dependant rate profiles' behaviour.

chime13, 2017-10-09:
@martinbudden, I appreciate the response. 
What exactly was the issue that required rewriting the function in the first place? It seems to me that it was indeed working correctly on all prior releases to 3.2. I used all it the time without performance issues or UI complications on versions before 3.1.7.

chime13, 2017-10-09:
@mikeller 
> When an 'adjustment range' switch is used to select rate profiles
Assignable stick commands would be great for TXs that have limited AUXs and no 'adjustment range' switches

martinbudden, 2017-10-09:
@mikeller , I understood what you meant. I'd rather try the simpler approach first - ie just have 6 rate profiles. If people still don't like that then we can think about extending it using your idea. Your change does add complications both in explaining how it works, in the UI and how things are stored. I always favour doing the simpler change first.

@chime13 one of the major "under the hood" changes to 3.2 was converting to using parameter groups. This vastly improved and simplified the code, but had little visible advantage to users (the main advantage to users probably being that many configuration changes did not need a reboot to take effect). However it does mean there is no going back. As I said, it would be possible extend the new code, but there are much more important and interesting things to implement. Also many F3 boards are approaching their ROM limit, and this functionality would also eat into that ROM limit.

You seemed very happy with my proposed stick solution before - why the change of mind?

mikeller, 2017-10-09:
@martinbudden: Totally agreed. As long as the `adjustment range` functionality stays the same and selects the first 3 rate profiles, doing it stepwise is a good way forward.

chime13, 2017-10-09:
@martinbudden 
> You seemed very happy with my proposed stick solution before - why the change of mind?

I am very please with the stick solution - no change of mind. Without knowing the reasoning behind the change(s), one could easily assume that more harm than good was done and why not revert? I now know there's no going back.

If I did have any reservations or concerns, it would be about being hopeful that my TX _will_ be able to implement this otherwise rudimentary function.

If 6 rateprofiles work for me with the implementation, I'll be much better off than I am right now.

Thanks guys!

fiveangle, 2017-10-10:
> @martinbudden: I'm not familiar with the DX6i - a quick look at the manual shows it has 6 channels, loads of switches and supports channel mixing. So I would think it would be possible to mix two switches onto a channel and so change between more than 2 rates.

FYI: it absolutly is. Airware is extremely capable, second only to OpenTX in configurability, and does so 10X easier.  No need to burn more than a single Aux channel for direct rate selection by mixing up existing switches.  But I digress :)

chime13, 2017-10-10:
@fiveangle , Before digressing - do you think you can point out the 10x easier procedure for the DX6i?
With all due respect - I think y'all are confused by the different Spectrum DX6 models as the 6i is not as robust as the 6 or the 6E.

etracer65, 2017-10-10:
@martinbudden Personally I'm not concerned about having more than 3 rate profiles, but I am concerned about how increasing the count to 6 rate profiles will complicate the selection process.  Currently the simplest way to set up your radio is to simply assign a 3 position switch to an aux channel and use that entire channel range for the adjustment that selects the rate profile.  So assuming aux 2, it would be:

`adjrange 0 0 1 900 2100 12 1`

[image]

Simply assigning a switch to a channel in pretty much every radio results in 1000us, 1500us and 2000us in the 3 positions.  So if now there were 6 positions it would require some complicated multi-switch multiplexing and much more involved radio programming.  So less capable radios might not be able to do this.

Then to add to the confusion, simply using a 3 position switch like now would result in rate profiles 1 and 6 being accessible (1000us and 2000us), but the middle position would select 1500us which would be right between rate profile 3 and 4.

martinbudden, 2017-10-10:
@etracer65 , yes, I've thought about that. I plan to add a new adjustment range, so we have "Rate Profile Selection - 3 Rates" and "Rate Profile Selection - 6 Rates" (names subject to change). The first would use a 3 position switch to change between rates 123, and the second would use a 6 position switch to change between rates 123456 (or alternatively 2 switches multiplexed onto one channel).

Rickcau, 2017-10-11:
Martin,

I would assume that the rate profile selection would be accessible via the BF OSD.  If this is not the case it's useless for me.  Keep mind that some folks don't like having lots of switches, and I am one of those people.  If there is no way to have 3 rate profiles and to be able to select the rate profile from BF OSD then I think most won't use it.  The only think I personally like on a switch is my ARM/DISARM.

Also, some folks are using the Evolution which I don't think has a 3 position switch.

For me, 3 rate profiles are fine if I never want to do freestyle on my race quads, but when my freestyle QUADS are down I find myself doing acro on my race quads and it's a royal pain to reset the rates, for obvious reasons.

Rickcau, 2017-10-11:
I moved back to BF for the OSD on all my race rigs and before 3.2 I use the 3 PID profiles using the same PID settings but have 3 different rate profiles for each PID profile.  For me, this is what worked the best.  But, I understand the need to simplify the code as I am a developer myself.  So, if you could simple add a 4th rate profile I would be very happy.

mikeller, 2017-10-11:
@etracer65, @martinbudden: I did a little test last night, and the way that adjustment ranges are implemented means that even after #4313, the corresponding adjustment range only has 3 positions, so only rate profiles 1, 2, and 3 are available on a 3 position switch.

martinbudden, 2017-10-11:
This is still a work in progress.

And, yes, we can also add this to the OSD menu system.

bigb2000, 2017-10-22:
@chime13: 2 years ago i flew helicopters with the DX6i and used it with a mixer (DX6i has 2 mixers) to use 2 switches for one aux channel, so you can assign 4 output values on one channel. So i selected between 3 rpm's on the heli and 4 different banks on Vstabi (or Vbar; a flybarless system for helicopters). It's a bit tricky to set up, but it works.

fiveangle, 2017-10-22:
@chime13 - you need to boost up your google-fu 😉, but this should give you some ideas:

https://youtu.be/gh01RqrJw0o

-=dave

mikeller, 2018-06-03:
@Rickcau: We have gone back to 6 rate profiles now. Can this be closed?

closed 2018-06-10 by mikeller