betaflight/betaflight #4303: I can not disable the beeper

Melokman, 2017-10-08:
betaflight 3.2 RC6 
Activated the beeper on aux2. A turtle mode on aux3. When I turn on the 4s the beeper begins to ring.
I delete the turtle mode in the configurator, the beeper works fine. What could be the reason?

mikeller, 2017-10-08:
Could it be that your AUX3 channel signal is activating crash flip mode? This causes the craft to issue warning beeps.

Melokman, 2017-10-08:
I decided that AUX did not have anything to do with it. so I adjusted the theme and content

mikeller, 2017-10-08:
Ah, my post crossed with your update. Your description now sounds like you have to adjust the maximum cell voltage up in order to get your batteries to be recognised properly.

Also, please note that GitHub is for questions and discussions around the development of Betaflight. For questions about the configuration and use, please see [RCGroups](https://www.rcgroups.com/forums/search.php?searchid=62590553&query=betaflight%20max%20battery%20voltage).

closed 2017-10-08 by mikeller