betaflight/betaflight #4304: BF3.2 Arming in Receiver tab with stick input

PitFlyerFPV, 2017-10-08:
When I was checking the stick endpoints (in Receiver tab), my quad accidentally armed. 
This could happen because I had to plug in my batterie to have power on the RX and I didnt set up the arm switch at this point. In this situation the default arming with the sticks was active and during the endpoint check the quad can arm. 

Nothing bad happend today but if you dont remove your props it could cut your fingers or more seriously. It is not expected in this situation that the motors are arming. 

My FC is a DYS F4

dump:
```
# version
# Betaflight / OMNIBUSF4 (OBF4) 3.2.0 Sep 28 2017 / 16:38:25 (ceb6f70) MSP API: 1.36

# name
name -

# resources
resource BEEPER 1 B04
resource MOTOR 1 B00
resource MOTOR 2 B01
resource MOTOR 3 A03
resource MOTOR 4 A02
resource MOTOR 5 A01
resource MOTOR 6 A08
resource MOTOR 7 NONE
resource MOTOR 8 NONE
resource SERVO 1 NONE
resource SERVO 2 NONE
resource SERVO 3 NONE
resource SERVO 4 NONE
resource SERVO 5 NONE
resource SERVO 6 NONE
resource SERVO 7 NONE
resource SERVO 8 NONE
resource PPM 1 B14
resource PWM 1 B14
resource PWM 2 B15
resource PWM 3 C06
resource PWM 4 C07
resource PWM 5 C08
resource PWM 6 C09
resource PWM 7 NONE
resource PWM 8 NONE
resource SONAR_TRIGGER 1 NONE
resource SONAR_ECHO 1 NONE
resource LED_STRIP 1 A01
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 NONE
resource SERIAL_TX 3 B10
resource SERIAL_TX 4 NONE
resource SERIAL_TX 5 NONE
resource SERIAL_TX 6 C06
resource SERIAL_TX 7 NONE
resource SERIAL_TX 8 NONE
resource SERIAL_TX 9 NONE
resource SERIAL_TX 10 NONE
resource SERIAL_TX 11 NONE
resource SERIAL_TX 12 NONE
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 NONE
resource SERIAL_RX 3 B11
resource SERIAL_RX 4 NONE
resource SERIAL_RX 5 NONE
resource SERIAL_RX 6 C07
resource SERIAL_RX 7 NONE
resource SERIAL_RX 8 NONE
resource SERIAL_RX 9 NONE
resource SERIAL_RX 10 NONE
resource SERIAL_RX 11 NONE
resource SERIAL_RX 12 NONE
resource INVERTER 1 C00
[... 576 lines cut ...]
set gps_pos_d = 0
set gps_posr_p = 34
set gps_posr_i = 14
set gps_posr_d = 53
set gps_nav_p = 25
set gps_nav_i = 33
set gps_nav_d = 83

# rateprofile
rateprofile 0

set rc_rate = 100
set rc_rate_yaw = 100
set rc_expo = 0
set rc_expo_yaw = 0
set thr_mid = 50
set thr_expo = 0
set roll_srate = 86
set pitch_srate = 84
set yaw_srate = 84
set tpa_rate = 10
set tpa_breakpoint = 1650

# 
```

mikeller, 2017-10-08:
This is expected behaviour.

GitHub is for questions and discussion around development of Betaflight. For questions about configuration and use, please see [RCGroups](https://www.rcgroups.com/forums/search.php?searchid=62590336&query=betaflight%20newbie%20setup).

PitFlyerFPV, 2017-10-09:
But it is a dangerous expected behaviour. 
In the motor tab you have a warning label. You should also see an info that switch arming is active or even bether disable arming when in receiver tab.

martinbudden, 2017-10-09:
@mikeller , I'm re-opening. Even if this is expected behaviour, a user was caught out. Something like a "Disable arming" checkbox in the receiver tab could be a useful addition.

mikeller, 2017-10-09:
@martinbudden: Good point. How do you think this 'disable arming' checkbox should be working? Should we add a 'disable_arming' MSP command?

However, since this is only one out of a near endless list of how users can create dangerous situations by misconfiguring their craft, my proposed solution would be to add a big red warning like `ALWAYS remove props BEFORE connecting battery while connected on USB!` to the top bar of the configurator.

Jahnkeanater, 2017-10-10:
> Should we add a 'disable_arming' MSP command?

However, since this is only one out of a near endless list of how users can create dangerous situations by misconfiguring their craft, my proposed solution would be to add a big red warning like ALWAYS remove props BEFORE connecting battery while connected on USB! to the top bar of the configurator.

This should not be a configurable feature. Arming should never be allowed unless the user specifically flips a switch In the motors tab. Leaving the tab should then disable arming again.

There is already a disable beeper while plugged into USB feature so its possible for the quad to know when its plugged in without MSP telling it so.

mikeller, 2017-10-10:
@Jahnkeanater: How about simply not connecting a battery to set up your modes?

> There is already a disable beeper while plugged into USB feature so its possible for the quad to know when its plugged in without MSP telling it so.

Unfortunately no. The craft detects that it 'must be plugged into USB' by measuring the voltage, and if it is outside the range of what can be attributed to any number of LiPos (like the 5V supplied by USB), it decides that it 'must be on USB'. This obviously does not work when on USB with a battery connected. (Go try it out. :-P)

Jahnkeanater, 2017-10-10:
> How about simply not connecting a battery to set up your modes?

There are a lot of FC's that do not supply 5v to the Rx when powered from USB. People are going to plug their batteries in to configure there radio and modes. 

There has to be a simple way for the FC to figure out its plugged in. If the only way to do this is to create a MSP command then why not. The longer this issue is ignored the more people that are not being increadibly careful will be injured.

mikeller, 2017-10-10:
> The longer this issue is ignored the more people that are not being increadibly careful will be injured.

I disagree. Removing the props before plugging in a battery while on the bench is not 'incredibly careful', not doing so is careless and ignorant of all safety precautions.

Jahnkeanater, 2017-10-10:
Even with the props removed it is dangerous to arm the quad while plugged it. Yaw oscillations can easily cause the quad to snag on things. As soon as you plug the quad in you have to be 'incredibly careful' that while reassigning modes you do not accidentally cause the arming channel to arm.

I'm not saying this needs to be fixed asap but it shouldn't just be ignored. Closing the issue and calling people 'careless and ignorant' isn't ethical. The hobby is growing and this is an issue that should eventually be addressed.

closed 2017-10-11 by mikeller