betaflight/betaflight #4306: Roll of Death

SupaCoop, 2017-10-09:
Hi, I've been having weird stuff going on with my new build.

The Build:
- Armattan Chameleon 5 inch
- Red Rotor RROSD v2
- Lumenier Lux v2
- Lumenier 30a DSHOT Blheli_s esc's (Re-branded racerbee's with signal cap removed)
- T-motor F40iii 2400kv
- Runcam Eagle 2
- TBS Unify Pro HV
- Spektrum quad race receiver with telemetry
- HQ 5x4.3x3
- Thunder Power Adrenaline 4s 1300mah 100c

Running betaflight 3.2 at 4k/4k with dshot 600, esc's running blheli_s 16.63, Airmode, Anti-gravity, Telemetry and Dynamic Filter enabled with pt1 enabled and first notch filter disabled. Both fc and motors are soft mounted. Idle motor speed at 4.5, min throttle command in receiver tab at 1010, and esc startup power at .50. 

 I've seen other people experience a similar problem, but with very simple fixes; I have yet to find a fix. In a nutshell, when I do a snappy roll or flip, my quad continues to roll or flip after I release the stick until it hits the ground. When it makes contact with the ground, I disarm, and hear the esc's or esc restart and go through their tones again. I can then re-arm and take off again. I can fly around just fine without having any issues, the quad is very smooth, about 80% percent tuned, and flies great without doing anything crazy. The most frustrating thing about this is the inconsistency of it. I can get through 2 or 3 packs really pushing it without any issues, and on the 4th pack right when I take off and do a roll it will spiral into the dirt. I don't necessarily know if it's a desync because I've tried everything that's supposed to fix desync's. I've upped the idle percentage, upped min throttle command in the receiver tab, played with startup power in the blheli config, I've tried earlier blheli_s and betaflight firmware's, lowered gyro and pid loop frequencies, played with different dshot levels, and I've gotten nothing. My motors are running cool and the esc's don't get hot what so ever. I then went to blackbox and saw that motor 3 was staying at 100% when this problem occurs to I replaced that esc, same thing still happens. My next log showed that motor 1 was staying at 100%, so I replaced that esc with no success. I don't think it's the motors because their brand new and their just doing what the esc tells them. Motors 1 and 3 are still spooling up to 100% and staying there. I switched out the lux for a working one from another quad with no success. I honestly don't know what to do next so that's why I've come here. I don't think this is a defective hardware issue because I've replaced and switched out parts, I don't know if it's a firmware issue because it sometimes flies great with no issues. I've practically rebuilt the quad so it can't be a loose wire or connection. The latest thing I've noticed is that the last couple times I've landed after testing (when it doesn't crash itself) the esc's or esc restarts when I touch down; when it bobbles a couple times cause airmode is on. 

Thanks for your time, Cooper

digitalentity, 2017-10-09:
Try going waaaay back to OneShot125 at 1khz. If it flies fine - you have noise issues and can't use DHOT on that setup.

SupaCoop, 2017-10-09:
digitalentity Thanks for the fast reply man! I'll definitely give that a shot tomorrow!

SupaCoop, 2017-10-09:
@digitalentity Nope, just did it in my backyard. Not a noise problem I guess.

digitalentity, 2017-10-09:
Ok, at least it's not a DSHOT-related problem. Can you upload a few blackbox logs of the problem?

SupaCoop, 2017-10-09:
@digitalentity sorry man I'm not very tech savy, how would I do that? I tried dragging the file into the message box and it doesn't accept it.

digitalentity, 2017-10-09:
No worries, Try compressing the file first (to a zip-file). 
Alternatively you can use a cloud service (like DropBox of OneCloud - I'm sure you have one) and post a link here.

SupaCoop, 2017-10-09:
[Log1.zip](https://github.com/betaflight/betaflight/files/1367404/Log1.zip) 

@digitalentity Ok, I'm not sure if I did that right, but if I did there's an example of motor 1 and an example of motor 3 sticking at full throttle.

digitalentity, 2017-10-09:
Thanks! I'll have a look later today.

SupaCoop, 2017-10-09:
@digitalentity Alright thanks man, I really appreciate it.

digitalentity, 2017-10-09:
Yeah, looks like motor 1 is stopping mid-air. Btw, do you have signal ground from ESC connected to flight controller or you have only signal wire connected?

SupaCoop, 2017-10-09:
@digitalentity Motor 1 is stopping mid-air? Yes I always ground my esc's directly to the fc.

KSCopter, 2017-10-10:
It is very likely you have a bad ESC that is de syncing when you throw a burst of power through it.  I have had the same problem on two different setups and fixed it with new esc(s).  I have seen Boris comment on this elsewhere.  I would start by replacing esc #1.

SupaCoop, 2017-10-10:
I have already replaced esc 1 and 3.

SupaCoop, 2017-10-10:
@digitalentity

backwoodsneckbeard, 2017-10-13:
I am having the same issue.  Intermittent rolls / flips of death.  Some packs the quad flies absolutely fine, other packs and a quick flip or roll sends it into a spiral of death. Had no issues with these escs on the other quad I had them on running BF 3.2 RC5.  

My build is
BF 3.2 stable
Super Racer Bee ESC on latest BLeli_S (DShot 600 enabled in config)
Tempest f3 FC
FrSky XSR
Unify Race
Runcam Swift 2
Tmotor F40 Pro

Jahnkeanater, 2017-10-13:
Check for loose prop nuts and lose ESC wires. If the prop slips on the shaft it will make a high pitched grinding/squeaking noise. 

Also, make sure the terminals inside the xt60 plug are making good contact. Double check your solder joints.

backwoodsneckbeard, 2017-10-13:
None of the above apply. Thanks though.

dgrigg123, 2017-10-15:
I just had the exact same issue:
Armattan Chameleon 5 inch
BF F3. BF 3.2.0. latest release
4k/4k
Wraith blheli_33  35A
DShot 1200
Riddle motors 3550kV

I'll upload logs once I pull them down and review, but I had the exact experience as SupaCoop.
Flies great, but 2 times when doing a snap roll iy went into an infinite spin when return to center. In my case,  putting some command back in during spiral corrected the spiral and Luckily it did not cause a crash and I continued to fly.

KSCopter, 2017-10-16:
Interesting...just had the roll of death issue, again,  this weekend.  Only when doing fast rolls or punch outs.  This time I noticed small nicks on my XT60 connector.  Turns out, the centrifugal force generated during the fast rolls was pushing/pulling the wires/connector toward the prop.  Instant roll of death.  The wires return to the original position when the craft is not moving, making it impossible to see what happened. Same thing on fast acceleration, the wires/connector were pulled/pushed ever so slightly into the path of the prop.  Again, instant roll of death.  It only takes a little bit of contact, not even enough to visibly damage the prop.  

Solution:  Ultra strong rubber band to better secure the battery cable.  

This is not the only cause for roll of death (ROD), but it is certainly one of them. For what it's worth..

choas0422, 2017-10-18:
Same thing here! I have one of the BFight's.

omnibus pro f3 fc
blheli_s esc's
2205 motor

After the esc firmware update to 16.67, and the betaflight firmware update to 3.2 rc6 i have one motor that randomly drops out causing the quad to tumble uncontrollably.

I've tried:
-upping digital idle speed to 8 from 4.5 (helped i think)
-changed the esc min and max values to 1016min and 2000max
-increased the esc timing to high (helped i think)
-increased dmag to high (helped i think)
-tried toggling airmode
-tried with and without antigravity
-motors in betaflight run fine but when using the sticks the one motor stalls
-tried every version of dshot up to 600 (only supports up to 600)
-lubed the bearings just to make sure it wasn't a failing bearing in the motor
-tried to downgrade the esc firmware (1 step back and 2 steps back)
-checked all solder joints, plugs, and wires.
-tried banging my head into the wall while crying over my sad non flying quad... also did not help...

choas0422, 2017-10-19:
If i was going to try swapping the motors around to check if it's a hardware issue what motor wires go where? does it matter what wires i solder to what pads? or can i just solder it up and make adjustments in blheli config?

choas0422, 2017-10-19:
I figured it out! It took a week but I got it! If you look real close at the motor in question, on hard throttle moves the windings on the inside are moving back and forth. also probably why increasing the motor timing and the dmag seamed to help.

tavdog, 2018-01-10:
I'm having the same issue.  I've seen it happen on both of my rear motors on separate occasions.  Seems to only happen on 4s.  Would an overpropped motor cause a desync and motor stop ?
5050 props on 2204-2300 
https://www.rcgroups.com/forums/showthread.php?3004863-Death-spiral-blackbox-logs#post38957768

mikeller, 2018-01-10:
@choas0422: Is this fixed now and can it be closed?

@tavdog: Have you tried increasing the Dshot idle value?

tavdog, 2018-01-10:
I've decreased the esc startup power and increased the idle percent to 10%

tavdog, 2018-01-11:
I've concluded that my issue is caused by running a 4s battery on cheap motors that can't handle the higher voltage.  

I do not believe this is a betaflight issue so my vote is to close this issue.

KSCopter, 2018-01-11:
I, too, have traced my more recent ROD issue to cheap motors desyncing.  Also, none of my previous ROD occurrences can be traced to BF. 
I, too, vote to close the issue.

closed 2018-01-11 by mikeller