betaflight/betaflight #4307: Beta Flight and Taranis RX issue

iskymaster, 2017-10-09:
I have a very specific question related to Eachine X220S.
1. I am using Taranis Plus as a TX.
2. Using Taranis X8R Rx.
3. Beta Flight 3.2.1
4. Flight Controller F3 6DOF

The receiver is correctly "Bind" with Tx.

Using S-BUS to connect to F3 6DOF (yes using S-BUS connector on the FC).

Beta flight does connect with FC properly. Serial and S-BUS features are enabled in Beta Flight.

The problem I have is that Beta Flight doesn't see the receiver inputs when you move the sticks. The TX is properly set for AETR, D16 Channel 1 to 8.

Any advice? Spent few hours reading, watching few videos still can't get Beta flight to register stick movements or to show anything.

wind0r, 2017-10-09:
Since Betaflight 3.2.1 doesnt exist I have a big problem to reproduce your "Bug".
My crystal ball say that you may configured the wrong port in ports tab.

Also I do not want to be harsh but:
Please consider using other user support options such as asking the manufacturer of the hardware you are using, [RCGroups](https://rcgroups.com/forums/showthread.php?t=2464844), [Slack](https://slack.betaflight.tech) or other user support forums & groups (e.g. facebook).

Since this should be for Betaflight bugs and not configurator issues or company support.

closed 2017-10-09 by mikeller

iskymaster, 2017-10-09:
Interesting reply, pretty much a smart ass reply. If you can't help why bother chiming in?

3.2.1 is the latest configurator.

Why Taranis X8R receiver will provide help for beta flight? The receiver works just fine with the TX, it's the Beta flight that doesn't recognize the S-BUS. Ports are configured correctly, Beta flight is set for Serial and S-BUS. It doesn't recognize it.

I did post this question on some other forums, same thing, all smart ass people don't wanna help :)

wind0r, 2017-10-09:
Please explain me how you I help you?
Yes. 3.2.1 is the last configurator, that still does tell which firmware you use. 
A simple issue just can be that you do not use the matching firmware, but somehow you are not willing to provide this information.

if your firmware is supported with the configurator you use it just can be as i said Ports / Wireing or wrong Protocol is choose.
I own pleanty SBus receivers which all are working. Also the major part of the Betaflight Userbase is using those. No one else reports such kind of Bug. So a Bug we can fix in the Source is very very unlikely. It just can be bad documentation from manufacture side or a configuration issue. which we do not want to have the bug tracker filled with.

I never said you should as FrSky. The manufacture of the Flight Controller should help you. They are in the control to fix bugs with thier Target. As i allready said. We are allready spending our freetime here. We do not want spend time on topic the flight controller manufacture should do and also we are not willing to buy all flight controllers and quads just because they write betaflight on it.

iskymaster, 2017-10-10:
I managed to solve the problem and no I was not trying to be mean towards you. I didn't find your comment giving any help.

Yes, I understand people are spending free time, so do I as I have helped 100s of people with PixHawk and other FC related issues as I am very comfortable with those FC and mission planner software etc. So you are not alone spending free time helping the community.

1. The Firmware needed to be flashed with new firmware. The factory default firmware was whatever and not sure what it was.
2. The FC F3 6DOF has not listed in Bata Flight controller drop-down menu. Not sure why, so you have to select SPRACINGF3 (as per Oscar Liang) which is another thing which I find very confusing. Maybe its the same product that has two different names?

Not sure what is the official name of this FC, however by selecting SPRACINGF3 and flashing the FC, now beta flight controller can receive S-BUS signals.

Problem solved.

Perhaps in beta flight in the future, we may consider adding a feature where Beta flight detects the flight controller and automatically provides several options? Not sure if the FC has enough smarts built into its processor to provide such information back to Beta Flight to make sense out of it.

iskymaster, 2017-10-10:
By the way, what is the difference between Beta Flight and Clean flight? To me, they both look the same.

wind0r, 2017-10-10:
There isn't any real way to detect the flightcontroller, beside using CLI and typing version but that can return what ever the manufacture want and most people know which flightcontoller they own. Also there are ideas to move for single binarys per mcu architektur and not per target.

To understand for you why we do not like this kind of issue here is that it looks like any other configuration issue. didnt provied enough information for any help and the biggest part is that 
Eachine simply cloned the SPRACINGF3 from @hydra the create of cleanflight. If they clone hardware they atleast should do the support on thier own and not also let do us the work.

cleanflight is the clean code version of baseflight.. since boris was playing around with the pidcontroller he create betaflight.. some months ago cleanflight rebased on betaflight.

iskymaster, 2017-10-11:
I am an electronics engineer by profession and we are developing a new FPV product for the community which will be launched next year.

What I find frustrating is that since the inception of RC hobby, we had well defined (almost) hardware standards. Since drones came into play all hell has broken loose. Any manufacture can make a FC whatever the way they want, no industry standards, no centralized place for any information, manufacturer pick and choose whatever connector types they want, no standards around pin configuration, software are developed based on whoever think is the right way to develop a software :) list goes on.

I see the benefit of open source concept but due to lack of a governing body setting standards for all is the real reason their is now a Chaos in this hobby. Its anybody game. RC support web sites are full of people posting questions, confused, not sure how to solve problems.

I understand this site is for pointing out bugs related to Beta Flight (Which what I thought I had)...but the question remains what would define a software bug vs. hardware bug vs. if someone has a clone or not a clone? :)) Getting to that conclusion is the first challenge to begin with due to lack conformity in this field.

Actually after noticing this issue myself, we are developing a well organized, tightly controlled community forum where the real experts will answer specific to specific problems rather allowing any joe blow to become an expert and confuse the hell out of the user base (That is the case on DIY Drone, RC Group, etc. etc.) Wish us luck!!