betaflight/betaflight #4310: Airbot F4 / Flip32 F4 VBAT Detect 3.2.0 (Stable Version)

despotic931, 2017-10-09:
I have discovered a potential bug involving the detection/sensing of VBAT voltages on the Airbot F4 boards using the stable release of 3.2.0. I have two builds using this board which were flashed with 3.2.0 over the weekend, and then set up from scratch. No backups or CLI dumps were used in the process. 

Upon flashing with 3.2.0 both boards fail to display battery voltage in the BetaFlight configurator (version 3.2.1) or on the external osd boards flashed with MWOSD.

If I revert to 3.1.7 the issue goes away and VBAT sensing functions again, then if I flash 3.2.0 the issue returns. This was duplicated on both of my boards, and no backup restorations or CLI dumps were done in the process. One of these boards was purchased directly from Airbot and the other from a local hobby shop, and both are connected to MicroMinimOSD boards flashed with MWOSD. They are not new builds and have functioned flawlessly with 3.1.7.

I have forgone the inclusion of a diff dump as this issue is present even without making any changes after flashing on both boards.

jflyper, 2017-10-09:
@despotic931 
Is source for battery meter set to onboard ADC?

despotic931, 2017-10-09:
Ok, so I had the battery meter set to ADC, but for some reason it was not reading per my comment above. Then I just redid the process, selected ADC, saved but with a battery connected this time, and it actually worked. If I just flash, set ADC, and save it doesn’t seem to work (won’t read battery in OSD or in configurator when battery is plugged in), but same process with a battery plugged in works.

Very weird, but it is working now and I am closing the issue.

closed 2017-10-09 by despotic931

jflyper, 2017-10-09:
@despotic931 
Battery tab's save doesn't actually save the sources. This is a BFC's problem, or the procedure is under documented...

mikeller, 2017-10-10:
@jflyper: What do you mean by this? The battery tab _should_ save the resources when 'Save' is clicked. Or else it's broken...

jflyper, 2017-10-10:
@mikeller I have a habit to do save on battery tab and goto other tab to do save & reboot to make the change permanent. Can you double check if save only works? I'm a way from my bench and boards.

mikeller, 2017-10-10:
Just checked, and the 'Save' button on the Battery tab in configurator 3.2.1 works as advertised.

jflyper, 2017-10-10:
Thanks for checking!