betaflight/betaflight #4311: Matek F4-OSD and Dianton F3 V3 no dshot command bep

devbloggs, 2017-10-09:
Flashed to 3.2 Stable with Matek F405-OSD and Diatone F3 V3 and cannot get the motor beep to work, only the regular beeper sounds.

I used the "beeper_dshot_beacon_tone = 1" command and saved with no result, only the buzzer sounds

Using DSHOT 1200 32 Bit Esc's using 32.1 Firmware
AND
Dshot 600 esc's using 16.6 Firmware

Anti Turtle mode works great btw :)

adrianmiriuta, 2017-10-09:
@devbloggs
you need new firmware for ESC's
blheli_s  v 16.63
blheli32  v 32.15

devbloggs, 2017-10-09:
Where would I get newer firmware, blheli suite says I have the newest already

elektronisk, 2017-10-09:
Had same issue myself. Use the blheli configurator app for Chrome to update.

mikeller, 2017-10-09:
https://chrome.google.com/webstore/detail/blheli-configurator/mejfjggmbnocnfibbibmoogocnjbcjnk?hl=en

^ That guy.

devbloggs, 2017-10-09:
Doesn’t work with blheli32

mikeller, 2017-10-09:
Correct. Looks like the downloadable versions were pulled from https://github.com/bitdump/BLHeli/tree/master/BLHeli_32%20ARM . :-(

Best open an issue in that repository and ask for them.

adrianmiriuta, 2017-10-10:
@devbloggs 
these are release candidates (very beta)
only blheli_s
https://github.com/bitdump/BLHeli/tree/master/BLHeli_S%20SiLabs/Rev16.63%20Dshotprog%20test%20code/Hex