betaflight/betaflight #4316: LED color from vtx channel

Jahnkeanater, 2017-10-10:
BF 3.2 stable. 

      This feature still does not work the way it should. All Raceband channels are red. Changing to Fatshark band did change the color to purple. The color should be based on channel not band.

      Maybe simplify the code to only look at the frequency range to determine color. You can also allow the VTX color feature to disable the color from aux channel feature. This feature can be useful for races and the VTX color should represent the frequency. 

5640-5670           White    1
5671-5710           Red       2
5711-5750           Orange 3
5751-5787           Yellow   4
5788-5824           Green   6
5825-5861           Blue     10
5862-5909           Purple  11
5910-5950           Pink     12

The aux color modifier should be able to sweep through every color in this range. This allows copters that don't have VTX control to manually select an appropriate color.

Jahnkeanater, 2017-10-10:
A simple equation to determine what range the frequency is in can be formed to simplify the code more.

floor((Frequency-5640)/37)+1

Jahnkeanater, 2017-10-13:
Lines 460-471 of ledstrip.c why is the range limited to "previousColor through color to nextColor"? It would be nice if an aux channel could fade through all HSV values from WHITE through DEEP_PINK.

mikeller, 2017-10-13:
Define 'all HSV values'. ;-)

The motivation behind this code is that, because HSV is really a 3 dimensional space (hue, saturation, brightness), 'all values' is not a defined transition. The current implementation allows the user to define the length and direction of the transition vector in a concise way, by selecting the colour for the starting point, mid point, and end point.

Any suggestions on how to improve on this are welcome.

Jahnkeanater, 2017-10-13:
The user can define the start and endpoint? 

I was under the impression that previousColor and nextColor was just +/- 30 hue. I really just want a knob on my radio to fade through white/red/orange/yellow/green/blue/purple/pink. So hue 0-360. 

How do you redefine the starting point and endpoint?

Jahnkeanater, 2017-10-13:
This guy seems to have the same issue.
https://www.youtube.com/watch?v=FeSw-YBUJms
https://github.com/betaflight/betaflight/commit/5e1c0a273ceb81185dcb98ba1c11aca3779cef58

mikeller, 2017-10-13:
@Jahnkeanater: See https://github.com/betaflight/betaflight/wiki/LED-Strip-functionality

Jahnkeanater, 2017-10-14:
Ok so if I don't select a different default color the aux channel allows me to fade through the entire spectrum. That's not very intuitive but I understand. Can we add one of those grey question marks to the configurator to explain this?

Maybe change the label when you hover over color 15. Instead of saying black it should say Aux.

Jahnkeanater, 2017-10-14:
LED function color this does not work. You cannot fade through the entire spectrum. ARM state function you can.

mikeller, 2017-10-14:
Yes, the new functionality only applies to fading based on a fixed colour, since for everything else, there is no such thing as previous and next colour in the list.

Jahnkeanater, 2017-10-14:
I'm testing my code but I'm running into an issue. No matter what frequency I set my smart audio vtx to the LEDs always show white. On BF 3.2 Stable the LEDs are always red. It seems like the frequency returned by vtx58frequencyTable[band][channel]; is always less than 5711.

Is this value only updated by lua scripts? I'm using the OSD to change channels. When I hit set the leds blink but never change colors.

mikeller, 2017-10-14:
What values do you get for `band` and `channel`? `vtx58frequencyTable` is constant, so the values in there don't change.

Jahnkeanater, 2017-10-14:
I'm not sure what the values are. I have no way of finding out I'm just testing it on a quad.

I think I found a bug with smart audio. Everything seems to be shifted by 1 band and 1 channel.

Fatshark band (4) works except for channel 8 is white not pink.
Raceband band (5) is all white.

vtxCommonGetBandAndChannel(&band, &channel);
returns the band and channel before
vtx58frequencyTable[band][channel]; returns the frequency.

I also noticed that in vtx_smartaudio.c
 saSetBandAndChannel(band - 1, channel - 1);
are the bands and channels for SmartAudio from 0-4 and 0-7 instead of 1-5 & 1-8?
or is the table vtxCommonGetBandAndChannel indexed starting at 0 not 1.
I think vtxCommonGetBandAndChannel(&band, &channel); needs to be subtracted by 1 before looking up the frequency?

I'm worried that simply subtracting band and channel by 1 might break the values returned using a tramp vtx.

Jahnkeanater, 2017-10-14:
So for example when vtxCommonGetBandAndChannel(&band, &channel); returns
band = 3 and channel = 5
vtxCommonGetBandAndChannel
should return 5885 if its indexed starting at 1
but instead it returns 5840 because its indexed starting at 0?

This gives the wrong frequency which results in the wrong color and raceband being white since its trying to look outside the table.

mikeller, 2017-10-14:
> I'm not sure what the values are. I have no way of finding out I'm just testing it on a quad.

For quick and easy live debugging, just assign the values in question to `debug[0-3]`, and they will be accessible in the 'Sensors' tab in the configurator (and in blackbox logs).