betaflight/betaflight #4321: Betaflight 3.17 cc3d Lost Buzzer setup on motor 6 Help?

alanfj, 2017-10-10:
# If your issue looks like a hardware fault or a configuration problem please don't raise an issue here.

## Please consider using other user support options such as asking the manufacturer of the hardware you are using, [RCGroups](https://rcgroups.com/forums/showthread.php?t=2464844), [Slack](https://slack.betaflight.tech) or other user support forums & groups (e.g. facebook).

## If you believe there is an issue with the firmware itself please follow these steps:
1.) Describe your problem;
2.) Include ways to reproduce the problem;
3.) Provide information about your flightcontroller and other components including how they are connected/wired;
4.) Add the used configuration and firmware version;
5.) Create a `diff` and post it here in a code block. Put \`\`\` (three backticks) at the start and end of the `diff` block; and
6.) Remove this Text :).

alanfj, 2017-10-10:
How do I setup Lost Buzzer on a CC3D w/ Betaflight 3.17?

AndersHoglund, 2017-10-10:
By reading the WiKi: https://github.com/betaflight/betaflight/wiki/Board---CC3D
Also, please follow the quide lines that is added in your own first post, especially last item no 6.

closed 2017-10-10 by AndersHoglund