betaflight/betaflight #4326: Connect Failure After Flashing Firmware

GladiatorDev, 2017-10-10:
I believe this is a software issue with BetaFlight 3.2.1. Before this version we never experienced this problem. We have two completely different quads. After flashing the latest stable firmware on both, BetaFlight 3.2.1 will tells us that it successfully connects to the quad then immediately it says that it successfully closes the serial port - disconnecting it. The exact same problem was reproducible on both Windows 10 and the latest MacOS. Is this a bug in 3.2.1?
Thanks!

mikeller, 2017-10-11:
Betaflight 3.2.1 (now 3.3.0) is under active development. As such it can be unstable / nonfunctional at times. Please use a released version.

closed 2017-10-11 by mikeller

GladiatorDev, 2017-10-11:
Thanks for the info!
Betaflight 3.2.1 is the version on the Chrome app store. 
How do I get Chrome to let me use the previous version of BetaFlight?

mikeller, 2017-10-11:
Ah. That's the *Betaflight configurator* and not *Betaflight* (as in the firmware).

Can you please fill in the details that were requested from you when you opened this issue?

GladiatorDev, 2017-10-11:
Sorry, I didn't clarify that it was the Configurator.
No details were requested from me.

mikeller, 2017-10-11:
[image]

Yes, still works.

GladiatorDev, 2017-10-11:
Sorry I missed that. I will provide those details.

GladiatorDev, 2017-10-11:
The following bug is in the BetaFlight Configurator version 3.2.1. We don't believe this is a hardware bug because we can reproduce it with two different quads. And we can reproduce the bug on both Mac (latest version) and Windows (Windows 10).
On both we flashed the firmware using BetaFlight Configurator 
Before we flashed the firmware, the configurator worked for configuring the quads no problem.
On one:
Spracingf3/3.2.0 spracingf3 latest stable
On the other:
Spracingf3evo/3.2.0 Spracingf3evo latest stable
The other settings like Manual baud rate didn't appear to make a difference as we tried different combinations.
After flashing the firmware, we unplugged everything (using a data micro USB to the quad) and restarted the application.
On opening it, and selecting Connect. Either the serial port would fail to open or it would open "successfully" then immediately close. 
Therefore, we cannot use the application to configure the quads and as a result we cannot fly them. 
One of the quads was a Eachine FB90, and the other was the Eachine E010S. 
We never hooked up a controller.

mikeller, 2017-10-11:
Ok, this could be caused by an incorrectly migrated configuration. Can you try to reflash them, with 'Full chip erase' enabled? (If this does not work as is, read the instructions for fixing a failed update on the firmware flasher tab, and follow them.)

GladiatorDev, 2017-10-11:
For the "instructions for fixing a failed update", I assume you're referring to the instructions that appear below the "Recovery / Lost Communication" section on the "Firmware Flasher" screen. I will try your suggestion. Thanks!

mikeller, 2017-10-11:
Yes that's the ones.

GladiatorDev, 2017-10-12:
OK. I talked to my buddy and he told me that we did do the Erase Chip option when we flashed the firmware. Looks like now we cannot even flash the firmware on either Eachine. The program says it fails. We tried all this with BetaFlight and TestFlight and we get the exact same results. We can no longer flash or load to configure either machine. These Eachines were supposed to be BNF. 
Are there any steps I can take to verify without question that the problem lies in the hardware and not the software so we can contact the makers (in China) that the problem lies with their hardware and not the software? I am a programmer. Would I be able to verify the problem is in the hardware by getting the source code and debugging the problem or do you have a easier way I can do this?
Thank you for your help!

AndersHoglund, 2017-10-12:
You can always re-flash an FC, unless you have damaged the hardware.
Bridge boot pads or press boot button when connecting USB, flash with "No Reboot Seq" and "Full Chip Erase" selected. 
All you describe are typical symptoms of mixing up targets, flashing the wrong type. I.e. Flashing SPRF3EVO on SPRF3 and visa vers.

GladiatorDev, 2017-10-12:
We did what you suggested in your second line. My buddy is pretty adamant that we are flashing the correct type. So it might be the hardware. Any suggestions how we might prove its the hardware for when we contact the makers of this quad? 
Thanks again for your help!!!

AndersHoglund, 2017-10-12:
Make sure you are using USB2, not USB3. There are a few reports on troubles with this.
None of this is betaflight related issues, please use RCG, FB or some other support fora.

GladiatorDev, 2017-10-12:
OK. Thanks again.

GladiatorDev, 2017-10-12:
Just a couple more questions (somewhat) related to the software...
FYI: we are flashing the firmware type specified by the manufacturer (assuming their info is correct).

1. Do you know if these boards suddenly develop hardware issues because we’ve flashed the firmware to them? 
2. And is there a certain number of times you can flash firmware to a board before it develops hardware problems?

If you wouldn't mind answering these last two questions, I'd really appreciate it.

GladiatorDev, 2017-10-12:
And to tie in to something you said, if we did flash the incorrect firmware using BetaFlight, can that permanently damage a board?

AndersHoglund, 2017-10-12:
1. Yes, I do know. And, No it will not develop issues. 
2. A couple of 100k times or more.
And your third Q is really the same as no 1.

Assume nothing, make sure. Do not trust manufacturer info, often misleading and even completely wrong. As you can not brake an FC by flashing, try flashing both types, EVO and non-EVO.

And again, this forum is for discussing Betaflight Firmware development issues. Not for end user support.  Please respect that this issue was once closed for this reason.

GladiatorDev, 2017-10-12:
Thanks again. And I apologize for posting to the wrong forum. However, as a developer, your insights have been extremely helpful in case I ever wish to dive into development on this platform.

mikeller, 2017-10-12:
@GladiatorDev:

> Do you know if these boards suddenly develop hardware issues because we’ve flashed the firmware to them?

One caveat to this is that it is possible for the board to be damaged by flashing the _wrong_  firmware. Let's assume that an MCU pin on your board is used as an input, and fed 0V by an external component. Now you flash the wrong target, and on this target this pin is an output, and the MCU feeds 3.3V to it, resulting in a short. Not sure how much of this the STM32 MCUs can handle, and even less for any external components involved in the short.