betaflight/betaflight #4327: Feature request: GPS flight modes

SlipstreamFPV, 2017-10-10:
Now that long range is becoming a lot more popular and inav flight modes are more mature it would be a great time to bring GPS flight modes such as RTH across into betaflight.

mikeller, 2017-10-11:
The clone of cleanflight that is specialising in advanced navigation is iNav: http://inavflight.com/

Considering how much we are already restricted by the available flash space on F3, adding advanced navigation to Betaflight is not realistic, and is also not the goal of this project.

SlipstreamFPV, 2017-10-11:
I'm aware of inav, I mentioned it in my post. I think with time this will become a feature that a lot of people will look for in flight control software. It's unrealistic to dismiss it entirely based on current F3 boards limitations and current goals of the project. I think betaflight is way beyond what Boris had intended when he first set out to improve cleanflight for acro flight.

digitalentity, 2017-10-11:
I doubt INAV's navigation stack is easily portable to Betaflight. Long range was no (and probably still is not) a high priority for Betaflight. Betaflight's main goal is racing, freestyle, ACRO performance and low loop times. It doesn't pay much attention to accelerometer, which is a strict requirement for INAV's flight stack. 

I second @mikeller - it's not realistic to combine Betaflight's superior ACRO performance and INAV's navigation capabilities and keep them both at high performance. The result will be either poor navigation performance or noticable drop in ACRO performance. Betaflight is a de-facto standard for drone racing, I think we shouldn't compromise it.

However, INAV for most users will fly on par with Betaflight in ACRO mode, only experienced pilots will notice the difference and even for them it will probably be quite small.

fiveangle, 2017-10-11:
Maybe we could also incorporate a sky-writing feature that spells out the craft name ?  Or maybe a "sky hook" function for rooftop grapling hook quad retrieval ?  And since there's "anti-turtle" mode, why not "anti-anti-dolphin" mode that locate tuna netting operations via GPS coordinates and tries to cut them loose via kamakazee sabatoge right into the ocean ? *[head spins]* 😉

McGiverGim, 2017-10-11:
I don't know why every time some ask some functionality with the GPS the only solution is iNav and some people enter in "ironic mode on".

BetaFlight has GPS Hold and RTH functionality, that in first versions worked more or less. They were not perfect, but could help in emergency situations. But with the modifications and aditions in newest versions by some reason this functionalities are broken and don't work, so it can be considered a bug.

Another thing is that developers can spend the time where they want, and if they don't want spend time in GPS Hold and RTH, then the bug will stay. No problem about that.

mikeller, 2017-10-11:
@McGiverGim: I think the irony stems from a steady stream of people suggesting that Betaflight should essentially implement all of the things that iNav already does, which gets a bit tedious. One reason that Betaflight does the things it does is because that is what the developers want to focus on. Starting to do other things that are outside of this scope will just mean that there is less time spent on making sure the things that we already do work as well as they can.
In my view, having multiple projects that share a common architecture, with each one having their own focus area is preferrable to having only one project that 'does it all'. I think the healthy exchange of ideas and features between Betaflight and iNav shows that this is working well.

fiveangle, 2017-10-11:
But in all seriousness, it would probably be more fruitful campaigning with the iNav dev team to implement whatever few pieces from BF that it is missing (if any).  I have implemented iNav on my LRS quad for the same RTH need and I find it flies identically to BF when doing local FPV (with my hobbyist pilot skills).  It doesn't have the absolute latest flight mode <strike>improvements</strike>features, but they always get ported in eventually, especially after being well-vetted.

mikeller, 2017-10-11:
@fiveangle:

> But in all seriousness, it would probably be more fruitful campaigning with the iNav dev team to implement whatever few pieces from BF that it is missing (if any). 

It has been done before. The iNav team's stance is pretty much the same as the mood here: They don't feel that things like Dshot will improve the things that they want to focus on, so they see them as distractions more than enhancements.

Jahnkeanater, 2017-10-12:
I completely agree that if you want GPS you should use iNav not betaflight. I feel like cleanflight might be a good in-between if you want GPS and flight performance.

The issue is that adding GPS to betaflight will impede flight performance. Good GPS requires a lot of computation that uses up a lot of cpu time. You cant have 8k loop times and GPS at the same time. If they ever fix GPS I would expect it to be very limited. No waypoint or position hold just RTH.

Since 3.2 stable just came out the developers are busy fixing bugs. This is a pretty low priority.

mikeller, 2017-10-12:
There **is** GPS support in Betaflight (for >= F4). But it is limited to simple GPS readout. (Still a valuable feature if you ever have to go looking for your craft in the undergrowth.)

From personal experience and talking to iNav guys, implementing a reliable position hold (which is kind of the precondition for anything more advanced like RTH) on a quad is actually harder than doing the same on a fixed wing aircraft, since you can not use a motion vector to infer orientation, and have to rely solely on the MAG to get this information. Even in iNav, position hold for multirotors is only experimental.

Since MAGs are notoriously sensitive to proximity to high current, high electrical noise, this will be even worse if one tried to do the same on a craft running high performance ESCs with high frequency switching...

DzikuVx, 2017-10-12:
@mikeller 

> Even in iNav, position hold for multirotors is only experimental.

Oh c'mon, works like a charm for more than a year now.

digitalentity, 2017-10-12:
>Oh c'mon, works like a charm for more than a year now.

Not for everybody and not out of the box though, but non-working cases appear to be caused by either hardware or setup issues.

AndersHoglund, 2017-10-12:
Should we not remove all GPS, Baro and Mag support completely from Betaflight? As we are not supporting it and only recommend users to go iNav anyhow. We could use the memory space for other things. Are there any good reasons for keeping it?

digitalentity, 2017-10-12:
GPS coordinates in OSD are useful when searching for a lost quad. Baro altitude in OSD is a good proof that you didn't violate allowed altitude. My vote is that baro and GPS are needed in Betaflight, but solely for informative purposes.

mikeller, 2017-10-12:
> Not for everybody and not out of the box though, but non-working cases appear to be caused by either hardware or setup issues.

I have issues. :cry: 

Probably I would just have to spend more time on iNav, instead of going back to testing Betaflight after the first failure with iNav. :wink: 

Also, using a heavy, sluggish hex to start with iNav probably wasn't the world's best decision. This thing  falls out of the sky like a rock as soon as the position hold loses its sh*t and commands it into a hard bank to one side.

NitroNAF, 2017-12-28:
I agree with OP. I have been experimenting with iNav and coming from betaflight the system feels more geared toward wings. All I, and many other pilots want, is Return to Home. Thats it. Nothing more. 

The hoops I have had to jump through just to attain an RTH iNav multi is rediculous. I tried to change over to iNav, while fantastic on my wings, its a plain joke on a multi. The flying experience is rubbish and tuning it is too thanx to its PID system and Rate system being so inferior to BF. 

Hours on hours of set up just to get a poor flying multi with RTH. 

Ended up going back to Betaflight and ditching the idea. 

Really wish BF had RTH.

CircuitBender, 2017-12-29:
I can t understand why nobody came to the idea to take betaflight, integrate the nice features of inav (maybe improve them) and while we are at it to merge all mission planer functions and a pipe to python (for rasp projects) all into one professional thing.
a setup wizard guides you through the features you want , for the fpv quad pilots a similar setup to Betaflight now, a setup for noobs, a setup for wings, and planes and hexacopters and one for all the exotic stuff.

After testing CL PF inav and Mission planer I am sure that BF is the most advanced and rigid software, the others have nice features which should be implemented into BF

CircuitBender, 2017-12-29:
if you take all the programmers hours from BF CL inav and mission planner and add them up and then think about what could have be done pulling all on one string -
we would be already at a software outplaying a lot of professional competitors

fiveangle, 2017-12-29:
I suspect you've never worked on a large hardware / software development project ?

von Karmen called this the _Pregnant Woman_ or _Too Many Cooks_ fallacy.  This is believing that if one woman can make a baby in 9 months, 9 can make one in 1 month.  Believing that just throwing more bodies into a larger group working on a single project will result in a sum-is-greater-than-the-parts scenario is wishful thinking, but it just doesn't work like that in practice.

This thinking isn't uncommon, and it is a constant struggle to prevent from falling into the trap of team consolidation and scope expansion -- so much so, that it's described many times throughout history under various titles, including _Second-system effect_:
https://en.wikipedia.org/wiki/Second-system_effect

And at extremes, _Software bloat effect_:
https://en.wikipedia.org/wiki/Software_bloat

Nowhere is this a greater issue than for embedded systems such as Betaflight, where the feature support must be painstakingly balanced against relatively scarce resource constraints of the most commonly available target hardware.

And this doesn't even get into the very punctuated differences between corporate software development and open-source software development, which adds the factor of _volunteer will_: nobody wants to work on a project where they are constantly at odds with others who envision a different direction for the project.

Happy New Year ! 😋 

-=dave

ElectricSpectre, 2018-02-25:
I am glad these posts aren't too old as I have a little bit of a horse in this race as far as some drones I am building.  I like using BetaFlight, but recently have built a larger, long range drone with an hour flight time.  I want to be able to tune and fly it as I do with my freestyle quads and fly it in altitude locked angle mode, or in acro mode as I feel like it.  RTH is a nice feature as well.  I have GPS Hold and RTH set up on BetaFlight, although haven't tested them as I have heard they don't work well and am nervous.  Looking for alternate options I stumbled across this and checked out iNav.  Being able to set check points would be nice as well.

Now, I have education dealing more with hardware then software so this may sound really stupid, but why couldn't you combine BetaFlight and iNav, or the features of them together into a package that can do both, call it OmniFlight or something?  I realize that hardware has its limitations (Trust me, I know the pain of being asked to make something by ignorant people) but if they are both based off CleanFlight, how hard would it be to combine the features of both together, and have it configurable in a menu that if you are using an F3 or F4 board, you have to choose to use more BetaFlight features, or more iNav features based on what you want to do, but if F7 boards which are becoming more prevalent are capable of using the features from both, having it allow you to use features from both, if an F7 board is detected?  I guess the question I should really be asking is would an F7 board be capable of running all the features of both?  I don't have this level of programming skill or knowledge to be able to tell if an F7 would have the processing power, memory etc that would be needed to run all of this.  From the sounds of it calculations from GPS take up more processing power then I would have thought, but my only experience with GPS is making an LCD read out the coordinates from a GPS breakout board using a microcontroller.

I hope to hear whether this could be possible, or hear why its not.  I figure **IF** F7 boards have the capability of running everything then why not combine the features from several software packages into one place, especially since F7 boards are becoming more prevalent, it would make sense to move in a direction to get the most out of them.

Or maybe you can tell me I am just a weird asshole that wants a quadcopter that is both a DJI Phantom 4 and a Racing Drone in one, but not as good at either task as the dedicated thing.

mikeller, 2018-02-25:
@ElectricSpectre: See https://github.com/betaflight/betaflight/issues/4327#issuecomment-354490557. Couldn't have been summed up any better.

KnuckleUpFPV, 2018-05-01:
Finding this old request is kind of funny. Is betaflight not currently working on a return to home with no mag needed? From what I understand its in the works. I guess the flood of long range and demand of the people have changed the go to inav attitude? Inav doesnt have dshot or fly as well. If it did, then everyone would just use inav and betaflight would have 0 reason to exist. I hate that I cannot remap in inav. Without building custom firmware you are limited to what you can do with ports or if you damage a soldering pad. I hope the attitude truly is changing from telling anyone that ask about gps to go to inav. We use inav on our wings because its great for wings. We fly betaflight on our quads because it flys amazing.  I do believe there are a ton of us in the community that would love a basic rth system with the benefit of betaflights wonderful dshot and filtering. We want tp be able to do proximity and then blast off out to a mile or more and be able to save our rear ends if we lose video or signal.

mikeller, 2018-05-02:
@KnuckleUpFPV: I think you are misunderstanding this. What's changed is that somebody actually stood up and implemented a GPS mode that is suitable for Betaflight in Betaflight, instead of just demanding 'I want all of the GPS features that iNav has in Betaflight'.

KnuckleUpFPV, 2018-05-02:
I dont see lots of demands. I see lots of request. In the place to do request.  Good for Diego for helping the community with their request and not pushing them to inav.

mikeller, 2018-05-03:
@KnuckleUpFPV: Call it however you want. The reality is that 'GPS flight modes such as RTH' will not be coming to Betaflight any time soon, and users who want them will have to go to iNav.

Closing this.

closed 2018-05-03 by mikeller