betaflight/betaflight #4328: Average cell voltage in OSD

OtterNas3, 2017-10-10:
Since BF3.2 the voltage for the average Cell that shows up in the OSD seems to be off sometimes.

mikeller, 2017-10-11:
- calibrate your voltage sensor;
- if this does not help, change your maximum cell voltage up a bit to accommodate for the higher than usual charge level

GitHub is for questions and discussions around the development of Betaflight. For configuration questions please see [RCGroups](https://www.rcgroups.com/forums/search.php?searchid=62616043&query=betaflight%20voltage%20calibration).

closed 2017-10-11 by mikeller

OtterNas3, 2017-10-11:
Was not meant as a configuration question, but cool

mikeller, 2017-10-11:
Well the problem is that your craft is misconfigured, so it is a configuration problem. ;-)

FasaArg, 2017-10-15:
I have a similar problem, (with my battery alarm), a fully charged 4 cell lipo battery give me low voltage alarm.

Searching the CLI i seeing this:

vbat_detect_cell_voltage = 30
Allowed range: 0 - 200

16.8 / 3 = 5.6 

If I discharge the battery a little and replug the system everything goes ok.

This happened today so i can´t try anithing elese.... this value should not be a little higher? like at least 3.4 or 3.5.

Thanks in advance.
Sorry about my poor english.

mikeller, 2017-10-15:
@FasaArg: Increase `vbat_max_cell_voltage`.

FasaArg, 2017-10-15:
@mikeller I did that yesterday on the field, but it did not work. (i was not able to do more test)

It was on 4.1 and then i raised it to 4.2v (same result)

My vbat range is accurate.

I will do further test tomorrow on the bench and i will report back.

Thanks.

FasaArg, 2017-10-15:
@mikeller thanks, i'm really sorry, on 3.1.7 vbat_full_cell_voltage was 42 and now on 3.2.0 is 43. 

I didn't realize this and I ported this from 3.1.7
Now is working fine with the original value of 43.

I apologize again.
Regards.