betaflight/betaflight #4334: VTX power auto change with arming

Gaz-B, 2017-10-11:
With smart audio ETC would it be possible to have a selection of two power levels IE low power on the ground and higher when armed it could even have a selectable delay to get away from flight line before it ups the output. I mainly want it to stop over heating of the VTX on the ground and bench.

Vidalcris, 2017-10-11:
Awesome idea, the delay after arming too ! :)

jflyper, 2017-10-11:
I think it's a duplicate...

Gaz-B, 2017-10-11:
If it is a duplicate is it already being developed ? I did search but didn't find anything.

jflyper, 2017-10-12:
There is a long list of feature requests, scattered around multiple issue tickets 😭

Jahnkeanater, 2017-10-18:
I like this idea. Could we just add VTX Full power to the modes tab. That way people can easily enable this feature if they want it.

It would be nice if we could still increase the VTX power without arming the quad to help find it when it crashes in a field. Or even just have a separate switch for setting the VTX power from high to low.

The user can customize the behavior easily this way.

Using the osd to change the VTX power should update the variable for the high power setting. 25mw should be the default for the low setting.

Vidalcris, 2017-10-20:
I hope someone is interested working on this ..
I'm trying to do simple modification on my side but i'm not even able to change the power ... this part is pretty hard for a noob like me XD

Is there a way to change the power with just one command or not ?

lostcontrol, 2017-10-30:
I killed the video of a buddy this week-end as I landed next to me and forgot about him sitting not far away. Terrible crash, GoPro Session 5 dead, 2 broken arms, etc :sob: Besides me being a moron, I think this feature could have changed the ending of that event. I heard that KISS has something like that already.

I would really couple this with arming/disarming. Probably a bad idea to change settings while flying. Instead of adding a mode for that, we could add a "disarmed" VTX power settings (set to the "armed" or normal VTX power settings per default, meaning same power).

@codecae @ethomas997 @AndersHoglund I've seen you are actively working on the VTX code at the moment. Any thoughts on this feature? Any time to invest on this? If not, I might give it a try. My knowledge of the code is very limited but I could see if I can manage to get something working. I should receive a AKK X2 soon which support SA.

lostcontrol, 2017-10-31:
@mikeller why are you confused if I may ask?

mikeller, 2017-10-31:
@lostcontrol: Sorry, that's not meant to be 'confused', but 'sad face' (which isn't available in the 'comment reaction').

lostcontrol, 2017-11-01:
@mikeller mouauahahaha 🤣 even worst... you don't like this feature? why?

mikeller, 2017-11-01:
Nooo, you totally misunderstand me. The sad face was because of the crash. The VTX power change seems like a reasonable feature request to me - after all I added the 'Feature Request' label.

lostcontrol, 2017-11-01:
ah oh sorry, I got it now!!! Waiting for more comments about this feature from people working now on that part of the code.

codecae, 2017-11-01:
Couldn't this be more easily implemented using lua?

codecae, 2017-11-01:
I guess not everyone uses a lua compatible transmitter, though.

codecae, 2017-11-01:
The most recent update to PR #4455 uses a scheduler in the task process function.  A step could be added to the power update step that could evaluate the armed state.  Only problem is that I don't think the task process function is called when armed.  Easy to fix that, though.

lostcontrol, 2017-11-01:
@codecae thanks for the reply. No, I don't think I have seen the scheduler change. And damn, that might make this feature request even more easy to implement. I had done it the other way around, setting the power from tryArm() and disarm() but actually checking the arm state in the task should be much simpler. Nice idea!!! Indeed the VTX control task is not scheduled when armed but as you said that should not be too hard to fix. Will cherry-pick your latest PR and give it a try except if you want to do it yourself. Would be more than happy to contribute a bit to ßF 😀

codecae, 2017-11-01:
Not sure that creating a feature dependency on an unapproved PR would get much traction, as this feature should probably be a separate PR altogether.  Never hurts to play around, though.

lostcontrol, 2017-11-01:
An initial attempt at addressing this feature can be found here: https://github.com/lostcontrol/betaflight/commit/c83a23a0164ffd51e1734a78a7d1539c79f3d4e4

It is not tested at all. I should receive my AKK X2 with SmartAudio this week so I should be able to give it a try soon. Any comments are welcome.

lostcontrol, 2017-11-02:
@codecae proposed the `vtx_low_power_disarm` CLI option in https://github.com/codecae/betaflight/commit/79027001d066f78c6b0a8e50e313db0d11346d51

I did a `vtx_low_power` and `vtx_high_power` attempt with both value settable via CMS which is AFAIK more or less similar to what KISS does. https://github.com/lostcontrol/betaflight/commits/vtx_power_on_arm

@mikeller @codecae How do we go further with this? Would be good if we could agree on the user facing changes. Also @codecae do you want to work on this or should I do it? Nothing worst than investing time and seeing your PR rejected or already implemented 😉

mikeller, 2017-11-02:
I have not yet converted to using externally adjustable VTX, so I am not in a position to judge what level of adjustability is the right level here. ;-)

In general, I think these refinements of VTX control should be addressed after @codecae's #4455 has been completed, otherwise at least one party will be attempting to shoot a moving target, so to speak. This also leaves us with plenty of time to discuss and decide who's proposal is the one that should be implemented.

codecae, 2017-11-02:
Completely agree.. cart is firmly before the horse.

lostcontrol, 2017-11-02:
Yes, #4455 is clearly a must. Still, the feature can already be discussed. So people, if you have ideas or wishes, feedback welcome.

I received my AKK X2 today and just finished the installation on my main quad. Smartaudio seems to work fine :smiley: By the way, as I have also seen on some videos, there is a video break up when switching power so definitely not something you want to do during flight.

lostcontrol, 2017-11-05:
@jflyper I guess the duplicate you were referring to is #3094

Vidalcris, 2017-11-05:
Oh i also found this issue 30min ago XD

codecae, 2017-11-10:
removed that one and reissued it with a new "brand" and a few adjustments 
#4529

codecae, 2017-11-10:
This feature has been merged into master if you'd like to take a look and try it out!

For now it's enabled by using 'set vtx_low_power_disarm=ON' in CLI.

Vidalcris, 2017-11-11:
Thanks a lot @codecae i will try it for sure :)

Vidalcris, 2017-11-11:
I just tried, this is not working for me :-/
The VTX power stay the same.

"osd_vtx_channel_pos = 2072
Allowed range: 0 - 3071

vtx_band = 4
Allowed range: 0 - 5

vtx_channel = 1
Allowed range: 1 - 8

vtx_power = 5
Allowed range: 0 - 5

vtx_low_power_disarm = ON
Allowed values: OFF, ON

vtx_freq = 5740
Allowed range: 0 - 5999

vtx_halfduplex = ON
Allowed values: OFF, ON"

Vidalcris, 2017-11-11:
@codecae With the fix #4542 this is not working but the smartaudio is working and i can change the channel.
But the power is stuck at max power and arming or disarming do nothing.

At first boot the vtx as been reset to channel 1-1 and power was 25mw, when i armed the first time the power as been switch automatically to max.
But the VTX was screen was freeze at this moment. from here the power stay stuck to max i cannot change it anymore.

Vidalcris, 2017-11-11:
After #4532 this is pretty the same
At first boot it was at 25mW.
at first arm the VTX screen freeze again to Max power, then after disarm stay at Max power.

Now if i change the power in the OSD nothing happen but if i reboot then again at first arm the VTX freeze with the new power i just set. and again when disarmed the power stay at the same level.

Vidalcris, 2017-11-11:
@codecae @jflyper are my comments usefull ?
I can see that you are doing modification but could you just to say me if i should continue to make report ? ;)

codecae, 2017-11-11:
@Vidalcris  the first boot setting to F1 25mw is normal.  This is part of the parameter group change.  Once you change it, the parameter group will update and settings will persist going forward.

I'm looking into the mw issue being stuck... wondering if it's just the cms string no updating.

Vidalcris, 2017-11-11:
ok thanks i was going to reflash to be sure ;)
When that will work the power setting in OSD will not change the power until arming right ?
Do i have to save when i change the power now ?

codecae, 2017-11-11:
I've found that if you 'set vtx_low_power_disarm=OFF' and return to the CMS screen, it will update the string when setting a new power value.  For now, I would leave the feature off until I can figure out what's going on.

Vidalcris, 2017-11-11:
Ok, but anyway i think i will not fly before this work ;)
This feature is vital for this particular build :p

Vidalcris, 2017-11-11:
Oh just to be sure, when i said that the power was stuck i was talking about the VTX itself.
I was not looking at the cms string.

codecae, 2017-11-11:
Yeah I see the same thing.  If you set vtx_low_power_disarm = OFF it will no longer get 'stuck'

Vidalcris, 2017-11-11:
Yeah sure i could also flash 3.2.1 again ;)
Your are working on this and i cannot say you how much i'm happy because i was waiting this feature !!
Thanks a lot for your work !

codecae, 2017-11-11:
I'm confident it will work fine with the setting turned off, but do whatever you'd like :)

I found the issue but would like to modify a few things before submitting the PR.

My pleasure!

Vidalcris, 2017-11-11:
Oh sorry i missunderstood then XD
You mean that with the feature off the feature will work ? :p

codecae, 2017-11-11:
The suggestion is to disable the feature until I send the PR so the power does not get stuck.  No reason to flash a different version to restore the original functionality.

Vidalcris, 2017-11-11:
Ok thanks.
As i said i will not fly anyway dont worry :p

codecae, 2017-11-12:
Created PR #4555 to correct issues with this feature.  This change should also provide visual feedback of the power change by means of the OSD vtx channel widget.

nigelsheffield, 2017-11-28:
I'm thinking the power levels selectable via a spare channel so could be put on a switch with arming, or even better could switched via a logical switch in open tx based on rssi levels.

mikeller, 2017-11-28:
@nigelsheffield: A comment is not a good place to ask for a new feature. Please open an issue for it, and we can track it as a feature request.

anti-vaxxer, 2025-03-20:
I think this is solved with `vtx_low_power_disarm` in #4529

haslinghuis, 2025-03-20:
Resolved in #4529

closed 2025-03-20 by haslinghuis