betaflight/betaflight #4335: Dshot buzzer don't work after updating.

Bugg69s, 2017-10-11:
I updated to BF3.2 and my dshot buzzer don't work any more

Bugg69s, 2017-10-11:
Omnibus f3

tnuki, 2017-10-11:
I have the same issue on an omnibusf4, but its both the ESC buzzer and regular buzzer that dont work

JyeSmith, 2017-10-11:
Did you set it using below? By default it is off.

set beeper_dshot_beacon_tone = [0 - 5]

tnuki, 2017-10-11:
I just changing it from 0 to 5, but didnt seem to change anything

fiveangle, 2017-10-11:
5 is contextual depending on _S (duplicate of tone 4) or _32 (extremely low frequency - potentially inaudible ?) so depending on your ESC, behavior may be expected.

What if set to 3 ?

Bugg69s, 2017-10-12:
But turtle mode is working

tnuki, 2017-10-12:
Turtle mode is working, and changing it to 3 didnt help. What did help is re-enabling the USB Beeper option in the config tab.

Bugg69s, 2017-10-12:
I have more problems with stable BF 3.2 than with 3.2 RC5 :/

tnuki, 2017-10-12:
Tell me about it

mikeller, 2017-10-12:
@tnuki: As I already tried to explain to you in #4344, ytour problem in 3.2.0 is that voltage detection does not work with your configuration. The rest of it, like beeper not working is just a consequence of the voltage detection not working.

st0rm4g3dd0n, 2017-10-18:
I am having this issue as well. Using cl racing f4 board, wraith_32 escs, and dshot1200 on betaflight 3.2.1. Set the tone as above in cli to several different values [1-5] with no success.

Before updating to 3.2.1 from 3.2rc6 I could not get turtle mode to work either, but that is working now. So I know that dshot commands are working.

[BTFL_cli_KD8KIF_20171018_150114.txt](https://github.com/betaflight/betaflight/files/1395969/BTFL_cli_KD8KIF_20171018_150114.txt)

Also, I had USB beep disabled but just enabled it after seeing above comment and it does not seem to work anymore either, nor did it fix the esc beeper.

brycedjohnson, 2017-10-18:
@st0rm4g3dd0n You are disabling the beeper here:

beeper -RX_SET for the switch
beeper -RX_LOST for no tx beeper

st0rm4g3dd0n, 2017-10-22:
Based on the description in the gui config for the beeper I didn't think those were related. I didn't read the one for RX_SET closely enough, just saw the gps part not the first. Thanks!

RX_LOST should make it beep if you turn your transmitter off, correct? The switch is working now, but this will not. Am I missing something else for that?

mikeller, 2017-10-22:
@st0rm4g3dd0n: Have you read and verified point 1 of https://github.com/cleanflight/cleanflight/blob/master/docs/Failsafe.md#configuration ?

(If not, there's much more at stake than just a beeper / beacon not sounding.)

st0rm4g3dd0n, 2017-10-23:
No, but it worked when I actually went out and flew. I may have not had a battery connected, or it was because it was still connected via usb also. I will definitely go read through that though, looks much more in depth than my understanding was already. 

Is there any way to choose whether the beeper uses the one on my fc, or dshot commands to the motors?

mikeller, 2017-10-23:
> I will definitely go read through that though, looks much more in depth than my understanding was already.

Yes please. The alternative to having a properly set up failsafe is to watch your craft fly away towards the horizon (or a bystander) in case of a transmitter malfunction.

> Is there any way to choose whether the beeper uses the one on my fc, or dshot commands to the motors?

Currently not, but it's being worked on.

st0rm4g3dd0n, 2017-10-23:
Alright. Thanks!

SomeguyinPDX, 2017-11-07:
Sorry to butt in on your thread, but I'm having the same issues.  I cannot get the motor beepers or turtle mode to spin in reverse when upside down.   The motors make the normal startup tones on power-up but no sound when assigned to a switch even after updating to the latest BF & blheli32 firmware.  I did set the CLI setting for (beeper_dshot_beacon_tone = 1).  When I activate the beeper switch I can see the led on the board flashing but motors are silent.  I'm sure I've got something wrong somewhere but no matter what I try I keep getting the same results.  Any suggestions?

[BTFL_cli_Purple_Haze_20171106_172614 Nov 6.txt](https://github.com/betaflight/betaflight/files/1450968/BTFL_cli_Purple_Haze_20171106_172614.Nov.6.txt)

brycedjohnson, 2017-11-07:
Check what version of blheli32 you loaded.  If they both aren't working
then it is a blheli32 issue.

SomeguyinPDX, 2017-11-07:
It shows 32.1 in Blheli32 configurator.   Just a few minutes ago while fiddling with this I heard the motors beep.  I had set my beacon timer from 10min to 1:00 in the blheli32 and while powered up on the bench un-armed w/transmitter on & bound, I heard the motors beep after the minute was up?  I also tested the failsafe and the motors beeped 1 min after motor shutdown w/transmitter off. 

[BTFL_cli_Purple_Haze_20171106_172614 Nov 6.txt](https://github.com/betaflight/betaflight/files/1451619/BTFL_cli_Purple_Haze_20171106_172614.Nov.6.txt)
.

mikeller, 2017-11-07:
@SomeguyinPDX: Yeah, that won't work. The first BLHeli32 version to support Dshot commands was 32.12. You'll need to update the firmware. The beeps you heard could just have been the init beeps because the ESCs were resetting.

SomeguyinPDX, 2017-11-07:
BLHELI32.com showed 32.1 as the latest firmware available for my ESC's?  I will look again.  Thanks.

SomeguyinPDX, 2017-11-07:
could you please post a link or tell me where I can get that firmware?  Blheli32.com's website shows 32.1 is the firmware for the Racerstar Tattu 35A 4in1?   Thank you for the help.

mikeller, 2017-11-07:
@SomeguyinPDX: No I can't. You'll have to contact the manufacturer / supplier of your hardware to get them to make the updated firmware available to you.

SomeguyinPDX, 2017-11-07:
Ok, Thanks.  So who or where is the official BLHELI32 site?

mikeller, 2017-11-07:
@SomeguyinPDX: https://blhelisuite.wordpress.com/

haslinghuis, 2022-05-08:
Closing as inactive

closed 2022-05-08 by haslinghuis