betaflight/betaflight #4336: Horizon moves to wrong direction - Omnibus F4 Hobbywing XRotor

MR444, 2017-10-11:
Hi, I activeted artifical horizon but it moves to wrong direction. So when I roll 45°left, artificial horizon is 90°. So it does not stay level. I am using betaflight 3.1.5 with Hobbywing XRotor Omnibus F4. Flying works well and copter directions are correct in the GUI.
[image]

AndersHoglund, 2017-10-11:
Right or wrong is a matter of definition. This is the Russian standard. https://www.skybrary.aero/index.php/Western_and_Russian_Artificial_Horizon_Formats
But we should have the Western std, I aggree.

MR444, 2017-10-11:
You are right it is just another definition. Now I remenber that it is how many (more old) russian airplanes have it. So it is not an issue but it looked strange for me. Will there be a possibility to change it?

DanNixon, 2017-10-11:
This has already been fixed (IIRC in version 3.1.7).

AndersHoglund, 2017-10-11:
Oooh, I did not notice 3.1.5. We can not accept any trouble reports on old versions unless they are verified to exist also in later/latest releases.

closed 2017-10-11 by AndersHoglund