betaflight/betaflight #4341: Mavlink Telemetry Not Outputting Data

davidbitton, 2017-10-11:
# If your issue looks like a hardware fault or a configuration problem please don't raise an issue here.

## Please consider using other user support options such as asking the manufacturer of the hardware you are using, [RCGroups](https://rcgroups.com/forums/showthread.php?t=2464844), [Slack](https://slack.betaflight.tech) or other user support forums & groups (e.g. facebook).

## If you believe there is an issue with the firmware itself please follow these steps:
1.) Describe your problem;
2.) Include ways to reproduce the problem;
3.) Provide information about your flightcontroller and other components including how they are connected/wired;
4.) Add the used configuration and firmware version;
5.) Create a `diff` and post it here in a code block. Put \`\`\` (three backticks) at the start and end of the `diff` block; and
6.) Remove this Text :).

I'm running 3.2.0 on an F3 Omnibus from RTFQ. I can get my board to output other telemetry formats such as FrSky and LTM, but when I switch to Mavlink, the board output is nill. I looked over the code and didn't see anything jump off the page. Are there addt'l setup steps or could this functionality be broken? Thanks.

mikeller, 2017-10-11:
A `diff` would help with diagnosing this. Please read the instructions.

davidbitton, 2017-10-11:
@mikeller diff of what though? just tell me what to look for. Also, I moved telem from UART1 to UART2, and SerialRX from UART2 to UART3 and I see the same behavior.

mikeller, 2017-10-11:
@davidbitton: Just post it here, so I (and others) get to look at it.

davidbitton, 2017-10-12:
A code diff?

davidbitton, 2017-10-12:
Also, 3.1.7 outputs Mavlink.

mikeller, 2017-10-12:
> A code diff?

https://oscarliang.com/use-diff-not-dump-betaflight/

davidbitton, 2017-10-12:
```
# diff
# Betaflight / OMNIBUS (OMNI) 3.2.0 Sep 28 2017 / 16:38:02 (ceb6f70) MSP API: 1.36

feature -RX_PPM
feature -OSD
feature RX_SERIAL
feature TELEMETRY
serial 1 512 115200 57600 57600 115200
serial 2 64 115200 57600 0 115200
set align_gyro = CW270
set align_acc = CW270
set serialrx_provider = SBUS
profile 0

rateprofile 0

#
```

diff as requested

jflyper, 2017-10-13:
Yes. I confirmed that the diff above works with 3.1.7, but not with 3.2.0.

jflyper, 2017-10-13:
And this is due to MAVLINK support dropped from OMNIBUS target, to yield some flash space.

mikeller, 2017-10-13:
Well spotted, @jflyper.

jflyper, 2017-10-13:
@davidbitton 
You can start building your own firmware, which is not too hard if your goal is to turn on and off various features. See Development section of the Betaflight wiki.

MAVLINK requires another 1K (999) bytes, which is not hard to find.
Take a look at the section here
https://github.com/betaflight/betaflight/blob/v3.2.0/src/main/target/common_fc_pre.h#L108-L135
and pick whatever you are not using, and `#undef` it in `target/OMNIBUS/target.h`.
The first and most obvious choice would be the duplication of `VTX_SMARTAUDIO` and `VTX_TRAMP`. 

If you are not familiar with the process, we can help you at the Betaflight slack.

closed 2017-10-13 by jflyper