betaflight/betaflight #4342: No voltage reading on Omnibus F4 since 3.2 update

tnuki, 2017-10-11:
I've been flying 3.2RC5 since it came out and everything worked fine (seemingly anyway). And I updated to the 3.2 release today on my Omnbus F4 V2 board, and everything seems to work, but I'm not getting any sort of voltage reading, both in betaflight and on my radio (telemetry is working except for the voltage)

///
$M>l:Õÿz
Entering CLI Mode, type 'exit' to return, or 'help'

# diff

# version
# Betaflight / OMNIBUSF4 (OBF4) 3.2.0 Sep 28 2017 / 16:38:25 (ceb6f70) MSP API: 1.36

# name
name Licorice

# resources

# mixer

# servo

# servo mix

# feature
feature TELEMETRY
feature RSSI_ADC
feature AIRMODE
feature ANTI_GRAVITY
feature DYNAMIC_FILTER

# beeper
beeper -ON_USB

# map
map TAER1234

# serial
serial 0 64 115200 57600 0 115200
serial 2 32 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 6 1700 2100
aux 1 1 3 900 1300
aux 2 2 3 1300 1700
aux 3 13 5 1700 2100

# adjrange

# rxrange

# vtx

# rxfail

# master
set gyro_notch1_hz = 0
set gyro_notch2_hz = 0
set baro_hardware = NONE
set min_check = 1045
set fpv_mix_degrees = 30
set serialrx_provider = SBUS
set motor_pwm_protocol = DSHOT600
set failsafe_delay = 10
set failsafe_throttle_low_delay = 10
set align_board_yaw = 90
set current_meter = NONE
set deadband = 5
set yaw_deadband = 5
set pid_process_denom = 1
set debug_mode = NOTCH

# profile
profile 0

set dterm_lowpass_type = PT1
set dterm_notch_hz = 0
set anti_gravity_gain = 3000
set dterm_setpoint_weight = 30
set crash_gthreshold = 600
set crash_time = 1000
set crash_recovery_angle = 0
set p_pitch = 52
set i_pitch = 60
set d_pitch = 13
set p_roll = 47
set i_roll = 55
set d_roll = 12
set p_yaw = 85
set i_yaw = 55
set level_limit = 75

# rateprofile
rateprofile 0

set rc_rate = 130
set rc_rate_yaw = 130
set rc_expo = 30
set rc_expo_yaw = 30
set roll_srate = 75
set pitch_srate = 75
set yaw_srate = 75
set tpa_rate = 25

# 

///

capjuancho89, 2017-10-11:
same happened to my lumenier f4 aio since rc3 no voltage reading, im wondering if here's a solution to this.

mikeller, 2017-10-11:
@tnuki: Can you please check that the 'Voltage sensor' dropdown in the battery tab shows 'Onboard ADC'.

tnuki, 2017-10-11:
@mikeller It does, it was the first thing I checked, I've also tried reflashing 3.2 a bunch of times but no luck, I'm using the 3.2.1 version of the configurator if that matters

tnuki, 2017-10-11:
Is there anyway to go back to rc5? that version was the last time everything worked

mikeller, 2017-10-11:
You can get it from here: https://ci.betaflight.tech/job/Betaflight/187/artifact/obj/

Or you could try to backup your config in the CLI, do a 'Reset to defaults', and restore your config to see if that helps.

tnuki, 2017-10-11:
I reset to default but no luck, I went back to RC5 (and heres the odd part) theres still no voltage monitoring or beeper. Its an Omnibus F4 and I have it hooked straight to the battery, so idk what the issue is if its not the firmware.

Arakon, 2017-10-12:
The voltage divider used to monitor voltage may have gotten physically damaged, or the port on the MCU burnt out for some reason.

tnuki, 2017-10-12:
Odd, could be, though I'm not sure how it wouldve gotten damaged, it worked fine last time I flew, and its just been sitting around for a couple days

jflyper, 2017-10-12:
@tnuki 
I'm back at my bench.

I can't reproduce the issue. Do you have a backup config for RC5?

tnuki, 2017-10-12:
I think I found the problem, maybe anyway. Looks like one of the 2 resistors next to the VBAT pin popped off at some point:

https://prnt.sc/gwjdas

jflyper, 2017-10-12:
@tnuki 
Yes. The R7 is the high-side resistor of the voltage divider.

tnuki, 2017-10-12:
Alright thank you, so it is a hardware problem, would it still be safe to use the FC as normal? just without voltage monitoring?

jflyper, 2017-10-12:
If that resistor is the only component that fell off, yes.