betaflight/betaflight #4344: Beeper issue on OmnibusF4

tnuki, 2017-10-12:
There seems to be an issue with the beeper on the latest 3.2 release with OmnibusF4 firmware, with the option to not beep while plugged in over USB is enabled, all beeping (both ESC and physical beeper) is turned off. It might just be me, if youre board is on the OmnibusF4 firmware I'd be interested to know if you turn off USB Beeping, can you still control the beeper from the radio?

mikeller, 2017-10-12:
Yes. But voltage detection needs to be configured and display the correct voltage, since USB detection is driven by the battery detection.

tnuki, 2017-10-12:
I have it Voltage meter source set to Onboard ADC, I'm not really sure how else to configure it, it just shows 0V with the lipo plugged in.

mikeller, 2017-10-12:
There you go. As long as the voltage shows as 0V, the flight controller 'thinks'  it's on USB...

tnuki, 2017-10-12:
Any chance you know how to fix it? Even without the USB plugged in, it still gets 0V on the OSD.

mikeller, 2017-10-12:
I think @jflyper is looking into this, but I can't find the ticket now. :-|

jflyper, 2017-10-12:
So, this one is solved, too?

Use/ignorance of VBUS sensing is another problem... I actually didn't know the beeper is estimating the VBUS presence 😭

mikeller, 2017-10-12:
@jflyper: The sad thing is, the firmware supports proper USB detection, but there is only about 2 targets that have the extra line from the USB port the an MCU pin that is required for it. As a consequence, the USB detection is not used for anything that is actually important (I think it only reduces the IR transmitter output). It's a total chicken and egg - if you suggest to a hardware manufacturer to add an USB detection pin, they tell you 'why, it's not used anyway'. :cry:

jflyper, 2017-10-13:
@mikeller Ah... okay, that explanation didn't come up in #3411 😉