betaflight/betaflight #4349: Feature Request: blackbox while disarmed

Zethyl, 2017-10-13:
I would like to formally request a feature to allow blackbox logging whilst disarmed.

My intention is the ability to measure the airframe resonance without needing to have anything but the FC mounted. Currently one needs to provide a valid RC signal to enable arming, in most cases that means a RC RX and some type of battery to power it. The less things touching the frame the better.

There are ways of doing this in PX4 and ArduPilot in the safety parameters.

I suggest making it enabled via a debug_mode

mikeller, 2017-10-13:
This is already supported. If you set `blackbox_on_motor_test` to `on`, blackbox logging will be enabled as soon as the motors are not at their minimum setting.

Zethyl, 2017-10-13:
Then I would need to be connected to USB. I am requesting the ability to
log without connection to USB while disarmed.

mikeller, 2017-10-13:
Ah I see. I assumed you were connected over USB anyway, since you stated above that you did not want to have a battery connected...

But nonetheless, you should be able to use `blackbox_on_motor_test`: Connect USB to your test set up, set the motors to 'run', then disconnect USB (without clicking 'Disconnect' first), and the logging will keep going.

closed 2017-10-26 by jflyper