betaflight/betaflight #4352: IMPULSERCF3 boards lock up after connecting GUI using v3.2.0

sblakemore, 2017-10-13:
When flashing 3.2.0 to an IMPULSERCF3 board and then connecting the GUI the board locks up within about 15 seconds. No more flashing LED and GUI stops responding.

Downloading and building the v3.2.0 tag with no changes to the code results in a hex file that does not have the same issue.

Perhaps a bad build on the releases page?

mikeller, 2017-10-13:
@sblakemore: Can you please confirm that this version is working:

[betaflight_3.2.0_IMPULSERCF3.zip](https://github.com/betaflight/betaflight/files/1381535/betaflight_3.2.0_IMPULSERCF3.zip)

sblakemore, 2017-10-13:
@mikeller It's been connected for 10 minutes now and is working fine!

mikeller, 2017-10-13:
Ok, changed the binary in the release, thanks for bringing this up.

sblakemore, 2017-10-17:
@mikeller unfortunately this seems to have surfaced again in 3.2.1, exactly the same behavior with the version on the release page.

I have pulled down the source and built the 3.2.1 tag and not seeing any problem in my build. Please let me know if there is anything I can do to assist.

mikeller, 2017-10-17:
@wind0r: Do you have time to look into this? The hex files that are in the 3.2.1 release are from https://ci.betaflight.tech/job/Betaflight%20Maintenance%203.2/9/. 
I'd already swapped the IMPULSERCF3 file in the 3.2.0 release for one that I'd built earlier, since the same problem had been reported for 3.2.0.

mikeller, 2017-10-17:
@sblakemore: Is it possible that you post the hex file (zipped) that you built here? Other users who aren't set up to build will appreciate it.

sblakemore, 2017-10-17:
Of course, here is my build.

[betaflight_3.2.1_IMPULSERCF3.zip](https://github.com/betaflight/betaflight/files/1390182/betaflight_3.2.1_IMPULSERCF3.zip)

AndersHoglund, 2017-10-17:
Anyone tried the binaries from the old Jenkins site? 
http://andwho.sytes.net:8080/job/BorisB_BetaFlight_Maint/115/

Would be good to know if there are some nasty environment dependencies. 
But what could possibly be specific to this FC ?? Other F3s are working fine....or?  Mysterious.

mikeller, 2017-10-17:
Good idea, @AndersHoglund.

@sblakemore: Can you please test this hex file:

[betaflight_3.2.1_IMPULSERCF3.zip](https://github.com/betaflight/betaflight/files/1392415/betaflight_3.2.1_IMPULSERCF3.zip)

AndersHoglund, 2017-10-17:
What is different with this hex file?

mikeller, 2017-10-17:
That is the one from the CI build you referenced above. I don't have an IMPULSERCF3, and I want to make sure that this fixes the problem before I replace the files in the release.

sblakemore, 2017-10-18:
Same problem with that hex unfortunately.

More than happy to send you hardware to add to the collection if it helps in the future.

AndersHoglund, 2017-10-18:
Interesting.... I do not have this FC to test and verify. But is it not a bit similar to BFF3? I do not see any such problem on BFF3. Any other target with the same kind of problem?
If you can build a working binary, on what environment is that ? Me using Ubuntu 16.04LTS, fully up-to-date.

AndersHoglund, 2017-10-18:
Have you tried the current HEAD of master, BF3.3.0? Same problem there??

AndersHoglund, 2017-10-18:
More builds to try, from different envs:
[betaflight_3.2.1_IMPULSERCF3_cygwin_build.zip](https://github.com/betaflight/betaflight/files/1394290/betaflight_3.2.1_IMPULSERCF3_cygwin_build.zip)
[betaflight_3.2.1_IMPULSERCF3_vagrant_build.zip](https://github.com/betaflight/betaflight/files/1394291/betaflight_3.2.1_IMPULSERCF3_vagrant_build.zip)

sblakemore, 2017-10-19:
I'm on Win 10 with cygwin, set up as per the [Wiki](https://github.com/martinbudden/betaflight/blob/master/docs/development/Building%20in%20Windows.md)

Of those two zipped files, the vagrant build exhibits the issue and the cygwin build works perfectly.

The 3.2.0 build provided in the second post by @mikeller also works fine, not certain of his environment...

mikeller, 2017-10-19:
Ok, I've replaced the IMPULSERCF3 binary in the 3.2.1 release with a locally generated one (I use linux), to keep more users from having spend time debugging this.

@sblakemore: Can you please verify that https://github.com/betaflight/betaflight/releases/download/v3.2.1/betaflight_3.2.1_IMPULSERCF3.hex is working?

sblakemore, 2017-10-19:
Sorry to say that build is not working, same problem. Board becomes unresponsive after around 20 seconds of being connected by USB.

mikeller, 2017-10-19:
Gah, this is getting worse and worse. I've uploaded @AndersHoglund's cygwin build. Also, I managed to organise an IMPULSERCF3 sample, so I should be able to test myself shortly.

richard-scott, 2017-10-19:
I had something similar happen in #3932

AndersHoglund, 2017-10-19:
> The 3.2.0 build provided in the second post by @mikeller also works fine, not certain of his environment...

@sblakemore , interesting. If I understand @mikeller correctly, that file was unzipped from the artifacts archive downloaded from the old Jenkins site. Correct? In that case I have a correctly working environment there... It may be good if you could verify this by downloading it yourself and test. 
http://andwho.sytes.net:8080/job/BorisB_BetaFlight_Maint/
Also the 3.3.0 master branch builds might be interesting to run a test on:
http://andwho.sytes.net:8080/job/BorisB_BetaFlight/

This is weird. Same compiler versions 2017-q2-update in all cases, different builds Linus/Win though.  Not sure how compiler optimizations are the same in different environments. Can host memory size affect the result? My Vagrant (failing) env is limited, not sure how much mem I have. Will check.

Should we try a compiler down-grade on Linux?  Not much good if I try to do tests on this, as my Linux env seems to be OK. I would however like a verification of that. I have done a manual, non-Jenkins, build. This to verify the builds in Jenkins and a manual interactive shell gives the same results. Both Attached.
Playing with compiler versions in Vagrant would of course be an alternative, but that env is soo, slow.

[betaflight_3.2.1_IMPULSERCF3_andwho_manual_build.zip](https://github.com/betaflight/betaflight/files/1397844/betaflight_3.2.1_IMPULSERCF3_andwho_manual_build.zip)
[betaflight_3.2.1_IMPULSERCF3_andwho_jenkins_build.zip](https://github.com/betaflight/betaflight/files/1397854/betaflight_3.2.1_IMPULSERCF3_andwho_jenkins_build.zip)

Both built on the v3.2.1 release label, and the last (jenkins) binary is taken from the Jenkins maint build linked above. Sorry @sblakemore , this will keep you busy testing, I have no hardware to run this on.

AndersHoglund, 2017-10-19:
For the record, a few host memory sizes.
Vagrant on Win10, failing:
```
ubuntu@ubuntu-xenial:~$ grep MemTotal /proc/meminfo
MemTotal:        4046480 kB

```
CygWin on Win10, OK:
```
>grep MemTotal /proc/meminfo
MemTotal:        8333004 kB
```
Ubuntu (on real HW), OK (?):
```
andwho@pucko:~$ grep MemTotal /proc/meminfo
MemTotal:        7343432 kB

```
Conclusion, small machines fail, bigger OK. Or ?????

AndersHoglund, 2017-10-19:
@wind0r, are https://ci.betaflight.tech/ running directly on silicon or in a WM? Memory resources??

wind0r, 2017-10-19:
@AndersHoglund that jenkins is a docker container on a real server with plenty of memory

AndersHoglund, 2017-10-19:
Aha. And docker has no memory limitations, can use all of available memory? Including virtual memory/swap?? What does `free `or `/proc/meminfo` say, running in docker???

wind0r, 2017-10-19:
docker is "just" cgroups. Just a way to run binarys without caring about libs and such ;)
There isnt any resource limit set from my side. Free will give you over 10G avaible and meminfo will return the same message as if i would run it from the "host" system.

i think If hardware would be an issue the build would stop. 
i think we should check the binarys to see if there is some compiler/link/cache issue or so.

i know @mlyle is good at that. maybe if we provied him with both hex files he will help us.

mikeller, 2017-10-19:
@AndersHoglund:

> @sblakemore , interesting. If I understand @mikeller correctly, that file was unzipped from the artifacts archive downloaded from the old Jenkins site. Correct? In that case I have a correctly working environment

No, the 3.2.0. binary I provided was build locally on a linux machine with `arm-none-eabi-gcc (GNU Tools for ARM Embedded Processors 6-2017-q2-update) 6.3.1 20170620 (release) [ARM/embedded-6-branch revision 249437]`. The 3.2.1 that does not work was built on a different linux machine, but with the same compiler suite, installed from the same package. Once I got a board to test with, I will dissect the differences to see if I can find what is causing this.

sblakemore, 2017-10-20:
Latest results from testing:

betaflight_3.2.1_IMPULSERCF3_andwho_jenkins_build - BAD
betaflight_3.2.1_IMPULSERCF3_andwho_manual_build - BAD
betaflight_3.3.0-1742-97c3e4b.zip - GOOD
betaflight_3.2.2-120-31251de.zip - BAD

@mikeller I believe your sample boards shipped today, so hopefully will be with you soon.

mikeller, 2017-10-20:
:-)

mikeller, 2017-10-25:
@sblakemore: That was quick, thank you! Try this:

[betaflight_3.2.1_IMPULSERCF3.zip](https://github.com/betaflight/betaflight/files/1413548/betaflight_3.2.1_IMPULSERCF3.zip)

closed 2017-10-26 by mikeller

mikeller, 2017-10-26:
Replaced `betaflight_3.2.1_IMPULSERCF3.hex` in the 3.2.1 release with one built with the change from #4439 backported into it.

sblakemore, 2017-10-27:
Thanks so much for you work on this @mikeller bit of a crazy bug!

PR #4439 (2017-10-26): Fixed IMPULSERCF3 target definition.
Fixes #4352.