betaflight/betaflight #4353: Can't activate GPS

madmilchmann, 2017-10-13:
Hi there!

I have a Problem with activating the GPS in BF3.2.
First I went to the Ports Tab and chose GPS as sensor input for UART2 
after saving i went to the configuration tab and switched on the GPS option. when i click Save and Reboot the GPS is deactivated again.

gear:
HGLRC F3 v3.1 FC
Ublox M8N GPS Module
XSR-E REceiver

\\\
# diff
# Betaflight / SPRACINGF3 (SRF3) 3.2.0 Sep 28 2017 / 16:45:44 (ceb6f70) MSP API: 1.36
name H2

feature -RX_PPM
feature -TELEMETRY
feature -RSSI_ADC
feature -TRANSPONDER
feature RX_SERIAL
feature MOTOR_STOP
feature LED_STRIP
map TAER1234
serial 1 2 115200 9600 0 115200
serial 2 64 115200 57600 0 115200
aux 0 0 1 1300 2100
aux 1 2 1 1300 1700
aux 2 13 3 1700 2100
aux 3 19 0 1300 2100
aux 4 28 1 1650 2100
set rssi_channel = 9
set serialrx_provider = SBUS
profile 0

rateprofile 0
\\\

McGiverGim, 2017-10-13:
The GPS support has been removed for F3, it don't fit in the available space.

madmilchmann, 2017-10-13:
is there a way to enable it? by deactivating other stuff maybe?
Or do i have to downgrade to an older version?

McGiverGim, 2017-10-13:
The only way is build your own firmware deactivating the stuff that you don't need.

madmilchmann, 2017-10-14:
Ok... thank you for you responses.
 building my own firmware probably goes over my head. My coding skills are rather limited :( - but i'll give it a look.
when you say the GPS support has been removed for F3, do you mean in BF3.2? so would downgrading solve my problem?

mikeller, 2017-10-14:
@madmilchmann: Building your own custom firmware isn't too hard, as long as all you want to do is enable / disable some features. There's  a primer on how to configure your own version at: https://github.com/cleanflight/cleanflight/blob/master/docs/Customized%20Version.md , and for the actual build you can use a prebuilt Docker container: https://hub.docker.com/r/betaflight/betaflight-build/

closed 2017-10-14 by mikeller