betaflight/betaflight #4355: AUX 4 not responding

tnuki, 2017-10-13:
On the OmnibusF4 firmware, in the reciever tab, my Aux4 switch is stuck at 1971, on my radio (qx7) in the outputs tab, it shows that the switch is still working properly.

Here's some screencaps of my reciever tab:

http://prntscr.com/gx0lh8
http://prntscr.com/gx0lng

and heres the mixer (on the radio):

http://prntscr.com/gx0mji
http://prntscr.com/gx0mra

///
# diff

# version
# Betaflight / OMNIBUSF4 (OBF4) 3.2.0 Sep 28 2017 / 16:38:25 (ceb6f70) MSP API: 1.36

# name
name Licorice

# resources
resource MOTOR 1 A02
resource MOTOR 2 A03
resource MOTOR 3 B01
resource MOTOR 4 B00

# mixer

# servo

# servo mix

# feature
feature TELEMETRY
feature RSSI_ADC
feature AIRMODE
feature ANTI_GRAVITY
feature DYNAMIC_FILTER

# beeper

# map
map TAER1234

# serial
serial 0 64 115200 57600 0 115200
serial 5 32 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 6 1700 2100
aux 1 1 3 900 1300
aux 2 2 3 1300 1700
aux 3 13 5 1700 2100
aux 4 35 4 1700 2100

# adjrange

# rxrange

# vtx

# rxfail

# master
set gyro_notch1_hz = 0
set gyro_notch2_hz = 0
set baro_hardware = NONE
set rssi_channel = 16
set fpv_mix_degrees = 30
set serialrx_provider = SBUS
set motor_pwm_protocol = DSHOT600
set failsafe_delay = 10
set failsafe_throttle_low_delay = 10
set align_board_yaw = 270
set bat_capacity = 1300
set ibata_scale = 200
set ibatv_scale = 200
set beeper_dshot_beacon_tone = 3
set deadband = 3
set yaw_deadband = 3
set pid_process_denom = 1
set osd_rssi_alarm = 40
set osd_cap_alarm = 1300
set osd_tim2 = 769
set osd_vbat_pos = 2466
set osd_rssi_pos = 2114
set osd_tim_1_pos = 54
set osd_tim_2_pos = 2486
set osd_flymode_pos = 2146
set osd_throttle_pos = 225
set osd_vtx_channel_pos = 377
set osd_crosshairs = 200
set osd_ah_sbar = 200
set osd_ah_pos = 200
set osd_current_pos = 424
set osd_mah_drawn_pos = 432
set osd_craft_name_pos = 2097
set osd_gps_speed_pos = 218
set osd_gps_lon_pos = 82
set osd_gps_lat_pos = 65
set osd_gps_sats_pos = 51
set osd_home_dir_pos = 302
set osd_home_dist_pos = 303
set osd_compass_bar_pos = 266
set osd_altitude_pos = 247
set osd_pid_roll_pos = 423
set osd_pid_pitch_pos = 455
set osd_pid_yaw_pos = 487
set osd_debug_pos = 1
set osd_power_pos = 321
set osd_pidrate_profile_pos = 345
set osd_warnings_pos = 2409
set osd_avg_cell_voltage_pos = 76
set osd_pit_ang_pos = 257
set osd_rol_ang_pos = 289
set osd_battery_usage_pos = 392
set osd_disarmed_pos = 2314
set osd_nheading_pos = 311
set osd_nvario_pos = 279
set osd_esc_tmp_pos = 82
set osd_esc_rpm_pos = 83
set osd_stat_max_spd = OFF
set osd_stat_max_curr = OFF
set osd_stat_used_mah = OFF
set osd_stat_bbox = OFF
set osd_stat_endbatt = ON
set osd_stat_bb_no = OFF

# profile
profile 0

set dterm_lowpass_type = PT1
set dterm_notch_hz = 0
set anti_gravity_gain = 3000
set dterm_setpoint_weight = 30
set crash_gthreshold = 600
set crash_time = 1000
set crash_recovery_angle = 0
set p_pitch = 52
set i_pitch = 60
set d_pitch = 13
set p_roll = 47
set i_roll = 55
set d_roll = 12
set p_yaw = 85
set i_yaw = 55
set level_limit = 75

# rateprofile
rateprofile 0

set rc_rate = 130
set rc_rate_yaw = 130
set rc_expo = 30
set rc_expo_yaw = 30
set roll_srate = 75
set pitch_srate = 75
set yaw_srate = 75
set tpa_rate = 25

# 
///

mikeller, 2017-10-13:
What RX do you use? Some of them (XM+ with special firmware for example) inject RSSI on an AUX channel.

Also, nice name! ;-)

pulquero, 2017-10-20:
My aux4 is also not responding, stuck on 1673. It displays correctly on my Q X7. I'm using it for digital RSSI. FC running FURYF4OSD, RX is XSR-M. Used to work before I changed FC and upgraded betaflight.
{code}
set rssi_channel = 8
set max_aux_channels = 4
{code}

mikeller, 2017-10-20:
@pulquero: This sounds more of a problem with your TX / RX. The RX seems to be sending a constant value for RSSI.

pulquero, 2017-10-20:
Yep, your right. You made me realise that I had reset the sensors during the upgrade and they didnt come back in the same order so I had some other sensor input linked to that channel instead of rssi. Thanks!

closed 2017-10-20 by jflyper