betaflight/betaflight #4356: Slow forward flip after flashing 3.2.0, motors not spin when armed with OneShot125

MR444, 2017-10-13:
Hi, I just updated from 3.1.5 to 3.2.0.
With 3.1.5 I used ONESHOT125 protocol without anti gravity and my quad flew perfect. No movement.

Since then it is hard to fly. When moving throttle to 0, my quad does a slow front flip. It is still controlable with zero throttle but very hard. I have to move sticks almost to end position. I tried to trim tail-heavy but it did not change anything. Another thing I noticed. When activating ONESHOT125, motors do not spin after arming although motor_stop is not actice. So I tried DSHOT150 and 600. Then the motors spin when armed but flight behaviour is still bad. I also tried airmode and anti gravity but it did not make it better.

My hardware is: 
Hobbywing XRotor F4 
Hobbywing 4in1 ESC
2206-2400kV
5043 3 blade
Symmetrical frog frame.
All components new and uncrashed.

Is it because of a changed PID controller? Is there a quick way to fix this problem? Was there something changed in oneshot125? I wonder about theese changes in flight behaviour and oneshot125. I changed nothing else but flashing the firmware. Thanks!

Jahnkeanater, 2017-10-13:
Maybe try increasing the min command/idle % . If your motors do not spin fast enough at idle that could cause strange flight behavior. My quad idles at 1040 or 4%.

Also, check that your pitch I gain is not 0. The default is 40. Maybe try increasing it to 50.

jflyper, 2017-10-13:
If all else fails, I would suggest you wipe the current configuration with CLI "defaults" and start configuring from scratch.

Bugg69s, 2017-10-14:
Mine does same thing in *ACRO* 
But in AIR MODE it is not a problem