betaflight/betaflight #4358: Betaflight configuration deleted when lipo connected.

Bugg69s, 2017-10-13:
Today I was flying like normal. 
I connected my lipo and it showed low voltage (BF though that I have 5s).
Like normally I was connecting and disconnecting lipo until it showed that I have 4s.

But today I done it and when I connected and disconnected my lipo my whole configuration was gone.
I had OSD full of that sh*t, and then I known that something is wrong.

Omnibus f3
BF 3.2 "stable"
About ~120 cycles

Bugg69s, 2017-10-13:
I had saved my configuration in my PC when updating from 3.2 RC5 to 3.2 stable

jflyper, 2017-10-13:
Is it the FC configuration or OSD font that was gone?

Bugg69s, 2017-10-14:
It was like a new FC, everything was deleted.

jflyper, 2017-10-14:
Sounds like you had a bad luck.

I suggest not to power off the FC until it's fully powered.

closed 2017-10-14 by jflyper

Bugg69s, 2017-10-14:
Ok :)