betaflight/betaflight #4361: Cant connect any quad to betaflight ?

noobnumber1, 2017-10-14:
# If your issue looks like a hardware fault or a configuration problem please don't raise an issue here.

## Please consider using other user support options such as asking the manufacturer of the hardware you are using, [RCGroups](https://rcgroups.com/forums/showthread.php?t=2464844), [Slack](https://slack.betaflight.tech) or other user support forums & groups (e.g. facebook).

## If you believe there is an issue with the firmware itself please follow these steps:
1.) Describe your problem;
2.) Include ways to reproduce the problem;
3.) Provide information about your flightcontroller and other components including how they are connected/wired;
4.) Add the used configuration and firmware version;
5.) Create a `diff` and post it here in a code block. Put \`\`\` (three backticks) at the start and end of the `diff` block; and
6.) Remove this Text :).

noobnumber1, 2017-10-14:
all of a sudden betaflight and cleanflight will not let me connect? It opens the port for approx. 30 sec then closes and does not go to the regular first gyro screen? Tried different usb cable and different quads and different computers in my home?

mikeller, 2017-10-14:
@noobnumber1: GitHub is for questions and discussion around the **development** of Betaflight. For questions around the configuration and use of Betaflight, please visit [RCGroups](https://www.rcgroups.com/forums/search.php?searchid=62637789&query=connection%20problems).

closed 2017-10-14 by mikeller

noobnumber1, 2017-10-14:
well! excuse meeeeeeeee! hope you crash!

wind0r, 2017-10-14:
@noobnumber1  with this attitude you should not realy expect much help here.
You didnt even read the Issue Text (See Point 6 in the list), do not provied enoug Informatiosn and having a "Problem" which is allready is adressed multply times.
Also this isnt a Betaflight Problem. Its a Problem with your computer. Just install the correct drivers as written in the Configurator help message or everywhere around the Internet.

Every unneeded Issue here will schedule the next release one month more away. 

Have a nice day and some nice flights :)