betaflight/betaflight #4363: Feature Request : option to set the maximum throttle on anti turtle mode

grhookgr, 2017-10-14:
when upside down sometimes we push the stick too much trying to recover and making the copter to jump roll again upside down.
it will be nice if we have the option to limit the maximum throttle on anti-turtle mode.

i was thinking a "set max_recover_throttle= 1750" or the desired value it may just work to help us recover faster.
it may be even configurable from osd

Gaz-B, 2017-10-20:
Why not just have the turtle mode switch also change the pitch and roll rates on your TX not everything has to be done in betaflight computer radios are very capable tools too.

richard-scott, 2017-10-20:
@Gaz-B Sadly my Turnigy Evolution TX doesn't have that capability.

I think the flip should be an automatic process where the FC slowly increases throttle to move the quad in the direction the TX requests. If the FC detects no roll movement it can back off and stop as that would indicate the quad is stuck and we don't want to burn out a motor or an ESC. Once the FC has passed 90 degrees (of the 180 required to flip) it could automatically cut throttle and the quad should flop to the ground right-side-up.

Gaz-B, 2017-10-21:
I mostly use turtle by banging the stick round and round until it drops back out the tree.

fiveangle, 2017-10-23:
Limiting throttle based on a situation where you have no idea what condition you might need is a very bad idea.

There was more discussion about how to make turtle flip better here:  https://github.com/betaflight/betaflight/issues/4259#issuecomment-336739110

-=dave

anti-vaxxer, 2025-02-20:
solved in #13905

closed 2025-02-20 by nerdCopter