betaflight/betaflight #4372: turtle mode

joefpv, 2017-10-16:
when turtle mode is activated, when roll to the right, motor # 1 does not spin, only motor # 2

joefpv, 2017-10-16:
Using  Dshot 1200.

brycedjohnson, 2017-10-17:
Your going to need give us more info to go off of...

## If you believe there is an issue with the firmware itself please follow these steps:
1.) Describe your problem;
2.) Include ways to reproduce the problem;
3.) Provide information about your flightcontroller and other components including how they are connected/wired;
4.) Add the used configuration and firmware version;
5.) Create a `diff` and post it here in a code block. Put ``` (three backticks) at the start and end of the `diff` block (instructions  on how to do a diff: https://oscarliang.com/use-diff-not-dump-betaflight/); and

closed 2017-10-18 by joefpv