betaflight/betaflight #4373: LED color representation.

Jahnkeanater, 2017-10-16:
The Red LED inside WS2812 LEDs is typically not as bright as the Blue and Green. I've noticed that it takes very little green to make the color orange. 

Instead of hue 30 orange is actually hue 15.
And to make yellow the hue is 28 not 60.

Is this the same with most LEDs? Can we change the values in ledstrip.c hsvColor_t hsv[] so that the colors are accurate. Or should we change colorconversion.c so the hue values are correct?

Maybe just decrease the strength of green for case 0?

Can other people confirm that their LEDs do not correctly produce the color orange. I would like to make sure this is a common issue and not just my LEDs before changing it.

Jahnkeanater, 2017-10-16:
Why is case 0 different?

case 0:
r.rgb.g = (((val - base) * hue) / 60) + base;

all other cases:
r.rgb.r = (((val - base) * (60 - (hue % 60))) / 60) + base;

Maybe this is just a bug in colorconversion.c ?
There is also a small whitespace error. :)

Jahnkeanater, 2017-10-16:
This code is copied from Arduino code from 7 years ago. I think the original code has an error but nobody has bothered to make sure it correctly converts the HSV values. The code works but its not acurate.

mikeller, 2017-10-16:
@Jahnkeanater: Any contribution that improves the quality of Betaflight is highly welcome!

Jahnkeanater, 2017-10-16:
I'm going to test the original code from 10 years ago on my arduino leonardo. If hue 30 gives me the wrong RGB values I'll fix it. Funny how old this code is.

Jahnkeanater, 2017-10-16:
The original code makes sense. Seems to return the correct value. I think its just the fact that the red led is dimmer.

DieHertz, 2017-10-16:
The face that blue LEDs have less irradiance than red and then green is will known and is accounted for by a special color space conversion, I believe it's correct in betaflight.
It has to do with manufacturing complications for blue LEDs as well as human eye perception of colors

Jahnkeanater, 2017-10-16:
@DieHertz In this case the blue LED is always 0. hue 0-60. 
RGB ( 255, 127, 0 ) should be orange but by playing with the hue manually I'm noticing RGB ( 255, 62, 0 ) is orange. Right now orange is basically yellow and yellow has a light green color.

Dividing Green by 2 seems like it fixes it but I don't want to modify a perfectly good function to correct the color.

DieHertz, 2017-10-16:
I suggest extracting color space parameters into CLI-configurable settings with current values as defaults, that way you'll be able to get colors you need by choosing custom parameters, without affecting everyone else.

Jahnkeanater, 2017-10-16:
I'm hoping its just my LEDs and that most LEDs have the correct colors. For now, I think I'm going to leave it the way it is.

mikeller, 2017-10-16:
I think most users will be content if they get a christmas-tree-blinky effect out of their LED_STRIP, and won't be concerned about the christmas tree being off-colour. :rofl:

Jahnkeanater, 2017-10-16:
My concern is with the VTX color orange and yellow both look yellow. I'm thinking about getting rid of yellow since it looks too much like orange and adding light blue.

I would like to keep orange since it does stand out from the other colors pretty well. I'm playing around with the colors to find the best VTX colors. 

I could also just define custom hsv values for the vtx and leave the default colors alone.

mikeller, 2017-10-16:
Would it make sense to use the defined default colours for VTX, but leave them changeable (as they currently are)? This way, if a particular model of LED_STRIP isn't accurate, the user can 'tune' the colours by changing the predefined colours.

Jahnkeanater, 2017-10-16:
@mikeller I was unaware that the predefined colors were changeable. Does the color command in CLI do that?

mikeller, 2017-10-16:
@Jahnkeanater: Yes. You can also change them in the UI by double clicking onto them (albeit the UI for it is super clumsy).

Jahnkeanater, 2017-10-16:
Look at that I can modify the HSV values right there! lol

Jahnkeanater, 2017-10-16:
The way it is now will the UI change the VTX colors?
 color = HSV(RED);

mikeller, 2017-10-16:
Nope. The named colours are always read from the const array `hsv` at the top of `src/main/io/ledstrip.c`.

Jahnkeanater, 2017-10-16:
How do I use the defined default colors?

mikeller, 2017-10-16:
`color = HSV(RED)`

If you mean the customer defined colours: `hsvColor_t color = ledStripConfig()->colors[index]`

Jahnkeanater, 2017-10-16:
@mikeller I'm guessing the index corilates to the number in the UI.
Is this correct for say red?

color = hsvColor_t color = ledStripConfig()->colors[2];

what about the getSC() function?

color = getSC(COLOR_RED);

mikeller, 2017-10-16:
@Jahnkeanater: Yes. Since `ledStripConfig()->colors` is initialised from `hsv`, you can even use color_red = ledStripConfig()->colors[COLOR_RED]` to get what is red in the default settings for custom colours (and thereby enable users to customise the value for red if it isn't showing as red on their LED_STRIP.