betaflight/betaflight #4379: Crash detection does not stop with disarm/arm event

DungAdumbs, 2017-10-17:
When rearming after disarming during a detected "crash", crash detection recovery is still active.  I don't recall this issue with 3.2 rc 4 but 3.2 final and 3.2.1 suffer from this issue.  Can use turtle mode to flip quad but upon rearming crash detection is still active.   Sometimes quad will aggressively flip from upright to inverted upon rearm.  Same situation happens after picking up "crashed" quad and rearming.  If hand held, quad requires moving to calculated horizon which can be way off from reality after a crash/hard flying.  This is on Omnibus F3 (Eachine Lizard 95).  I've seen other RCGroups comments that crash detection was messed up after 3.2rc4.

DungAdumbs, 2017-10-18:
further conformation today.  Set crash_recovery = BEEP.  Post "crash", have tone after disarm & rearm until crash recovered by quad attitude.

`# diff
# Betaflight / OMNIBUS (OMNI) 3.2.1 Oct 15 2017 / 19:42:04 (434db9a36) MSP API: 1.36
name DfuknA

feature -RX_PPM
feature RX_SERIAL
feature AIRMODE
feature ANTI_GRAVITY
feature DYNAMIC_FILTER
beeper -ON_USB
map TAER1234
serial 0 1 115200 57600 0 115200
serial 2 64 115200 57600 0 115200
led 0 0,7::COI:0
led 1 7,7::CTO:0
led 2 8,7::CTO:0
led 3 15,7::COI:0
aux 0 0 0 1400 2100
aux 1 2 1 1800 1900
aux 2 13 1 1575 1925
aux 3 35 1 1800 1875
rxrange 0 1156 1838
rxrange 1 1158 1830
rxrange 2 1158 1840
rxrange 3 1158 1820
set gyro_sync_denom = 2
set gyro_notch1_hz = 0
set gyro_notch2_hz = 0
set acc_trim_pitch = 36
set acc_trim_roll = -24
set baro_hardware = NONE
set min_check = 1010
set max_check = 1990
set max_aux_channels = 2
set airmode_start_throttle = 1100
set blackbox_device = SERIAL
set min_throttle = 1050
set dshot_idle_value = 200
set motor_pwm_protocol = DSHOT600
set bat_capacity = 450
set vbat_warning_cell_voltage = 34
set vbat_hysteresis = 3
set current_meter = VIRTUAL
set ibata_scale = 1000
set ibatv_scale = 250
set ibatv_offset = 20
set small_angle = 180
set deadband = 4
set yaw_deadband = 4
set pid_process_denom = 1
set osd_units = IMPERIAL
set osd_cap_alarm = 500
set osd_tim1 = 18945
set osd_vbat_pos = 44
set osd_rssi_pos = 1399
set osd_tim_1_pos = 54
set osd_flymode_pos = 2264
set osd_throttle_pos = 2177
set osd_vtx_channel_pos = 200
set osd_crosshairs = 0
set osd_current_pos = 2199
set osd_mah_drawn_pos = 2232
set osd_craft_name_pos = 386
set osd_gps_speed_pos = 66
set osd_gps_lon_pos = 82
set osd_gps_lat_pos = 65
set osd_gps_sats_pos = 1410
set osd_home_dir_pos = 302
set osd_home_dist_pos = 303
set osd_compass_bar_pos = 266
set osd_altitude_pos = 161
set osd_pid_roll_pos = 423
set osd_pid_pitch_pos = 455
set osd_pid_yaw_pos = 487
set osd_debug_pos = 1
set osd_power_pos = 320
set osd_pidrate_profile_pos = 345
set osd_warnings_pos = 2249
set osd_avg_cell_voltage_pos = 2135
set osd_pit_ang_pos = 2440
set osd_rol_ang_pos = 289
set osd_nheading_pos = 311
set osd_nvario_pos = 279
set osd_esc_tmp_pos = 82
set osd_esc_rpm_pos = 83
set osd_stat_max_spd = OFF
set osd_stat_min_rssi = OFF
set osd_stat_bbox = OFF
set osd_stat_bb_no = OFF
set vcd_video_system = 1
set displayport_max7456_wht = 0
set camera_control_ref_voltage = 310
profile 0

set dterm_lowpass_type = PT1
set dterm_notch_hz = 0
set dterm_setpoint_weight = 60
set crash_recovery = BEEP
set p_pitch = 40
set i_pitch = 43
set d_pitch = 22
set i_roll = 43
set d_roll = 22
rateprofile 0

set rc_rate = 70
set rc_rate_yaw = 120
set thr_mid = 5
set thr_expo = 60
set roll_srate = 85
set pitch_srate = 85
set yaw_srate = 77
set tpa_rate = 20

# set crash
crash_dthreshold = 50
Allowed range: 0 - 2000

crash_gthreshold = 400
Allowed range: 0 - 2000

crash_setpoint_threshold = 350
Allowed range: 0 - 2000

crash_time = 500
Allowed range: 0 - 5000

crash_delay = 0
Allowed range: 0 - 500

crash_recovery_angle = 10
Allowed range: 0 - 30

crash_recovery_rate = 100
Allowed range: 0 - 255

crash_limit_yaw = 200
Allowed range: 0 - 1000

crash_recovery = BEEP
Allowed values: OFF, ON, BEEP
`

mikeller, 2017-10-18:
Thanks for the testing, @DungAdumbs!

Vidalcris, 2017-10-19:
I did the PR but i need help because i think ARMING_FLAG(DISARMED) is not working XD
https://github.com/betaflight/betaflight/pull/4396

commit 184fc3c7 (2017-10-19): Merge pull request #4399 from Vidalcris/patch-4

Crash recovery - Fixes #4379
commit 70973bad (2017-10-19): Crash recovery - Fixes #4379
commit 82ad1488 (2017-10-19): Merge pull request #4399 from Vidalcris/patch-4

Crash recovery - Fixes #4379
closed 2017-10-19 by mikeller