betaflight/betaflight #4380: BF 3.2.x, 32k pt1 filter, all dshot protocols; motors burning up only when arming

frizzle101101, 2017-10-17:
Testing on motor tab motors work fine with any settings. When arming the motors get crazy vibrations as if desync; This is related to enabling 32k and pt1 filter on my kakute f4 aio, only with dshot protocols on gs30 blheli_32 escs. All is fine with biquad on 32k. Pt1 works with 32k disabled. pt1 works with 32k when multishot is enabled. so it seems that the 32bit protocal issue.

wind0r, 2017-10-17:
Please provied a diff.
AND remember that 32khz is a high noise mode. 
Simply disabling filters and switching the filter type will lead in hot motors.
Not realy something the Firmware can change there. You simply need to tune the filters to fit your hype.

automatt1c, 2018-01-02:
I'm having the same issue as well on kakute f4 aio. If i Enable 32K mode and 16k/16k with PT1 with all the stock filters and lower than stock PIDS. The Quad Goes crazy when arming.  On the motors tab it shows the throttle going from full to zero very quickly.  On 8k/8k it works perfectly fine. it definitely seems like a bug with how its reading the gyro data. 

I know there is less filtering on 32k mode. Is it just not possible to run PT1 at 32k mode? is PT1 even needed on 32k?

wind0r, 2018-01-02:
If there is a bug with reading the gyro data why it works with 8k then?
As written above. You need to tune your filters!
PT1 is just a first-order differential equation which is not enough for noisy data from NO FILTER MODE (32khz doesnt mean less filter, it means no filter).
Use Biquad or FIR and check your Gyro Data.

RotorVulura, 2018-01-24:
Same here. Kakute F4 AIO BF 3.2.X
If you turn on 32k Gyro (switch) the motors starts grinding/growing up in resonance and starts howling. Hot motors / ESCs in few seconds. Same configuration, PID loop, Gyro, PIDs, Expo,  filters and so on, just the switch set to "enable".
So I can fly super smooth 8K-8K dshot 300/600 without filters - emax 2205S - fvt30A blheli_s (caps removed manually for dshot) or racerstar 30A V2 without 32k gyro enabled. I don't use particular features like antygravity, permanent air mode or other fancy help like auto filter.
This does not make sense. I'm investigating about this switch is intend for, but nothing strange found, yet. You are the master :)
Thx for your time and go go go Rotor Vulura !

RotorVulura, 2018-01-25:
Long story short : https://github.com/betaflight/betaflight/wiki/Gyro-&-Dterm-filtering-recommendations
Now I'm running 16k-16k. Cold motors. Very good piece of software :-)
Biquad and Dterm active only. I need more BB debug but it's almost perfect now playing P around 25 an D at 5, impressive low D.