betaflight/betaflight #4384: Turtle mode: small_angle changes by itself

copterella, 2017-10-17:
There is this weird behaviour in Betaflight 3.2.0 (release version) which after using "anti turtle mode" can leave the quad unable to arm. Basically, what happens is that Betaflight resets the value of "small_angle" back to 25 all by itself (it was at 180 before). So when the quad has a bigger tilt after the turtle flip it cannot be rearmed for take-off.

When I changed the small_angle value to 179 the problem disappeared. I was able to reproduce it on two of my quads but it does not seem to happen to everyone. I am using a Hobbywing X-rotor F4 controller with the accelerometers on and crash recovery off. I have made a video that explains the details: https://www.youtube.com/watch?v=KRsZXsVF1fE

brycedjohnson, 2017-10-17:
Turtle mode bypasses the small angle check when it is activated, although it doesn't change the value.  I don't see anywhere that is setting the small_angle in the code besides over the CLI and for the first initialization.  It could be the small_angle 180 setting had gotten lost somewhere long the way with updates, but hadn't been noticed because turtle mode now bypasses the small angle check?

copterella, 2017-10-17:
I'm positive that I had set and saved small_angle to 180 after the upgrade and before the turtle action...