betaflight/betaflight #4385: Cannot connect to drone.

KrisspyLettuce, 2017-10-17:
So I am a first time drone owner, I bought a eschine wizard x200 but I cant get it to connect to beta flight. what am I doing wrong? I plug it in but it says no connection found.

jflyper, 2017-10-17:
@KrisspyLettuce We are sorry to tell you this is not a place for user support.

Please redirect your problem statement to other forums on the net.

If your craft is not X200 but X220, there is a nice thread on RCG.
https://www.rcgroups.com/forums/showthread.php?2720753-Eachine-Wizard-X220-%28Under-construction%29

You can also some search in RCG.
https://www.rcgroups.com/forums/search.php?searchid=62686257&query=x220%20connection

closed 2017-10-17 by jflyper

rinyheuvelman, 2018-05-02:
First update your f3  racing fc board to latest betaflight, see https://www.youtube.com/watch?v=e7dGCZMJ6SA
that fixed it for me. 
Make sure your rx is plugged the right way
for ibus:horizontal top three pins on the right, to boardconnection IO1
for ppm: three vertical pins on left bottomside to boardconnection IO2
connection see http://nathan.vertile.com/assets/images/blog/fs_i6/500px-FlySky_IA6B_pinout-f89eed4f.jpg
FC: 
[SPRacingF3-Manual-latest.pdf](https://github.com/betaflight/betaflight/files/1966781/SPRacingF3-Manual-latest.pdf)
[FS-i6-MANUAL-CN-EN-20150601.pdf](https://github.com/betaflight/betaflight/files/1966783/FS-i6-MANUAL-CN-EN-20150601.pdf)