betaflight/betaflight #4386: Betaflight 3.2.1 : Pid tuning / angle-horizon/ Sensitivity stays on 0

Sixks, 2017-10-17:
Hello people from Betaflight

First of all, congratulations and thank you for your work.
Can you help me solve this problem:

 I successfully flashed a SPRacingF3 board with FW 3.2.1, reconfigured with
Betaflight, binded with a DX8 and of course activated the dynamics filters,
I have two identical multi, the difference in flight between firmware 3.1.5 and 3.2.1 is huge! What an advance!
However, I wanted to change the 'sensitivity' value in the pid tuning tab, but it remains at 0.
I noticed that in dump 3.1.5 there is a CLI command **set level_sensitivity = xx**.
This command does not appear in version 3.2.1 so I tried to write it, but it is invalid.
Does this command still exist?

mikeller, 2017-10-17:
No, this has been removed. `level_limit` is still there, and can be used to configure the angle at which the cut over happens

Sixks, 2017-10-17:
Thank you for the quick reply.
It's a shame, this function was interesting.
But I could find an alternative by increasing the D setpoint weight.

Hustleflyer, 2017-10-19:
Yes, I was wondering about this change from 3.1.7 to 3.2 and 3.2.1.  I have found that D setpoint weight and transition between 1.4 - 1.6 and 0.4 - 0.6, respectively -work very well.  I personally am running this at ~1.5 and ~0.5 on my craft...give or take a few hundredths.

closed 2017-10-19 by mikeller