betaflight/betaflight #4387: BF3.2.1 SPF3 Evo dshot300 motor 4 issue and Ledstrip

agropom, 2017-10-17:
Dear all,
on my SPF3 Evo I installed last BF3.2.1.
I'm using RS30A lite blheli_s with oneshot125 and all 4 motor spin correctly, if I use dshot300 only motor 1,2,3 will spin.
I read that is necessary remap motor 4 on another "pin" but I'm having some issue I receive this message:

> A06 already assigned to SERIAL RX11

Can someone help me to remap the 4th motor?

I also have another problem that the led strip doesn't work anymore. If I use BF3.1.6 led will work
Thanks!

> resource list
IO
A00: MOTOR 1
A01: MOTOR 2
A02: MOTOR 3
A03: MOTOR 4
A04: ADC_BATT
A05: FREE
A06: FREE
A07: FREE
A08: TRANSPONDER
A09: SERIAL_TX 1
A10: FREE
A11: FREE
A12: FREE
A13: FREE
A14: FREE
A15: SERIAL_RX 2
B00: FREE
B01: FREE
B02: ADC_RSSI
B03: SPI_SCK 1
B04: SPI_MISO 1
B05: SPI_MOSI 1
B06: I2C_SCL 1
B07: I2C_SDA 1
B08: LED 1
B09: MPU_CS
B10: SERIAL_TX 3
B11: FREE
B12: SDCARD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C13: MPU_EXTI
C14: SDCARD_DETECT
C15: BEEPER
F00: FREE
F01: FREE
F04: FREE

> DMA:
DMA1 Channel 1: MOTOR 3
DMA1 Channel 2: TRANSPONDER
DMA1 Channel 3: FREE
DMA1 Channel 4: FREE
DMA1 Channel 5: MOTOR 1
DMA1 Channel 6: FREE
DMA1 Channel 7: MOTOR 2
DMA2 Channel 1: ADC
DMA2 Channel 2: FREE
DMA2 Channel 3: FREE
DMA2 Channel 4: FREE
DMA2 Channel 5: FREE

> diff
feature -RX_PPM
feature RX_SERIAL
feature LED_STRIP
feature AIRMODE
feature ANTI_GRAVITY
beeper -BAT_LOW
beeper -ON_USB
serial 0 32 115200 57600 0 115200
serial 1 64 115200 57600 0 115200
serial 2 8192 115200 57600 0 115200
led 0 0,0::C:6
led 1 1,0::C:6
led 2 2,0::C:6
led 3 0,1::C:6
led 4 1,1::C:6
led 5 2,1::C:6
led 6 0,2::C:6
led 7 1,2::C:6
led 8 2,2::C:6
led 9 3,0::C:6
led 10 4,0::C:6
led 11 5,0::C:6
led 12 6,0::C:6
led 13 7,0::C:6
led 14 8,0::C:6
led 15 9,0::C:6
led 16 10,0::C:6
led 17 11,0::C:6
led 18 12,0::C:6
led 19 13,0::C:6
led 20 14,0::C:6
led 21 15,0::C:6
led 22 3,1::C:6
led 23 4,1::C:6
led 24 5,1::C:6
led 25 6,1::C:6
led 26 7,1::C:6
led 27 8,1::C:6
led 28 9,1::C:6
led 29 10,1::C:6
led 30 11,1::C:6
led 31 12,1::C:6
mode_color 7 0 7
aux 0 0 0 1700 2100
aux 1 1 1 1700 2100
aux 2 2 1 1300 1700
aux 3 13 2 1700 2100
aux 4 15 3 1700 2100
set gyro_sync_denom = 2
set baro_hardware = NONE
set min_check = 1010
set max_check = 1990
set serialrx_provider = SBUS
set min_throttle = 1017
set motor_pwm_protocol = DSHOT300
set align_board_yaw = 270
set vbat_scale = 108
set pid_process_denom = 2
profile 0
rateprofile 0

mikeller, 2017-10-17:
@agropom: GitHub is for questions and discussion around the **development** of Betaflight. For questions around the configuration and use of Betaflight, please visit [RCGroups](https://www.rcgroups.com/forums/search.php?searchid=62637789&query=betaflight%20resource%20remapping).

Also, have a look at https://github.com/betaflight/betaflight/wiki/Betaflight-resource-remapping.

closed 2017-10-17 by mikeller

richard-scott, 2017-10-18:
Did you read the [DShot Wiki ](https://github.com/betaflight/betaflight/wiki/DSHOT-ESC-Protocol#spracingf3evo----must-move-motor-4-to-new-pin-assignment-cli--resource-motor-4-a06-then-solder-esc-for-motor-4-to-motor-output-5-fixes-dma-conflict-with-motor-outputs-2-and-4) page? 

You also need to move the signal wire for motor 4 to the output pad for motor 5?