betaflight/betaflight #4390: RSSI Not Displaying in Betaflight 3.2.1 Omnibus F4

rinciong, 2017-10-18:
1.) Describe your problem
   - I moved from 3.17 to 3.2.1, and made no other changes, other than removing "set yaw_motor_direction" since the motor direction is accessible through the configuration page.
2.) Include ways to reproduce the problem;
   - On the OSD tab, setting the toggle switch for "RSSI" does not change anything on the preview screen of the OSD.
3.) Provide information about your flightcontroller and other components including how they are connected/wired;
  - I have two DYS Ominbus F4 FC's, VTX and Camera are wired directly to the board.  I can see all other OSD features.
4.) Add the used configuration and firmware version;
  - Betaflight Omnibus F4 3.2.1

Arakon, 2017-10-18:
No issues here, freshly flashed to 3.2.1 this morning.
[image]

Check that the RSSI position is not out of bounds.. try resetting to default and not copying over the OSD strings with the settings when restoring.

jflyper, 2017-10-18:
@rinciong 
Please do the following.
- Execute CLI "defaults" or full-erase flash the 3.2.1 and configure from scratch.
- Make sure you have the up-to-date configurator (3.2.1).
- Use DYSF4PRO target to avoid confusion (I think this is irrelevant to your issue).

If your issue persists, please redirect it to the RCGroups thread for DYSF4PRO, as this (the Betaflight github issues) is not a end user support forum. Thanks in advance.

https://www.rcgroups.com/forums/showthread.php?2856541-DYS-F4-OSD-BEC-AiO-Flight-controller

closed 2017-10-18 by jflyper