betaflight/betaflight #4391: Request:. Spektrum F400 target

ijustwannafly, 2017-10-18:
Will there ever be a target for the spektrum f400 flight controller?

Many thanks.

Arakon, 2017-10-18:
Targets usually have to be added by the manufacturer, and I doubt spektrum will want that.

ijustwannafly, 2017-10-18:
I figured as much but wanted to ask to be sure.  Now I have to convince a friend to ditch it :)

mikeller, 2017-10-18:
Looks like a bog standard F4 based board with no frills - well possible that it will already work to 90 percent or so with one of the existing targets (try REVOLT). Somebody with the board, a multimeter, and a steady hand could probably ring out the motor pin to MCU pin connections, and based on that it should be possible to find a candidate target.

ijustwannafly, 2017-10-18:
I was playing with the idea of loading a revolt target as raceflight was revolt specific for a while.  As for the multimeter and pinout I would need specific instructions on how to do that.

mikeller, 2017-10-18:
Essentially, what you want to do is for every (signal) pad / pinhole on the board, check with the continuity tester which pin on the MCU it is connected to (if any), and write down the mapping. Some pins (like inverted serial ports) won't be connected, but things like motor output pins or the LED_STRIP pin will be connected directly to the MCU. Once we got the map (even if it's incomplete), we can check if it matches to an existing target, or create a new target based on the map (and some probing / educated guesses).

ijustwannafly, 2017-10-18:
I'm up to the challenge.  Hopefully I can map it in the next day or so.  Many thanks.

jflyper, 2017-10-19:
@ijustwannafly 
We may be able to help you with high-res pics of front and back of the board, possibly taken at an angle that stands out traces 😉

ijustwannafly, 2017-10-19:
I will try to get some high res pics.  I only have a nikon600 and some cellphones at my disposal.  I will try both high res pics and the mapping.  I think this flight controller would greatly benefit from betaflight and I can't stand the raceflight environment.  Can't even run dshot :(

ijustwannafly, 2017-10-20:
So I don't think I will be able to get this done.  My meter probes are too big and any camera I have at my disposal can't capture the traces.  I kind of wish there was software I could load onto the board that would send pulses to each pin and then maybe probe it that way.  Oh well.

ijustwannafly, 2017-10-20:
Just found this

https://github.com/SpektrumFPV/raceflight

Perhaps this might help?

mikeller, 2017-10-20:
I think this is a leftover from the early RaceFlight days, before they did a 'full rewrite' and went closed source. There does not seem to be a definition for your board in there, probably because it did not yet exist at the time that this code was current.

If the board is legacy anyway then it probably does not make too much sense to put a lot of effort into it.

Re probing, it is indeed hard to probe single pins on the MCU. Probing the assignment of a pad / pinhole on the board to one of two possible MCU pins is easier, by holding the probe into the gap between the MCU pins. Since things like motor pins are almost certainly only connected to one MCU pin, this makes it possible to deduct their position by confirming the pin they're on from the left and from the right.

ijustwannafly, 2017-10-20:
Thanks mikeller.  Even though it's legacy for raceflight it is still an f4 board.  Would it be better to map every single pin on the board or would getting the motors and some of the uarts suffice?  The only reason I am even putting some effort into this is because a handful of my flying buddies bought into the spektrum board (at least two or three each) and now they are being left behind firmware wise.  It would be near impossible now to recoup their money unless betaflight were possible.

mikeller, 2017-10-20:
Getting the motors and other easy ones would be a good start, and would allow educated guesses about the board being a clone of an existing board. If that does not pan out, you can always put more effort into mapping the rest, in order to create a new target for the board.

ijustwannafly, 2017-10-20:
So a bit of inspiration hit me when you said the motors should be mapped to a single leg on the processor.

I verified on all sides that motors 1 through 4 only showed continuity on a single leg.  I hope my magnifier was spot on.  
[image]

https://www.dropbox.com/s/cwxj85dy55niamy/IMG_20171019_213901.jpg?dl=0

DieHertz, 2017-10-20:
F446, it may be a bit harder than defining a new target.

ijustwannafly, 2017-10-20:
@DieHertz,

Any clarification on why the difficulty?  Mainly for my curiosity.

Also, I have an st link v2 programmer and an ftdi adaptor if that might help in any way.  I am willing to do what I can with this board while I have it.

mikeller, 2017-10-20:
Yes, the F446 is indeed a bit of a problem. No existing target is using this MCU (most use an F405), so this means it will have to be a new target. Speaking from experience with getting F411 targets to work, there will probably be unexpected hurdles to be overcome when it comes to details in how this particular MCU's IO works.

I would say that without support from Spektrum (by making schematics available, or just supplying some test boards to a developer) it will be a very hard ask to get this going. I think @AndersHoglund has connection to Spektrum, maybe he could ask them about this? (If this board isn't supported by RaceFlight any longer, then they don't have much to lose anyway.)

DieHertz, 2017-10-20:
I don't think we have glue code for this particular MCU. I'll have a look whether we do when I get to a computer.

ijustwannafly, 2017-10-20:
Thanks mikeller and DieHertz.

If it is a true dead end then I will stop all efforts.  Hopefully @AndersHoglund can chime something here as well.

For now I found led and rx3.

[image]
[image]

DieHertz, 2017-10-20:
Boards AND schematics would be good.

mikeller, 2017-10-20:
@DieHertz: That is a valid consideration. Reverse engineering this is probably not worth the (considerable) effort. Writing a target definition based on schematics is a lot less effort, and might be worth it.

ijustwannafly, 2017-10-20:
I agree that having the schematics would make things easier.  But I'm sure spektrum wouldn't make those documents too easily available.

AndersHoglund, 2017-10-20:
Sorry guys. I have no information on this FC. Even if I had, I could not reveal it. NDA. Spektrum must make it public first.

atomiclama, 2017-10-20:
Just for information, getting things up and running on the F446 is not a big deal there are only a couple of changes required. I have BF running on a ST32F446Nucleo board. There maybe issues I have not come across yet though.

ijustwannafly, 2017-10-20:
@atomiclama 

How would your bf nucleo build translate to the spektrum?  I don't know the code side of things but I'm guessing the stm pins should correspond the same for the stm series or am I wrong?

mikeller, 2017-10-20:
@atomiclama: Good to know!

AndersHoglund, 2017-10-20:
There are not only binaries in the rs2k repo, target definition sources as well.
https://github.com/rs2k/raceflight_community/tree/master/src/main/target/SPMFC400
Is this not use-able?

mikeller, 2017-10-20:
Good finding, @AndersHoglund! Everything except for the motors seems to be in here: https://github.com/rs2k/raceflight_community/blob/master/src/main/target/SPMFC400/target.h

mikeller, 2017-10-20:
@atomiclama: Can we borrow your F446 target definition?

atomiclama, 2017-10-20:
You can have it. 
Unfortunately it's not in a single commit, it's hidden in the middle of the stuff I'm working on. 
https://github.com/atomiclama/cleanflight/tree/bf_work

From memory there were only a couple of changes required to get it going. Sorry of on hols in a bit so can't help out much.

ijustwannafly, 2017-10-20:
@atomiclama 

Many thanks.  Hopefully this is another step toward even a beta-betaflight target :)

atomiclama, 2017-10-20:
I've updated the bf_work branch with the targets I'm currently using, might be of help.

ijustwannafly, 2017-10-20:
If a test build can be made I'm ready to test.  This is getting exciting.

mikeller, 2017-10-20:
@atomiclama: Ooh, I have spotted the makings of 'openLRSng over SPI' in your target. :-D That's been on my to-do list for such a long time. Any chance of you contributing that back to Betaflight? ;-)

ijustwannafly, 2017-10-20:
Is there anything more I can contribute for a test build?  Or should I wait for an actual betaflight target?  Either way I have the board for a short while for testing.

Many thanks to all.

mikeller, 2017-10-20:
Here we go: 
[betaflight_3.3.0_SPEKTRUMF400.zip](https://github.com/betaflight/betaflight/files/1403552/betaflight_3.3.0_SPEKTRUMF400.zip)

Source is in https://github.com/mikeller/betaflight/tree/added_spektrumf400_target, with thanks to @atomiclama for making F446 work, and @AndersHoglund for finding the RaceFlight target definition.

ijustwannafly, 2017-10-20:
Woohoo!  I sure can.  Wrapping up work soon and I will flash first thing.  :)

ijustwannafly, 2017-10-20:
Since the test is a 3.3 build should I use the configurator zip from github or will the latest chrome extension be enough?

mikeller, 2017-10-21:
The version from the web store should be fine for testing (some 3.3 features might not work).

ijustwannafly, 2017-10-21:
Just tried to flash the test hex.  It succeeded programming but on reconnect I get a usb device descriptor request failed.  Gonna try again once more.

mikeller, 2017-10-21:
After flashing, do you get any activity on the LEDs when powering up?

ijustwannafly, 2017-10-21:
Green and Amber solid, then alternating Green and Amber on and off and then solid green.

ijustwannafly, 2017-10-21:
Also, I tried installing drivers via the zadig program and each time the driver fails.  I have about 5 different boards I connect on a regular basis so I am fairly versed in how to get boards to connect.

mikeller, 2017-10-21:
Yay, it is booting. After powering it up, does one of the LEDs blink when you tilt the board?

jflyper, 2017-10-21:
@mikeller I think I can help this using the bare F405 dev board. Can you publish the branch?

ijustwannafly, 2017-10-21:
@mikeller 
No led activity when moving the board.

jflyper, 2017-10-21:
@ijustwannafly Can you describe the "_then alternating Green and Amber on and off_" precisely? A video clip would be great...

ijustwannafly, 2017-10-21:
@jflyper 

Video on the way.  I just noticed it does one of two sequences.  The video captured both.

mikeller, 2017-10-21:
@jflyper: https://github.com/mikeller/betaflight/tree/added_spektrumf400_target

ijustwannafly, 2017-10-21:
Here is the vid https://www.dropbox.com/s/7hq6yia1krvhd59/VID_20171020_151408.mp4?dl=0

Now a third sequence has shown up.  After the blinks have finished both amber and green will remain solid and then amber will turn off after about a minute.

jflyper, 2017-10-21:
I decode the LED sequence as
1. ACC/GYRO missing
2. RXLOSS
3. Don't know 😓

ijustwannafly, 2017-10-21:
I notice when I flash the racefight bb bin back onto the board it flashes slower than when betaflight flashes the test hex.  I have no reboot sequence and full chip erase selected on betaflight.  Auto baud rate.

mikeller, 2017-10-21:
Ok, I did some mostly cosmetic fixes: 

[betaflight_3.3.0_SPEKTRUMF400.zip](https://github.com/betaflight/betaflight/files/1403668/betaflight_3.3.0_SPEKTRUMF400.zip)

mikeller, 2017-10-21:
@jflyper: If you want to compare, that's where I got the information for the target from: https://github.com/rs2k/raceflight_community/blob/master/src/main/target/SPMFC400/target.h

jflyper, 2017-10-21:
Okay, Nucleo-F405 detects MPU9250 on SPI2 (PB13,PB14,PB15).
If the above does not, then we can take a look at SPI related stuff, starting with clock.

ijustwannafly, 2017-10-21:
Still getting a usb device not recognized but this time the led dance is shorter/faster and goes to just green led solid.

jflyper, 2017-10-21:
@ijustwannafly Are you on Linux?

ijustwannafly, 2017-10-21:
Windows 10 surface pro 3.

I can connect to the board when I reflash the raceflight firmware back onto it.  But when betaflight is on it is not recognized via usb.

ijustwannafly, 2017-10-21:
I just found this http://raceflightinfo.com/2016/05/25/flashing-raceflight/

Perhaps it is a bootloader issue.  I had to use an stlinkv2 programmer to clear the bootloader on my cc3d boards.  Perhaps this is the same?

jflyper, 2017-10-21:
@mikeller `timerClock` in `timer_stm32f4xx.c` need some mod if max system lock is variable among timers; all 180 v.s. 90/180 mixed. Probably former.

https://github.com/atomiclama/cleanflight/commit/d0fb65df0ad711d20915fb4d1fa6875a318cf632

[53 more replies]

closed 2018-01-07 by ijustwannafly